raspberrypi/rpi-eeprom #4: The script fails to detect rpi4 running in 64bit mode.

asavah, 2019-09-07:
https://github.com/raspberrypi/rpi-eeprom/blob/master/rpi-eeprom-update#L355

This check should be changed to make use of /proc/device-tree which should be more reliable.
`cat /proc/cpuinfo` on 4.19.x ARM64 kernel built from from raspberrypi/linux repo
```
processor       : 0
BogoMIPS        : 108.00
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 1
BogoMIPS        : 108.00
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 2
BogoMIPS        : 108.00
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 3
BogoMIPS        : 108.00
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3
```

MilhouseVH, 2019-09-07:
```
grep Revision /proc/cpuinfo | awk '{print $3}'
od -tx1 /proc/device-tree/system/linux,revision | head -1 | tr -d ' ' | cut -b9-
```
works in LibreELEC and Raspbian, returning:
RPi4 (4GB):
```
c03111
0c03111
```
and
RPi3+:
```
2a020d3
2a020d3
```

asavah, 2019-09-07:
in 64 bit mode:
`grep Revision /proc/cpuinfo | awk '{print $3}'`
returns nothing

`od -tx1 /proc/device-tree/system/linux,revision | head -1 | tr -d ' ' | cut -b9-`
returns
```
0c03111
```

timg236, 2019-09-07:
I might change this to get the board id from 'vcgencmd otp_dump' because the output format is defined by the firmware.

asavah, 2019-09-09:
Fixed by https://github.com/raspberrypi/rpi-eeprom/commit/47f985542de707b624af5e293c571b7c86163706

Thanks

closed 2019-09-09 by asavah