raspberrypi/rpi-eeprom #26: "latest" firmware repo

lurch, 2019-09-26:
Should `raspberry_pi4_network_boot_beta.md` be recommending https://github.com/raspberrypi/firmware as the source of latest firmware, rather than https://github.com/Hexxeh/rpi-update ?
ping @popcornmix

popcornmix, 2019-09-30:
Well, rpi-update the tool you use to get latest firmware from  https://github.com/raspberrypi/firmware (technically it comes from https://github.com/Hexxeh/rpi-firmware but the two repos have the same firmware contents).

At some point it should refer to using apt to get latest firmware, but at this stage rpi-update firmware is required for testing this. There is a fair amount of dependency between the bootcode in eeprom and the start.elf firmware so I suspect needing firmware as new as latest eeprom update will continue to be required for quite a while.

lurch, 2019-10-02:
To clarify, I wasn't querying whether "rpi-update firmware" was required for network booting (which is of course entirely understandable, given the "beta" status of it all), I was just querying whether https://github.com/raspberrypi/rpi-eeprom/blob/master/firmware/raspberry_pi4_network_boot_beta.md should be linking to https://github.com/raspberrypi/firmware (since IMHO that looks much more like an "official" URL than https://github.com/Hexxeh/rpi-firmware , even if the two repos have the same contents).

Although I now see that https://github.com/raspberrypi/firmware gets updated more frequently than https://github.com/Hexxeh/rpi-firmware so perhaps the distinction is deliberate? (in which case feel free to close this issue!)

timg236, 2020-05-02:
Network boot is no longer beta and that document doesn't exist. Closing

closed 2020-05-02 by timg236