raspberrypi/rpi-eeprom #27: VideoCore stops to respond with pieeprom-2019-09-25.bin on some RPI4s

mcerveny, 2019-09-29:
Hello.
I have about 15 RPI4s and I tested pieeprom-2019-09-25.bin.
All boots to login prompt (eg. DHCP, TFTP, NFS seems to be OK - firmware+kernel from  rpi-update) but about 50% of RPI4 have VideoCore blocked ("vcgencmd" hung).

See differences in "bcm2835-codec bcm2835-codec" startup:
```
# ### DMESG from OK
grep bcm2835 dmesg_o
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 cma=64M cma=256M video=HDMI-A-1:1280x1024M@60,margin_left=0,margin_right=0,margin_top=0,margin_bottom=0 smsc95xx.macaddr=DC:A6:32:0B:63:09 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=/dev/nfs nfsroot=10.199.100.21:/nfs/rootfs,vers=3,nolock ro ip=dhcp elevator=deadline rootwait logo.nologo systemd.show_status=no bcm2835_wdt.nowayout=1 bcm2835_wdt.heartbeat=10
[    0.034134] bcm2835-mbox fe00b880.mailbox: mailbox enabled
[    0.069387] bcm2835-dma fe007000.dma: DMA legacy API manager at (ptrval), dmachans=0x1
[    0.326814] gpiomem-bcm2835 fe200000.gpiomem: Initialised: Registers at 0xfe200000
[    0.960963] bcm2835-wdt bcm2835-wdt: Broadcom BCM2835 watchdog timer
[    0.962895] bcm2835-cpufreq: min=600000 max=1500000
[    1.262263] bcm2835-power bcm2835-power: Broadcom BCM2835 power domains driver
[   10.548433] bcm2835_vc_sm_cma_probe: Videocore shared memory driver
[   10.563911] bcm2835_mmal_vchiq: module is from the staging directory, the quality is unknown, you have been warned.
[   10.573834] bcm2835_codec: module is from the staging directory, the quality is unknown, you have been warned.
[   10.602303] bcm2835-codec bcm2835-codec: Device registered as /dev/video10
[   10.602338] bcm2835-codec bcm2835-codec: Loaded V4L2 decode
[   10.613746] bcm2835-codec bcm2835-codec: Device registered as /dev/video11
[   10.613788] bcm2835-codec bcm2835-codec: Loaded V4L2 encode
[   10.621454] bcm2835-codec bcm2835-codec: Device registered as /dev/video12
[   10.621487] bcm2835-codec bcm2835-codec: Loaded V4L2 isp
[   10.626609] bcm2835_v4l2: module is from the staging directory, the quality is unknown, you have been warned.
[   10.849370] snd_bcm2835: module is from the staging directory, the quality is unknown, you have been warned.
[   10.859843] bcm2835_audio soc:audio: card created with 8 channels

# ### DMESG from hung
grep bcm2835 dmesg_e
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 cma=64M cma=256M video=HDMI-A-1:1280x1024M@60,margin_left=0,margin_right=0,margin_top=0,margin_bottom=0 smsc95xx.macaddr=DC:A6:32:0B:66:E3 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=/dev/nfs nfsroot=10.199.100.21:/nfs/rootfs,vers=3,nolock ro ip=dhcp elevator=deadline rootwait logo.nologo systemd.show_status=no bcm2835_wdt.nowayout=1 bcm2835_wdt.heartbeat=10
[    0.034189] bcm2835-mbox fe00b880.mailbox: mailbox enabled
[    0.069679] bcm2835-dma fe007000.dma: DMA legacy API manager at (ptrval), dmachans=0x1
[    0.326883] gpiomem-bcm2835 fe200000.gpiomem: Initialised: Registers at 0xfe200000
[    0.960830] bcm2835-wdt bcm2835-wdt: Broadcom BCM2835 watchdog timer
[    0.962780] bcm2835-cpufreq: min=600000 max=1500000
[    1.280745] bcm2835-power bcm2835-power: Broadcom BCM2835 power domains driver
[   10.525541] bcm2835_vc_sm_cma_probe: Videocore shared memory driver
[   10.607926] bcm2835_mmal_vchiq: module is from the staging directory, the quality is unknown, you have been warned.
[   10.643553] bcm2835_v4l2: module is from the staging directory, the quality is unknown, you have been warned.
[   10.643558] bcm2835_codec: module is from the staging directory, the quality is unknown, you have been warned.
[   10.655302] bcm2835-codec bcm2835-codec: Device registered as /dev/video10
[   10.655339] bcm2835-codec bcm2835-codec: Loaded V4L2 decode
[   10.933601] snd_bcm2835: module is from the staging directory, the quality is unknown, you have been warned.
[   10.964786] bcm2835_audio soc:audio: card created with 8 channels
[   13.671798] bcm2835_mmal_vchiq: timed out waiting for sync completion
[   16.711764] bcm2835_mmal_vchiq: timed out waiting for sync completion
```

[dmesg_e.gz](https://github.com/raspberrypi/rpi-eeprom/files/3667425/dmesg_e.gz)
[dmesg_o.gz](https://github.com/raspberrypi/rpi-eeprom/files/3667426/dmesg_o.gz)

timg236, 2019-09-29:
Do the failing RPI4s still hang in vcgencmd if booted from sd-card (with the same bootloader version)? 

Please can you also try "systemctl disable rpi-eeprom-update" to disable any bootloader related gencmds at startup then see if vcgencmd works. There was a fix to the userland vcgencmd library for handling large messages. So maybe it's getting confused by the EEPROM config and hanging there. 

The bootloader doesn't stay resident in memory and provide services to the firmware or Linux so that's why the SD vs NFS and versions of userland libraries is interesting.

mcerveny, 2019-09-29:
SD-card booting with the same binaries (/boot/* and /opt/vc/* (VC+kernel)) without problems.
The problem is very persistent - multiple power-on-off leads to same results.
Also analyzed TFTP protocol sequence (snoop/tcpdump/wireshark) but seems to be equal (many failed request for "recover*"...).
Already tested without rpi-eeprom-update startup (yes, it was hung in booting) but the same results (any use of "vcgencmd", "tvservice" or application with vc_tvservice.h/bcm_host.h/ilclient.h API... after initial login lead to hung) on some RPI4s. 
```
root@thinvdi-client:~# vcgencmd version
Sep 24 2019 17:34:30 
Copyright (c) 2012 Broadcom
version cd3add54955f8fa065b414d8fc07c525e7ddffc8 (clean) (release) (start)
root@thinvdi-client:~# uname -a
Linux thinvdi-client 4.19.75-v7l+ #1270 SMP Tue Sep 24 18:51:41 BST 2019 armv7l GNU/Linux
```

timg236, 2019-09-30:
That's interesting, do all the network booting boards have identical file-systems ?

The firmware doesn't do a lot more after starting the Linux kernel except the turbo clock boost then waiting for gencmd or mbox events. Does it fail if you set core_freq=500 and core_freq_min=500 in config.txt ?

It would be interesting to try "sudo vcdbg log msg" to see if that gives any clues as to the firmware status. Adding "start_db=1" to config.txt should give more information.

timg236, 2019-09-30:
Ah, found the issue. A fix in start.elf is required for the virtual TFTP file-system. I'll update this bug once the fix is available via rpi-update.

timg236, 2019-09-30:
This should now be fixed in the rpi-update firmware if you download all the start*.elf and fixup*.dat files.

mcerveny, 2019-10-01:
Yes, you repaired it !
I updated (rpi-update) to new versions.
```
root@thinvdi-client:~#  vcgencmd version
Sep 30 2019 14:37:51 
Copyright (c) 2012 Broadcom
version b1477501b8675de389acd184c2b7e7c4b2e39e4a (clean) (release) (start)
root@thinvdi-client:~# uname -a
Linux thinvdi-client 4.19.75-v7l+ #1271 SMP Mon Sep 30 13:51:15 BST 2019 armv7l GNU/Linux
```
I tested network boot with 11 RPI4s without problems.
Thanks!

closed 2019-10-01 by mcerveny

timg236, 2019-10-02:
Great. Thanks for verifying the fix :)