raspberrypi/rpi-eeprom #28: Boot from USB

dawidroz, 2019-10-04:
When are you planning implement booting from USB? 

Thanks

lenisko, 2019-10-04:
As it was already mentioned they will take care of it after network boot which is at BETA phase right now.

alesoft73, 2019-10-06:
We attend the usb boot.. many thanks!!!!

pepeEL, 2019-10-11:
Now we are waiting for boot USB :)

timg236, 2019-10-11:
Not a bug - closing issue. 

Support and feature requests are best discussed on the Raspberry Pi forums where this question has already been answered - typically the Advanced Users forum
https://www.raspberrypi.org/forums/

AndrewCarterUK, 2019-10-11:
Hello,

Is there a chance this could be re-opened? Issues don't have to be bugs - they can also be feature requests and GitHub has some helpful labels for categorizing them.

I was subscribed to this so that I had a notification when there was an update. Our company can't switch to using Raspberry Pi 4's until this issue has been resolved, so this thread was providing us value.

timg236, 2019-10-11:
Ok since it is actually a technical feature there is no active issue I’ll reopen it so you get a notification instead of having to keep an eye on all the commits. We generally don’t pre announce betas or post roadmaps so “are we there yet questions” tend to be ignored :)

ykla, 2019-10-21:
Will support USB 3.0？

lurch, 2019-11-22:
AFAIK (and I might be wrong!) I think the eventual plan is for the full-size USB ports to support [USB host boot](https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/msd.md) and for the USB-C port to support [USB device boot](https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/device.md).

ykla, 2019-12-13:
I have been waiting for it for a long time.

drduker, 2019-12-19:
All I want for Christmas is boot from usb.

timg236, 2019-12-19:
Locked conversation because of time-wasting posts (now deleted) so people tracking this issue don't get spammed. When USB MSD boot is available this issue will be updated along with normal forum announcements.
If you want to discuss something Raspberry Pi related then the forums are the best place. Issues on this Github are for technical discussion.

timg236, 2020-05-15:
The first beta #127 has been uploaded but it will require updated firmware via rpi-update. That's a few days away.

pepeEL, 2020-05-17:
Hi. When we can test this beta?

yassineim, 2020-05-17:
We still need the firmware, which is apparently a few days away

ghollingworth, 2020-05-17:
> Hi. When we can test this beta?

You have been told on a number of occasions that it's coming soon and requires an update to the firmware first.  We will then add a message here which you will get as an email if you've subscribed.  If you ask one more time I will block you from the organisation.

2play, 2020-05-17:
hi to all,
i get this checksum error
**lib/firmware/raspberrypi/bootloader/beta/recovery.bin: FAILED**
when I try to do -a update to the beta pieeprom-2020-05-15.bin
The recovery.bin with same date fails the test.
rip-update done also, apt update and upgrade.

Does this file needs updating so -a can work our end?

tx in advance and kind regards

timg236, 2020-05-18:
Wait until the package is published

2play, 2020-05-18:
@timg236 thank you . Just a quick que. Would it require rpi-update (Kernel 4.19.97 bumped to 5.4.40) or will be ok after published with existing kernel?

LufyCZ, 2020-05-18:
> hi to all,
> i get this checksum error
> **lib/firmware/raspberrypi/bootloader/beta/recovery.bin: FAILED**
> when I try to do -a update to the beta pieeprom-2020-05-15.bin
> The recovery.bin with same date [...]

You can use -i to ignore checksum checks.

Marcinoo97, 2020-05-18:
Will boot from type-c port on a dwc2 controller be possible?

lurch, 2020-05-18:
> Will boot from type-c port on a dwc2 controller be possible?

Just out of curiosity, why would you want to do that?

timg236, 2020-05-18:
> Will boot from type-c port on a dwc2 controller be possible?

no

Marcinoo97, 2020-05-18:
> > Will boot from type-c port on a dwc2 controller be possible?
> 
> Just out of curiosity, why would you want to do that?

I'm using an OS with a broken xhci support and fixing it will take a long time. Only dwc2 works there.

timg236, 2020-05-18:
Which OS? 

U-boot already has working xHC which should be a good reference but is probably GPL. Although at least the standard is open and whilst large isn't that bad. The VLI implementation didn't really need much in the way of Pi specific quirks.

Marcinoo97, 2020-05-18:
Windows 10

2play, 2020-05-19:
@timg236 
today i retested. changed to beta, update/upgrade and let me upgrade with -a
rebooted, tested if ok
`pi@playbox:~ $ sudo rpi-eeprom-update
BCM2711 detected
BOOTLOADER: up-to-date
CURRENT: Fri 15 May 2020 10:05:52 AM UTC (1589537152)
 LATEST: Fri 15 May 2020 10:05:52 AM UTC (1589537152)
 FW DIR: /lib/firmware/raspberrypi/bootloader/beta
VL805: up-to-date
CURRENT: 000137ad
 LATEST: 000137ad
`
Unfo the usb does not get power at all to boot. Am I missing something
Use a USB 3 NVMe m.2 Sabrent 512GB and also a SSD Sandisk Extreme Pro

timg236, 2020-05-19:
Locked until the software is released. As usual support questions should go to the forums first

timg236, 2020-05-20:
The beta firmware and bootloader releases are now available. Please see the announcement on the Raspberry Pi General forum.

https://www.raspberrypi.org/forums/viewtopic.php?t=274595&p=1663644#p1663644

Support questions should go to the forums first or bug reports in new specific Github issues. I'll close this bug once the release has made it out of beta.

2play, 2020-05-20:
Thank you for the update

KC-inDomus, 2020-06-18:
In the last release not i read "Promote the latest beta EEPROM and recovery.bin to stable and feature freeze USB MSD support until a production release is ready."

What you mean with "feature freeze USB MSD support until a production release is ready"?

2play, 2020-06-18:
@KC-inDomus  change to stable and update the firmware. its now officially published

lurch, 2020-06-18:
See the "Firmware release status" section of https://www.raspberrypi.org/documentation/hardware/raspberrypi/booteeprom.md for more information.

Misiu, 2020-07-01:
I'm a bit confused 😕
According to [the docs](https://www.raspberrypi.org/documentation/hardware/raspberrypi/booteeprom.md):

> USB boot is not currently supported. Once it is ready a beta release will be announced on the Raspberry Pi Forums.

Should the docs be updated? If the firmware is stable then it isn't a beta release and USB boot is officially supported.

timg236, 2020-07-01:
From releases.md "N.B. Although the bootloader has now been promoted to a stable / feature-freeze release USB MSD should still be considered BETA software because this requires updates to the GPU firmware."

pieter-groeneweg, 2020-07-01:
that IS confusing.
so how do we track the progress on the GPU firmware?

timg236, 2020-07-01:
See earlier comment on this bug 

"Support questions should go to the forums first or bug reports in new specific Github issues. I'll close this bug once the release has made it out of beta."

goon818, 2020-07-16:
Hello, i use the raspberry pi 4 one with 4GB and one with 8GB RAM. The new function Boot from USB only working on the 4GB device. On the 8 GB device i get the message "card not detected." and four time "failed to open device "scard". I try the last stable and the last beta bootloader. I try it with the same USB Stick. On the 4GB it's runing and on the 8GB not. Could it be? What can i try to run it?

Dark-Sky, 2020-07-16:
> Hello, i use the raspberry pi 4 one with 4GB and one with 8GB RAM. The new function Boot from USB only working on the 4GB device. On the 8 GB device i get the message "card not detected." and four [...]

Not sure what version you are running but the latest my 8G is working ok with sdcard and usb boot:

[ray@pi4 ~]$ sudo rpi-eeprom-update 
BCM2711 detected
VL805 firmware in bootloader EEPROM
BOOTLOADER: up-to-date
CURRENT: Thu 16 Jul 2020 10:44:17 AM UTC (1594896257)
 LATEST: Thu 16 Jul 2020 10:44:17 AM UTC (1594896257)
 FW DIR: /lib/firmware/raspberrypi/bootloader/beta
VL805: up-to-date
CURRENT: 000138a1
 LATEST: 000138a1

goon818, 2020-07-16:
> 
> [ray@pi4 ~]$ sudo rpi-eeprom-update
> BCM2711 detected
> VL805 firmware in bootloader EEPROM
> BOOTLOADER: up-to-date
> CURRENT: Thu 16 Jul 2020 10:44:17 AM UTC (1594896257)
> LATEST: Thu 16 Jul 2020 10:44:17 AM UTC (1594896257)
> FW DIR: /lib/firmware/raspberrypi/bootloader/beta
> VL805: up-to-date
> CURRENT: 000138a1
> LATEST: 000138a1
> 
> [ray@pi4 ~]$ free
> total used free shared buff/cache available
> Mem: 8013984 208340 7383176 24436 422468 7673320
> Swap: 12020972 0 12020972
> [ray@pi4 ~]$ lsblk
> NAME MAJ:MIN RM SIZE RO TYPE MOUNTPOINT
> sda 8:0 1 14.5G 0 disk
> ├─sda1 8:1 1 213.6M 0 part /boot
> └─sda2 8:2 1 14.3G 0 part /
> zram0 254:0 0 11.5G 0 disk [SWAP]

Thank you for your help. I compare your and my bootloader version. i have the same. Then i go to raspi-config and try to change the bootloader option to usb. i get an error. The i use the command sudo rpi-eeprom-update again. After this reboot the system and after this i can change the boot option to usb and it's run.  Without your answer i has not try it again. THX

timg236, 2020-07-16:
Please post support questions on the forum. This Issue is about USB boot as a feature not for general support requests

timg236, 2020-09-18:
USB msd boot support is now available in the default bootloader version 2020-09-03

closed 2020-09-18 by timg236