raspberrypi/rpi-eeprom #41: rpi-eeprom-update cannot get current VL805 firmware version when run as non-root user

andrum99, 2019-10-23:
When run as a non-root user, rpi-eeprom-update shows the current and latest applicable version of bootloader EEPROM. It shows a blank space where the VL805 EEPROM version should be, and shows the latest applicable VL805 EEPROM version that is available. When run using sudo it successfully retrieves the VL805 EEPROM version.

> pi@livingroom:~ $ rpi-eeprom-update
> Bootloader EEPROM is up to date
> BOOTLOADER
> CURRENT: Wed 16 Oct 17:00:03 UTC 2019 (1571245203)
> LATEST: Tue 10 Sep 10:41:50 UTC 2019 (1568112110)
> VL805
> CURRENT:
> LATEST: 000137ab
> pi@livingroom:~ $ sudo rpi-eeprom-update
> Bootloader EEPROM is up to date
> BOOTLOADER
> CURRENT: Wed 16 Oct 17:00:03 UTC 2019 (1571245203)
> LATEST: Tue 10 Sep 10:41:50 UTC 2019 (1568112110)
> VL805
> CURRENT: 000137ab
> LATEST: 000137ab

timg236, 2019-10-23:
Yep. Unfortunately the version number can only be obtained as the root user because it involves reading magic config space registers. A setuid cutdown vl805 tool might be an option but it’s a low priority because the automatic service runs as root and interactive usage which is relatively rare can be fixed by running as sudo

andrum99, 2019-10-24:
Fixed by #42

closed 2019-10-24 by andrum99