raspberrypi/rpi-eeprom #43: Raspbian Deb package depends on Python (Python 2)

jriguera, 2019-10-27:
Raspbian package `rpi-eeprom` depends on `python` (from `apt-cache show rpi-eeprom`):
```
Package: rpi-eeprom
Version: 2.0-1
Architecture: all
Maintainer: Serge Schneider <serge@raspberrypi.org>
Installed-Size: 427
Depends: rpi-eeprom-images, libraspberrypi-bin, python, raspberrypi-bootloader (>= 1.20190819)
Recommends: flashrom
Homepage: https://github.com/raspberrypi/rpi-eeprom/
[...]
```
Python dependency defaults to Python2 and it is getting out of support in 2020. I have checked the program `rpi-update` and it works with Python3 (checked with Python 3.7.3), so I am wondering if it is possible to update the shebang and the Raspbian Deb package to depend on Python3.

timg236, 2019-10-28:
rpi-eeprom-config must run on systems with only Python2. Happy to accept a PR which makes it work on both Python3 and Python2

MilhouseVH, 2019-10-28:
See #45 for my effort...

XECDesign, 2019-10-28:
It's still supported in Buster no matter what happens upstream.

From my point of view, there's absolutely no harm in leaving it as is or changing it (as far as Raspbian goes). If it's changed to Python 3, it may inconvenience LibreElec in the short term.

The only reasons I can see to change it are that it's just one less thing to do a few years from now (when we may be forced to) and to be more considerate of distros which don't ship with Python 2.

timg236, 2019-11-04:
Change #45 merged so the script will work with either Python2 or Python3. Exactly what the package depends upon is a different issue and only affects distros which re-use the the package directly but want different dependencies

closed 2019-11-04 by timg236