raspberrypi/rpi-eeprom #48: License clarification of vl805

nullr0ute, 2019-11-04:
The LICENSE files states that everything in the root is:
<pre>
Files: *
Copyright: 2019, Raspberry Pi (Trading) Ltd.
License: BSD-3"
</pre>

Is the vl805 binary under this license? If so is the source published somewhere so it could be built for aarch64 use too? If not I believe the license needs to be clarified.

timg236, 2019-11-04:
vl805 should be moved to the firmware directory

timg236, 2019-11-06:
I'm a bit busy at the moment but in-case anyone else needs it I plan to replace the usage of vl805 for version checks with lspci as described 'trejan' on the forums
https://www.raspberrypi.org/forums/viewtopic.php?t=255001&p=1559532
sudo lspci -d 1106:3483 -xxx | awk ' { if ($1 == "50:") { print "VL805 FW version: " $5 $4 $3 $2 } } '

The vl805 would then only be needed if the recovery.bin method is not available although given that vl805 to update the controller whilst it's running can hang I think we want to avoid usage of that

MilhouseVH, 2019-11-06:
LibreELEC doesn't currently include `lspci` in RPi4 images - maybe we can include it by building `pciutils` for RPi4 (need to check) but before I do, would it be possible for `rpi-eeprom-update` to fall back on `vl805` when `lspci` isn't available?

timg236, 2019-11-06:
No problem, I'll use command -v to check for lspci and failover to vl805 if lspci is not present

nullr0ute, 2019-11-07:
For reference the fwupd (https://github.com/fwupd/fwupd) are working with Via to add firmware update support via LVFS for all their USB controllers and hubs etc. Hubs support is already in the latest release, controllers will be added shortly, not sure if that helps here.

XECDesign, 2019-11-07:
I don't remember the details, but I found fwupd to be unsuitable. IIRC, it has a ton of unnecessary dependencies, including UEFI and it doesn't give as much control as we would like.

mi-hol, 2019-11-10:
Well using fwupd  means to make use of the established LINUX service "Linux Vendor Firmware Service" ([LVFS](https://fwupd.org))  has many long-term benefits, that should be weighted & considered properly when making short-term decisions. 
The biggest benefit I can see is to be on par with the simplicity of MSFT Windows Update!

XECDesign, 2019-11-11:
Agreed. It was the very first thing I looked at.

timg236, 2019-11-11:
vl805 binary moved to firmware directory with appropriate license so resolving issue.

closed 2019-11-11 by timg236