raspberrypi/rpi-eeprom #55: USB Drives dont spin down

workshopninja, 2019-11-20:
I am using my Pi as a NAS running OpenMediaVault - with a dual raid box connected via usb 3.
Before installing this update the USB drives lights and fans shutdown with the PI.
After the update fans and lights stay on - and then when I reboot the Pi the drives all reboot...

Cris--, 2019-11-20:
Maybe a solution for here:
https://forum.libreelec.tv/thread/20648-rpi4-usb-ports-does-not-turn-off-power-after-shutdown/

workshopninja, 2019-11-20:
I did the update to regain power button features using "dtoverlay=gpio-shutdown"

This allows a single button (between GPIO3 and gnd) to shutdown the system and wake from said shutdown.

It seems like the link above will disable that overlay from working.

I have limited knowledge of the hardware - are my desires mutually exclusive?

andrum99, 2019-11-21:
According to the documentation, these two things are mutually exclusive. A solution would be to use an external piece of hardware to do the power management.

timg236, 2019-11-21:
As andrum99 says these are mutually exclusive because POWER_OFF_ON_HALT disables power the SOC so there's no way to wakeup from a GPIO.

An external piece of hardware could pull GLOBAL_EN low to wakeup from HALT with POWER_OFF_ON_HALT.

andrum99, 2019-11-21:
> An external piece of hardware could pull GLOBAL_EN low to wakeup from HALT with POWER_OFF_ON_HALT.

In which case the solution is to pull GLOBAL_EN low instead of GPIO 3.

timg236, 2019-11-21:
> > An external piece of hardware could pull GLOBAL_EN low to wakeup from HALT with POWER_OFF_ON_HALT.
> 
> In which case the solution is to pull GLOBAL_EN low instead of GPIO 3.

Yes, but pulling GLOBAL_EN while the Pi is running will reset the PMIC instead of doing a nice shutdown. I think GPIO3 is sometimes used as a fake key for ACPI shutdown i.e. you need different pins for shutdown and wake up

andrum99, 2019-11-21:
> Yes, but pulling GLOBAL_EN while the Pi is running will reset the PMIC instead of doing a nice shutdown. I think GPIO3 is sometimes used as a fake key for ACPI shutdown i.e. you need different pins [...]

Ah, sorry. I was meaning that instead of using GPIO 3 for wake up, you can use GLOBAL_EN instead. Obviously you can continue to use GPIO 3 for shutdown.

workshopninja, 2019-11-22:
Thank you @timg236 - I thought that might be the case but wanted confirmation from someone who has read the documentation ;)...
Not really a problem - it’s a NAS - it will be on all the time, I just noticed the difference and wondered why.

workshopninja, 2019-11-22:
I don’t know how to close an issue?

timg236, 2019-11-22:
NP I'll close it.

closed 2019-11-22 by timg236