raspberrypi/rpi-eeprom #58: No DHCPACK

marcolefo, 2019-11-22:
Hi

I am trying to netboot my pi4 with no success.

After plug the pi, the green LED is fix. I saw DHCP requests on my server.
The Pi keep requesting DHCP server and the server answer with the IP.
I only see DHCPDISCOVER and DHCPOFFER. No DHCPACK.

I have tried with pieeprom-2019-11-18.bin and pieeprom-2019-10-16.bin.

My dhcp server is a isc-dhcp on a Debian Jessie. Note that it works for a pi3.
The definition for the subnet is :
subnet 172.16.7.0 netmask 255.255.255.0 {
        interface eth0;
        allow unknown-clients;
        range 172.16.7.230 172.16.7.234;
        option subnet-mask 255.255.255.0;
        option routers 172.16.7.254;

        next-server 172.16.7.1;
        option tftp-server-name "172.16.7.1";
}

Muxxxi, 2019-11-22:
Hi

I don't know if it helps, but the following configuration works for me so far.
I'm running the Docker Image `networkboot/dhcpd`.

```
subnet 192.168.1.0 netmask 255.255.255.0 {
        range 192.168.1.20 192.168.1.100;
      	next-server 192.168.1.1;
        option tftp-server-name "192.168.1.1";
    	always-broadcast false;
    	option vendor-class-identifier "PXEClient";
        option vendor-encapsulated-options "Raspberry Pi Boot";
        option routers 192.168.1.1;
        option domain-name-servers 8.8.8.8;
}
```

timg236, 2019-11-22:
Please can you post wireshark capture. Might be expecting SI_ADDR but won’t be able to look at this for a few days.

marcolefo, 2019-11-25:
> Hi
>
> I don't know if it helps, but the following configuration works for me so far.
> I'm running the Docker Image `networkboot/dhcpd`.
>
> ```
> subnet 192.168.1.0 netmask 255.255.255.0 {
>         range [...]

No change with adding these options

marcolefo, 2019-11-25:
> 
> 
> Please can you post wireshark capture. Might be expecting SI_ADDR but won’t be able to look at this for a few days.

With fiIter `udp.port==68` see only DHCP Discover
```
No.     Time           Source                Destination           Protocol Length Info
     40 17.643001      0.0.0.0               255.255.255.255       DHCP     364    DHCP Discover - Transaction ID 0x5ad50db4

Frame 40: 364 bytes on wire (2912 bits), 364 bytes captured (2912 bits) on interface 0
Ethernet II, Src: dc:a6:32:27:76:bc (dc:a6:32:27:76:bc), Dst: Broadcast (ff:ff:ff:ff:ff:ff)
    Destination: Broadcast (ff:ff:ff:ff:ff:ff)
    Source: dc:a6:32:27:76:bc (dc:a6:32:27:76:bc)
    Type: IPv4 (0x0800)
Internet Protocol Version 4, Src: 0.0.0.0, Dst: 255.255.255.255
    0100 .... = Version: 4
    .... 0101 = Header Length: 20 bytes (5)
    Differentiated Services Field: 0x00 (DSCP: CS0, ECN: Not-ECT)
    Total Length: 350
    Identification: 0x506a (20586)
    Flags: 0x0000
    Time to live: 64
    Protocol: UDP (17)
    Header checksum: 0x2926 [validation disabled]
    [Header checksum status: Unverified]
    Source: 0.0.0.0
    Destination: 255.255.255.255
User Datagram Protocol, Src Port: 68, Dst Port: 67
    Source Port: 68
    Destination Port: 67
    Length: 330
    Checksum: 0xfad3 [unverified]
    [Checksum Status: Unverified]
    [Stream index: 0]
Bootstrap Protocol (Discover)
    Message type: Boot Request (1)
    Hardware type: Ethernet (0x01)
    Hardware address length: 6
    Hops: 0
    Transaction ID: 0x5ad50db4
    Seconds elapsed: 0
    Bootp flags: 0x0000 (Unicast)
        0... .... .... .... = Broadcast flag: Unicast
        .000 0000 0000 0000 = Reserved flags: 0x0000
    Client IP address: 0.0.0.0
    Your (client) IP address: 0.0.0.0
    Next server IP address: 0.0.0.0
    Relay agent IP address: 0.0.0.0
    Client MAC address: dc:a6:32:27:76:bc (dc:a6:32:27:76:bc)
    Client hardware address padding: 00000000000000000000
    Server host name not given
    Boot file name not given
    Magic cookie: DHCP
    Option: (53) DHCP Message Type (Discover)
        Length: 1
        DHCP: Discover (1)
    Option: (55) Parameter Request List
        Length: 14
        Parameter Request List Item: (1) Subnet Mask
        Parameter Request List Item: (3) Router
        Parameter Request List Item: (43) Vendor-Specific Information
        Parameter Request List Item: (60) Vendor class identifier
        Parameter Request List Item: (66) TFTP Server Name
        Parameter Request List Item: (67) Bootfile name
        Parameter Request List Item: (128) DOCSIS full security server IP [TODO]
        Parameter Request List Item: (129) PXE - undefined (vendor specific)
        Parameter Request List Item: (130) PXE - undefined (vendor specific)
        Parameter Request List Item: (131) PXE - undefined (vendor specific)
        Parameter Request List Item: (132) PXE - undefined (vendor specific)
        Parameter Request List Item: (133) PXE - undefined (vendor specific)
[... 322 lines cut ...]
        Parameter Request List Item: (128) DOCSIS full security server IP [TODO]
        Parameter Request List Item: (129) PXE - undefined (vendor specific)
        Parameter Request List Item: (130) PXE - undefined (vendor specific)
        Parameter Request List Item: (131) PXE - undefined (vendor specific)
        Parameter Request List Item: (132) PXE - undefined (vendor specific)
        Parameter Request List Item: (133) PXE - undefined (vendor specific)
        Parameter Request List Item: (134) PXE - undefined (vendor specific)
        Parameter Request List Item: (135) PXE - undefined (vendor specific)
    Option: (93) Client System Architecture
        Length: 2
        Client System Architecture: IA x86 PC (0)
    Option: (94) Client Network Device Interface
        Length: 3
        Major Version: 2
        Minor Version: 1
    Option: (97) UUID/GUID-based Client Identifier
        Length: 17
        Client Identifier (UUID): 6aee7243-7243-6aee-4372-ee6a4372ee6a
    Option: (60) Vendor class identifier
        Length: 32
        Vendor class identifier: PXEClient:Arch:00000:UNDI:002001
    Option: (255) End
        Option End: 255

```

timg236, 2019-11-25:
I think we'd need to see the DHCPOFFER packets the DHCPDISCOVER is constant and looks reasonable

marcolefo, 2019-11-26:
I know it's strange but I have captured no DHCPOFFER packet... but in dhcp log I can see : 

```
Nov 25 10:47:14  dhcpd: DHCPDISCOVER from dc:a6:32:27:76:bc via 10.10.10.254
Nov 25 10:47:14  dhcpd: DHCPOFFER on 10.10.10.230 to dc:a6:32:27:76:bc via 10.10.10.254
Nov 25 10:47:22  dhcpd: DHCPDISCOVER from dc:a6:32:27:76:bc via 10.10.10.254
Nov 25 10:47:22  dhcpd: DHCPOFFER on 10.10.10.230 to dc:a6:32:27:76:bc via 10.10.10.254
Nov 25 10:47:38  dhcpd: DHCPDISCOVER from dc:a6:32:27:76:bc via 10.10.10.254
Nov 25 10:47:38  dhcpd: DHCPOFFER on 10.10.10.230 to dc:a6:32:27:76:bc via 10.10.10.254
Nov 25 10:47:59  dhcpd: DHCPDISCOVER from dc:a6:32:27:76:bc via 10.10.10.254
Nov 25 10:47:59  dhcpd: DHCPOFFER on 10.10.10.230 to dc:a6:32:27:76:bc via 10.10.10.254
Nov 25 10:48:07  dhcpd: DHCPDISCOVER from dc:a6:32:27:76:bc via 10.10.10.254
Nov 25 10:48:07  dhcpd: DHCPOFFER on 10.10.10.230 to dc:a6:32:27:76:bc via 10.10.10.254
Nov 25 10:48:23  dhcpd: DHCPDISCOVER from dc:a6:32:27:76:bc via 10.10.10.254
Nov 25 10:48:23  dhcpd: DHCPOFFER on 10.10.10.230 to dc:a6:32:27:76:bc via 10.10.10.254
```

ghollingworth, 2019-11-26:
Where are you trying to capture those packets from?  Are you doing that on the DHCP server itself?

timg236, 2019-11-26:
I think you might need to allow UDP port 67 as well as 68, or filter by MAC address instead

timg236, 2020-01-21:
Please could you retry with the latest beta ?

Mavyre, 2020-02-16:
Hi,

I'm answering to you @timg236 as I'm running into the exact same issue with a Raspberry Pi 4 and a Synology's DHCP server. Note that it works perfectly on a Raspberry pi 3B+

`21:39:02.683460 IP (tos 0x0, ttl 64, id 35665, offset 0, flags [none], proto UDP (17), length 371)
    0.0.0.0.68 > 255.255.255.255.67: BOOTP/DHCP, Request from b8:27:eb:e3:8b:5d, length 343, xid 0x48a5fcdb, Flags [none]
	  Client-Ethernet-Address b8:27:eb:e3:8b:5d
	  Vendor-rfc1048 Extensions
	    Magic Cookie 0x63825363
	    DHCP-Message Option 53, length 1: Request
	    Client-ID Option 61, length 7: ether b8:27:eb:e3:8b:5d
	    Requested-IP Option 50, length 4: 192.168.82.102
	    MSZ Option 57, length 2: 1500
	    Vendor-Class Option 60, length 45: "dhcpcd-6.7.1:Linux-4.19.97-v7+:armv7l:BCM2835"
	    Hostname Option 12, length 12: "net-failover"
	    T145 Option 145, length 1: 1
	    Parameter-Request Option 55, length 14: 
	      Subnet-Mask, Classless-Static-Route, Static-Route, Default-Gateway
	      Domain-Name-Server, Hostname, Domain-Name, BR
	      NTP, Lease-Time, Server-ID, RN
	      RB, Option 119
21:39:02.735598 IP (tos 0xc0, ttl 64, id 2605, offset 0, flags [none], proto UDP (17), length 374)
    192.168.82.2.67 > 192.168.82.102.68: BOOTP/DHCP, Reply, length 346, xid 0x48a5fcdb, Flags [none]
	  Your-IP 192.168.82.102
	  Server-IP 192.168.82.2
	  Client-Ethernet-Address b8:27:eb:e3:8b:5d
	  Vendor-rfc1048 Extensions
	    Magic Cookie 0x63825363
	    DHCP-Message Option 53, length 1: ACK
	    Server-ID Option 54, length 4: 192.168.82.2
	    Lease-Time Option 51, length 4: 86400
	    RN Option 58, length 4: 43200
	    RB Option 59, length 4: 75600
	    Subnet-Mask Option 1, length 4: 255.255.255.0
	    BR Option 28, length 4: 192.168.82.255
	    Hostname Option 12, length 12: "net-failover"
	    Domain-Name-Server Option 6, length 8: 192.168.82.2,192.168.82.1
	    Domain-Name Option 15, length 15: "home.vydees.net"
	    T119 Option 119, length 17: 4.104.111.109.101.6.118.121.100.101.101.115.3.110.101.116.0
	    Default-Gateway Option 3, length 4: 192.168.82.1
21:39:02.787538 IP (tos 0x0, ttl 64, id 20627, offset 0, flags [none], proto UDP (17), length 350)
    0.0.0.0.68 > 255.255.255.255.67: BOOTP/DHCP, Request from dc:a6:32:2a:ad:db, length 322, xid 0xd7910b3d, Flags [none]
	  Client-Ethernet-Address dc:a6:32:2a:ad:db
	  Vendor-rfc1048 Extensions
	    Magic Cookie 0x63825363
	    DHCP-Message Option 53, length 1: Discover
	    Parameter-Request Option 55, length 14: 
	      Subnet-Mask, Default-Gateway, Vendor-Option, Vendor-Class
	      TFTP, BF, Option 128, Option 129
	      Option 130, Option 131, Option 132, Option 133
	      Option 134, Option 135
	    ARCH Option 93, length 2: 0
	    NDI Option 94, length 3: 1.2.1
	    GUID Option 97, length 17: 0.195.120.127.194.195.120.127.194.195.120.127.194.195.120.127.194
	    Vendor-Class Option 60, length 32: "PXEClient:Arch:00000:UNDI:002001"
21:39:10.787356 IP (tos 0x0, ttl 64, id 20628, offset 0, flags [none], proto UDP (17), length 350)
    0.0.0.0.68 > 255.255.255.255.67: BOOTP/DHCP, Request from dc:a6:32:2a:ad:db, length 322, xid 0xd7910b3d, Flags [none]
	  Client-Ethernet-Address dc:a6:32:2a:ad:db
	  Vendor-rfc1048 Extensions
	    Magic Cookie 0x63825363
	    DHCP-Message Option 53, length 1: Discover
	    Parameter-Request Option 55, length 14: 
	      Subnet-Mask, Default-Gateway, Vendor-Option, Vendor-Class
	      TFTP, BF, Option 128, Option 129
	      Option 130, Option 131, Option 132, Option 133
	      Option 134, Option 135
	    ARCH Option 93, length 2: 0
	    NDI Option 94, length 3: 1.2.1
	    GUID Option 97, length 17: 0.195.120.127.194.195.120.127.194.195.120.127.194.195.120.127.194
	    Vendor-Class Option 60, length 32: "PXEClient:Arch:00000:UNDI:002001"`

And this indefinitely. The boot works fine if I flash the EEPROM with the TFTP address directly though. But I'd like it not to be hardcoded into the EEPROM...

ibot3, 2020-03-31:
From [raspberrypi.org](https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=252501&start=50#p1628250):
> Hello everybody,
>
> I successfully netbooted a RPi4 Model B with a dnsmasq PXE Server.
>
> However, the same setup does not seem to work when trying to boot via a DHCP relay.
> The Pi sends out DHCP Discovers and always receives the related Offer. But nothing more, no DHCP request, no TFTP requests.
> After some retries, the Pi gives up.
>
> Does someone have an idea why the Pi does nothing when receiving the DHCP offer?
>
> DHCP DISCOVER packet: https://gist.github.com/ibot3/f9ffda7b8 ... d76b476af2
> DHCP OFFER packet: https://gist.github.com/ibot3/be19ec699 ... c91290c91c

I have the problem described above when using a DHCP relay. I also attached the wireshark captures of the packets.

timg236, 2020-03-31:
There's no explicit support for relay agents so this would have to be transparent to the client Pi. I'd recommend updating to https://github.com/raspberrypi/rpi-eeprom/blob/master/firmware/beta/pieeprom-2020-03-19.bin and enabling BOOT_UART and capturing the log either via a USB serial or UART on another Pi.
That should tell you whether client Pi sees the DHCP OFFER packet

ibot3, 2020-04-15:
I tried it with the newest version (2020-04-09) an got this output after three DISCOVER/OFFER iterations:

```
Failed to open device: 'sdcard'
SD card not detected
SD boot failed
IHY
INK STATUS: speed: 1000 full duplex
GENET START: 64 1223 0x284600P: 0 IPV4: 0 MAC: 0 UDP: 0 UDP RECVX: 37 IP: 0 IPV4: 4 MAC: 2 UDP: 2 UDP RECV: 2 IP_CSUM_ERR: 0 UDP_CSUM_ERR: 0
RX: 37 IP: 0 IP_CSUM_ERR: 0 UDP_CSUM_ERR: 0
RX_DISCARD: 0
  GR64: 0x00000009   GR64: 0x00000014  GR127: 0x00000000  GR255: 0x00000008
 BCA: 0x00000000  GRFCS: 0x00000000
 GRXCF: 0x00000000  GRXPF: 0x00000000  GRXUO: 0x00000000  GRALN: 0x00000000
 GRFLR: 0x00000000  GRCDE: 0x00000000  GRFCR: 0x00000000  GROVR: 0x00JBR: 0x0000  GRPOK: 0x00000002   GP: 0x00000000  GRCRC: 0: 0x00000000  TR511: 0x00000003
TR1023: 0x00000000 TR1518: 0x00000000  TRMGV: 0x00000000 TR2047: 0x00000000
TR4095: 0x00000000 TR9216: 0x00000000  GTPKT: 0x00000003  GTMCA: 0x00000000
 GTBCA: 0x0000000XCF: 0x00000000  GTFCS: 0x00000000
 GTOVR: 0x00000000  POK: 0x00000T: 64 16 32
GENET: UMAC_START x28460000
RX: 37 IP: 0 IPV4: 4 MAC: 2 UDP: 2 UDP RECV: 2 IP_CSUM_ERR: 0 UDP_CSUM_ERR: 0
RUM_ERR: 0 UDP_CSUM_ERR: 0
RX: 72 IP: 0 IPV4: 10 MAC: 5 UDP: 5 UDP RECV: 5 IP_CSP_CSUM_ERR:4: 0x00000006   GR64: 0GR127: 0x00000000  GR255: 0x00000006
 GR511: 0x00000000 GR1023: 0x00000000 GR1518: 0x00000000  GRMGV: 0x00000000
GR2047: 0x4095: 0x00000000 GR9216: 0x00000023FCR: 0x00000000  GROVR: 0x000000003   GRUC: 0x00000000
00000  GTPKT: 0x000000000000000
 GTBCA: 0x00000003  GTXPF: 0x00000000  GTXCF: 0x00000000  GTFCS: 0x00000000
 GTOVR: 0x00000000  GTDRF: 0x00000000  GTEDF: 0x00000000  GTSCL: 0x00000000
 GTMCL: 0x00000000  GTLCL: 0x00000000  GTXCL: 0x00000000  GTFRG: 0x00000000
 GTNCL: 0x00000000  OT init failed
NET_BOOT: dc:a6:32:23:28:46 wait for link TFTP: 0.0.0uplex
GENET START: 64 ENET: UMAC_START 0xdca63223 0x28460000
RX: 72 IP: 0 IPV4: 10 MAC: 5 UDP: 5 UDP RECV: 5 IP_CSUM_ERR: 0 UDP_CSUM_ERR: 0
RX: 114 IP: 0 IPV4: 16 M8 IP_CSUM_ERR: 0 UDP_CSX: 114 IP: 8 UDP RECV: 8 IP_CSUM_ERR: 0 UDP_CSUM_ERR: 0
0 
  GR64: 0x0000000b   GR64: 0x00000016  GR127: 0x00000000  GR255: 0x00000009
 GR511: 0x00000000 GR1023: 0x00000000 GR1518: 0x00000000 0000000
GR2047: 0x00000000 GR40950000002a  GRPKT: 0x000018b0
 GRBYT: 0x0000001e  GRMCA: 0x00000009  GRBCA: 0x00000000  GRFCS: 0x00000000
 GRXCF: 0x00000000  GRXPF: 0x00000000  GRXUO: 0x00000000  GRALN: 0x00000000
 GRFLR: 0xCDE: 0x0000 0x00000000GRJBR: 0x00000000 GRMTUE: 0x0000002b  GRPOK: 0x00000003   GRUC: 0x00000000
 GRPPP: 0x00000000  GRCRC: 0x00000000

  TR64: 0x00000000 x00000003
TR1023: 0x00000000 TR1518: 0x0000 TR2047: 0x00000000
TR4095: 0x0000  GTPKT: 0x00000003  GTMCA: 0x000000
 GTOVR: 0x00000000 0000000  GTEDF: 0x00000000  GTSCL: 0x00000000
 GTMCL: 0x00000000  GTLCL: 0x00000000  GTXCL0  GTFRG: 0xR: 0x00000000  GTBYT: 0x00000450  000
GENET STOP: 0
NETBOOT CANCEL
NETBOOT init failed
NET_BOOT: dc:a6:32:23:28:46 wait for link TFTP: 0.0.0.0
LINK STATUS: speed: 1000 full duplex
GENET START: 64 16 32
GENET: UMAC_START 0xdca63223 0x28460000
RX: 114 IP: 0 IPV4: 16 MAC: 8 UDP: 8 UDP RECV: 8 IP_CSUM_ERR: 0 UDP_CSUM_ERR: 0
```

However, I don't really understand what's going on there :)

timg236, 2020-04-15:
The copy from the serial output is a little mangled but these are statistics from the software about the number of packets received for each layer

IPV4: 10 MAC: 5 UDP: 5 UDP RECV: 5 IP_CSUM_ERR: 0 UDP_CSUM_ERR: 0
e.g. 10 IPV4 packets with 5 UDP packets no checksum errors

The GR values are dumps from the ethernet registers.

It looks as though the link is up.

The absence of any DHCP logs suggests that either it didn't receive the packets, wrong address / port. 

I'd suggest using PiServer to validate with a known DHCP/TFTP server and comparing Wireshark logs https://hackaday.com/2019/11/11/network-booting-the-pi-4/

ibot3, 2020-04-15:
I corrected the output in my previous post to look better and more complete.

https://gist.github.com/ibot3/a4f168820bb956da1819770262f20b1a
This is the DHCP OFFER that the Pi is receiving on its link. AFAICS it is looking good (except that it is over a DHCP relay).
If I connect the Pi to my home router which does simple DHCP without any PXE, DHCP logs start to appear on the serial console.

The same setup is working when the Pi and the dnsmasq server are in the same L2 network.
Therefore the dnsmasq config can't be too wrong:
```
dhcp-range=set:local-mgmt-zw41,10.160.41.5,10.160.41.62,255.255.255.192,10.160.41.63,12h
dhcp-option=tag:local-mgmt-zw41,option:router,10.160.41.1
dhcp-host=10.160.41.1

dhcp-option=tag:user-pi,66,10.128.1.5

dhcp-host=dc:a6:32:23:28:46,set:user-pi,user-pi-zw41,10.160.41.20,12h

pxe-service=tag:user-pi,0,"Raspberry Pi Boot"
```

timg236, 2020-04-15:
Thanks, I can't see anything obviously wrong with that although it's not the raw binary and I can't tell if the checksums are correct, packets with incorrect checksums are discarded. The ports and mac-address look correct which would be things which could cause the packet to be discarded very early on.

It's not practical for me to duplicate network setups right not (that's also very time consuming) but if I get some spare time I could look at adding an EEPROM config to control the verbosity of network debug. This can be in a separate GPIO enabled section to allow very verbose debug e.g. dump all packets back from the netboot state machine.

I'm doing something similar for USB-MSD interop tests.

ibot3, 2020-04-15:
Thanks.
I updated the gist and enabled checksum validation and appended the raw hex at the end.

I also created a bash script for you which sets an complete test environment up using network namespaces. dnsmasq must be installed. For wireshark to diplay correct checksums, the optional commands with ethtool must be executed.

Just change the physical interface at the top to the interface where you want to connect the pi.
The only difference to my setup that I encountered is, that the dnsmasq dhcp relay evaluates the dhcp broadcast flag and is therefore sending the OFFER as a unicast. But this doesn't change anything on the current behavior.

https://gist.github.com/ibot3/2615bd5ae1164b32a0f7610b60755f3d

timg236, 2020-04-15:
Changed the title to indicate this bug is specifically about adding support for DHCP relay agents

timg236, 2020-04-15:
Thanks for the script, I'll add support for relay agents to the todo list after the MSD BOOT beta is released (hopefully quite soon)

ibot3, 2020-06-09:
@timg236 You added the "awaiting information" label. Do you need any further information or is this related to an internal process?

timg236, 2020-06-09:
As I understand it a DHCP relay should be completely invisible to the client device so the question is what's different about the packets. The difference may well be something which is valid but not handled by the Pi but nobody has had a chance to do the A-B comparison.

ibot3, 2020-06-11:
Yes, this image is working fine.

Thank you!

timg236, 2020-06-11:
Thanks @ibot3 I'll close this bug when the beta with this fix is released. We are currently trying to close out any blockers which would prevent the current USB MSD beta from going to stable and then production so this change should make it to critical/production reasonably soon.

timg236, 2020-06-16:
Resolved in pieeprom-2020-06-15

closed 2020-06-16 by timg236