raspberrypi/rpi-eeprom #60: PXE-Service order

skrenes, 2019-11-24:
I have the following `dnsmasq.conf` that network boots **both** x86 computers as well as Raspberry Pi 3B+'s. However, I recently switched to Raspberry Pi 4's with the latest firmware (pieeprom-2019-11-18.bin) and this fails.
```bash
port=0
dhcp-range=192.168.0.255,proxy
enable-tftp
tftp-root=/tftp
tftp-unique-root

# chainload to iPXE then run boot.ipxe script
dhcp-userclass=set:ipxe,iPXE
pxe-service=tag:!ipxe, x86PC, "Chainloading iPXE", ipxe.pxe
pxe-service=tag:ipxe, x86PC, "iPXE Script", boot.ipxe

# required to have the Raspberry Pi PXE boot
pxe-service=0,"Raspberry Pi Boot"

# x86 chooses first pxe-service option immediately; ignored by Raspberry Pi
pxe-prompt="PXE",0
```
If I move the "Raspberry Pi Boot" line, as in the following, the Raspberry Pi 4 boots, but the x86 machine will not (unless I remove the `pxe-prompt` line and manually select iPXE):

```bash
port=0
dhcp-range=192.168.0.255,proxy
enable-tftp
tftp-root=/tftp
tftp-unique-root

# required to have the Raspberry Pi PXE boot
pxe-service=0,"Raspberry Pi Boot"

# chainload to iPXE then run boot.ipxe script
dhcp-userclass=set:ipxe,iPXE
pxe-service=tag:!ipxe, x86PC, "Chainloading iPXE", ipxe.pxe
pxe-service=tag:ipxe, x86PC, "iPXE Script", boot.ipxe

# x86 chooses first pxe-service option immediately; ignored by Raspberry Pi
pxe-prompt="PXE",0
```

I'm currently trying to figure out if I can modify the code in such a way to get both the Raspberry Pi 4 and x86 computers to network boot automatically. But I thought I'd file a bug report since the behavior is different than Raspberry Pi 3B+.

I'm not sure which is correctly implemented, but the different behavior between the Pi 3B+ and Pi 4B merits a closer look. Thanks!

skrenes, 2019-11-24:
I now have both x86 computers and the Pi4's booting via the following `dnsmasq.conf`:
```bash
port=0
dhcp-range=192.168.0.255,proxy
enable-tftp
tftp-root=/tftp
tftp-unique-root

# required to have the Raspberry Pi 4 network boot
dhcp-mac=set:pi,DC:A6:32:*:*:*
pxe-service=tag:pi,0,"Raspberry Pi Boot"

# chainload to iPXE then run boot.ipxe script
dhcp-userclass=set:ipxe,iPXE
pxe-service=tag:!ipxe, x86PC, "Chainloading iPXE", ipxe.pxe
pxe-service=tag:ipxe, x86PC, "iPXE Script", boot.ipxe

# x86 chooses first pxe-service option immediately; ignored by Raspberry Pi
pxe-prompt="PXE",0
```
However, my original concern regarding the differences between the Pi 3B+ and 4B remains.

timg236, 2019-11-25:
Please can you post a wireshark capture?

skrenes, 2019-11-27:
I haven't used wireshark in years. Are you just wanting the DNS response? If you have docker setup somewhere (perhaps on a Pi ;-)) my `Dockerfile` is as follows
```
FROM alpine:edge
RUN apk --no-cache add dnsmasq
ENTRYPOINT ["dnsmasq", "-k"]
```

My `docker-compose.yml` file is as follows:
```yaml
version: '3'
services:
  dnsmasq:
    container_name: DNSmasq
    build:
      context: .
    command: --log-facility=-
    volumes:
      - /etc/localtime:/etc/localtime:ro
      - ./dnsmasq.conf:/etc/dnsmasq.conf:ro
      - ./tftp:/tftp:ro
    network_mode: host
    cap_add:
      - NET_ADMIN
    restart: always
```

timg236, 2019-12-04:
From code inspection I found a problem. Please could you try pieeprom-2019-12-03.bin from here
https://github.com/timg236/rpi-eeprom/tree/tim/stable_rc2/firmware/beta

skrenes, 2019-12-04:
Okay. I finally updated to 2019-12-03. I was trying to do it via a PXE booted RPi4 and it wouldn't take. After every command of `sudo rpi-eeprom-update -d -f` it would tell me to reboot. Afterwards though I'd see the following:
```bash
pi@raspberrypi:~ $ sudo rpi-eeprom-update
BOOTLOADER: up-to-date
CURRENT: Mon 18 Nov 11:06:55 UTC 2019 (1574075215)
 LATEST: Tue 10 Sep 10:41:50 UTC 2019 (1568112110)
VL805: up-to-date
CURRENT: 000137ab
 LATEST: 000137ab
```

I disabled DNSmasq docker, put in an SD card, and rebooted, then did it over again, and this time it updated. Any idea how I can update the EEPROM without an SD card?

As for the update itself, I reverted the `dnsmasq.conf` file back to:
```bash
port=0
dhcp-range=192.168.0.255,proxy
enable-tftp
tftp-root=/tftp
tftp-unique-root

# chainload to iPXE then run boot.ipxe script
dhcp-userclass=set:ipxe,iPXE
pxe-service=tag:!ipxe, x86PC, "Chainloading iPXE", ipxe.pxe
pxe-service=tag:ipxe, x86PC, "iPXE Script", boot.ipxe

# required to have the Raspberry Pi PXE boot
pxe-service=0,"Raspberry Pi Boot"

# x86 chooses first pxe-service option immediately; ignored by Raspberry Pi
pxe-prompt="PXE",0
```

There was no noticeable difference between 2019-11-18 and 2019-12-03. The Pi4 will still not boot. The logs from dnsmasq just keep saying `dnsmasq-dhcp[1]: PXE(eno1) dc:a6:32:[REDACTED] proxy` for about 10 times before it stops. Fortunately, it still works with the following `dnsmasq.conf` file:
```bash
port=0
dhcp-range=192.168.0.255,proxy
enable-tftp
tftp-root=/tftp
tftp-unique-root

# required to have the Raspberry Pi 4 network boot
dhcp-mac=set:pi,DC:A6:32:*:*:*
pxe-service=tag:pi,0,"Raspberry Pi Boot"

# chainload to iPXE then run boot.ipxe script
dhcp-userclass=set:ipxe,iPXE
pxe-service=tag:!ipxe, x86PC, "Chainloading iPXE", ipxe.pxe
pxe-service=tag:ipxe, x86PC, "iPXE Script", boot.ipxe

# x86 chooses first pxe-service option immediately; ignored by Raspberry Pi
pxe-prompt="PXE",0
```

With that `dnsmasq.conf` file, the output from dnsmasq is instead:
```bash
dnsmasq-dhcp[1]: PXE(eno1) dc:a6:32:[REDACTED] proxy
dnsmasq-tftp[1]: file /tftp/192.168.0.124/[REDACTED]/start4.elf not found
dnsmasq-tftp[1]: file /tftp/192.168.0.124/[REDACTED]/start.elf not found
dnsmasq-tftp[1]: sent /tftp/192.168.0.124/config.txt to 192.168.0.124
dnsmasq-tftp[1]: file /tftp/192.168.0.124/recover4.elf not found
dnsmasq-tftp[1]: file /tftp/192.168.0.124/recovery.elf not found
dnsmasq-tftp[1]: sent /tftp/192.168.0.124/start4.elf to 192.168.0.124
dnsmasq-tftp[1]: sent /tftp/192.168.0.124/fixup4.dat to 192.168.0.124
dnsmasq-tftp[1]: file /tftp/192.168.0.124/recovery.elf not found
dnsmasq-tftp[1]: sent /tftp/192.168.0.124/config.txt to 192.168.0.124
dnsmasq-tftp[1]: file /tftp/192.168.0.124/dt-blob.bin not found
dnsmasq-tftp[1]: file /tftp/192.168.0.124/recovery.elf not found
dnsmasq-tftp[1]: sent /tftp/192.168.0.124/config.txt to 192.168.0.124
dnsmasq-tftp[1]: file /tftp/192.168.0.124/bootcfg.txt not found
dnsmasq-tftp[1]: sent /tftp/192.168.0.124/cmdline.txt to 192.168.0.124
dnsmasq-tftp[1]: sent /tftp/192.168.0.124/bcm2711-rpi-4-b.dtb to 192.168.0.124
dnsmasq-tftp[1]: sent /tftp/192.168.0.124/config.txt to 192.168.0.124
dnsmasq-tftp[1]: sent /tftp/192.168.0.124/overlays/vc4-fkms-v3d.dtbo to 192.168.0.124
dnsmasq-tftp[1]: file /tftp/192.168.0.124/recovery8.img not found
dnsmasq-tftp[1]: file /tftp/192.168.0.124/recovery8-32.img not found
dnsmasq-tftp[1]: file /tftp/192.168.0.124/recovery7l.img not found
dnsmasq-tftp[1]: file /tftp/192.168.0.124/recovery7.img not found
dnsmasq-tftp[1]: file /tftp/192.168.0.124/recovery.img not found
dnsmasq-tftp[1]: error 0 Early terminate received from 192.168.0.124
dnsmasq-tftp[1]: failed sending /tftp/192.168.0.124/kernel8.img to 192.168.0.124
dnsmasq-tftp[1]: file /tftp/192.168.0.124/kernel8-32.img not found
dnsmasq-tftp[1]: error 0 Early terminate received from 192.168.0.124
dnsmasq-tftp[1]: failed sending /tftp/192.168.0.124/kernel7l.img to 192.168.0.124
dnsmasq-tftp[1]: file /tftp/192.168.0.124/armstub8-32-gic.bin not found
dnsmasq-tftp[1]: sent /tftp/192.168.0.124/kernel7l.img to 192.168.0.124
```

timg236, 2019-12-05:
"""
I disabled DNSmasq docker, put in an SD card, and rebooted, then did it over again, and this time it updated. Any idea how I can update the EEPROM without an SD card?
"""
You can select USE_FLASHROM=1 in /etc/default/rpi-eeprom-update but it's only safe to do this from console mode because the the SPI CS is muxed with audio and it may hang if the desktop is running and ALSA is holding the audio

lurch, 2019-12-05:
@timg236 Is that worth documenting anywhere? Or is it too much of a niche use-case to worry about?

timg236, 2019-12-05:
It's documented in the script but I wouldn't encourage people to use it because it will cause confusing hang issues. USB device boot might be a better alternative or simply getting the bootloader to update itself (just don't disconnect the power) but that's more like Q1 2020 stuff

skrenes, 2019-12-06:
Thanks! Five of my six Pi4's are PXE and headless. So being able to update the EEPROM remotely is a huge benefit.

timg236, 2020-01-09:
This should be fixed in the 2020-01-09 release. 
Please could you give this a try ?
https://github.com/timg236/rpi-eeprom/tree/rpiboot-2020-01-19/firmware/beta

timg236, 2020-01-15:
I believe this is fixed. Closing bug - please re-open if this is not the case.

closed 2020-01-15 by timg236