raspberrypi/rpi-eeprom #62: Easy way to query the version information inside pieeprom

dividuum, 2019-12-01:
Would it be possible to include the version string returned by `vcgencmd bootloader_version` in an easy to query way inside the pieeprom.bin file itself? My naive approach would to be include another file (version.txt?) next to the already existing `bootconf.txt`. That way the mechanism in rpi-eeprom-config could be used. Or maybe another VERSION_MAGIC that's otherwise unused by the firmware itself?

This change would make automatically processing firmware file a bit easier.

timg236, 2019-12-01:
It's already included.

The git short hash and timestamp identify non-modifiable part of the pieeprom.bin image. 
rpi-eeprom-update uses strings to get the BUILD_TIMESTAMP.

strings pieeprom.bin | egrep "VERSION:|BUILD_TIMESTAMP"

dividuum, 2019-12-01:
Good to know. I was searching for a way to get the long hex string mentioned in the bootloader_version output and completely ignored the timestamp value. Thanks.

closed 2019-12-01 by dividuum

timg236, 2019-12-01:
No problem. If you are writing your own scripts then watch out for the initial launch firmware where the time stamp was not returned by the gencmd. There’s a workaround in the rpi-eeprom-update script

dividuum, 2019-12-01:
In case anyone else is interested in that workaround:

https://github.com/raspberrypi/rpi-eeprom/blob/a84bd5c5223b7bdd1cbdea3806e15fed6e099973/rpi-eeprom-update#L195-L202)