raspberrypi/rpi-eeprom #64: Invalid/missing signature

mecko, 2019-12-08:
Raspberry Pi 4B (2GB). After failing to boot a fresh setup of the latest Raspbian, I followed the recommendation to make sure the EEPROM is in tact.

I took an sdcard, formatted with FAT32, and simply put the content of the .zip file from raspberrypi.org in it. I then put it in the Pi, plugged in the power and presented with 2 long flashes and then 4 short ones, repeatedly, which is apparently an undocumented error sequence.

Having no micro HDMI cable, I connected the serial pins to UART. The error in the console is:
```
Loading vl805.sig hnd: 0x00000031
Loading vl805.bin hnd: 0x0000002a
SHA256
00000000  12 67 75 0e 2e e8 49 b3  44 8a 93 fa 7b f2 92 90      | .gu...I. D...{...|
00000010  75 1c c8 98 a5 4d 76 ab  0d 6e c7 a4 7a c1 89 60      | u....Mv. .n..z..`|
File has Invalid/missing signature: 'vl805.sig' (65)
FATAL @ 0x8000654a
```

I tried to remove vl805.bin and vol805.sig completely, which brought me to the same error but in regards to pieeprom.sig/bin.

I also tried to replace the recovery.bin, vl805.bin and pieeeprom.bin files with more recent ones from this repo (and update the .sig files accordingly).

The voltage of the Pi is rock steady at 5.17vdc (exactly, constantly, for at least 15 seconds). I plugged the sdcard into a different pc with a different card reader, and checked the SHA256 of the files - they match the checksums inside the .sig files.

mecko, 2019-12-12:
After thorough fishing for more details, I can confirm that it detects the correct SHA256 for files that are 32 bytes or smaller, and incorrect SHA256 for files that are larger than 32 bytes.
I believe only somebody like @timg236 or so with access to the sources can be helpful here.

cleverca22, 2019-12-12:
feels like there is a 32 byte buffer somewhere, and the hash check fails for any file over 32 bytes in size, making it imposible to correctly compute the hash for real firmware files

timg236, 2019-12-12:
It's clearly not a software issue with the hash check because if it was it would fail on every board every time. 

This image is https://static.raspberrypi.org/files/rpi-boot-eeprom-recovery-2019-09-10-vl805-000137ab.zip is used on 1000s of boards if not more.

"I tried the procedure with 4 different sdcards, 1 of which is brand new and 1 is 100% working."

You mentioned that it fails on old sd-cards but not on a new one so that should give a clue that it's an issue with the sd-card . You can try using the latest recovery.bin from here 
https://github.com/raspberrypi/rpi-eeprom/blob/master/firmware/beta/recovery.bin

We've yet to find a SD high capacity card where this doesn't work except where the sd-card reader has been damaged.

Closing bug. If you find brands of sd-cards that are unreliable on Raspberry Pi then please raise firwmare bugs with full details of the cards and we can try to get samples to test.

closed 2019-12-12 by timg236

mecko, 2019-12-12:
@timg236 You got me completely wrong here.
I never mentioned that it "fails on old sd-cards" and I never said "not on a new one". I get the same problem with *all* sdcards that I tried - old and new.

I tried the process with a SanDisk 16GB, a Kingston 16GB and a Transcend 16GB. The SanDisk and Transcend are brand new, the Kingston has been running a Pi 3 for some time now. I also tried to work with an old Sony 64GB and an old SanDisk 16GB (but forget about them for now).

This is either a software or a hardware problem. It's clearly not an issue with the sdcards and not a compatibility or reliability problem. I'm reaching out for assistance in figuring out what's causing this.

Please consider reopening this issue as this is clearly an issue with either the software at this repo or a larger scale issue with the Pi 4B.

cleverca22, 2019-12-12:
@timg236 what about the fact that a 32 byte file containing plaintext, fails to match the `.sig` (intentionally) with its real sha256

but a 33 byte plain-text file, fails with a completely different sha256, that doesnt match its contents?

timg236, 2019-12-12:
Files which are the wrong size for the EEPROMs will be rejected. It isn't a general purpose hash generator

cleverca22, 2019-12-12:
but, we are only using that to verify why the hash function fails

for any file over 32 bytes in size (including the proper .bin files) the hash comes out completely wrong

for any file 32 bytes and under, the hash comes out correctly

cleverca22, 2019-12-12:
we have also confirmed the hashes are correct if reading it back from a pc, so the card isnt corrupting data

mecko, 2019-12-12:
### **32 bytes** file
#### hexdump -C vl805.bin
```
00000000  30 30 30 30 30 30 30 30  30 30 31 31 31 31 31 31  |0000000000111111|
00000010  31 31 31 31 32 32 32 32  32 32 32 32 32 32 3d 3d  |11112222222222==|
00000020
```
#### sha256sum vl805.bin
```b48b0337e47b6e577b1f05d21e09597b8a47f09b83b522220120fa1e5cc77fc5  vl805.bin```
#### UART
```
Loading vl805.sig hnd: 0x0000005d
Loading vl805.bin hnd: 0x00000050
SHA256
00000000  b4 8b 03 37 e4 7b 6e 57  7b 1f 05 d2 1e 09 59 7b      | ...7.{nW {.....Y{|
00000010  8a 47 f0 9b 83 b5 22 22  01 20 fa 1e 5c c7 7f c5      | .G...."" . ..\...|
File has Invalid/missing signature: 'vl805.sig' (65)
FATAL @ 0x8000654a
```

### **35 bytes** file
#### hexdump -C vl805.bin
```
00000000  30 30 30 30 30 30 30 30  30 30 31 31 31 31 31 31  |0000000000111111|
00000010  31 31 31 31 32 32 32 32  32 32 32 32 32 32 33 33  |1111222222222233|
00000020  33 33 33                                          |333|
00000023
```
#### sha256sum vl805.bin
```f6bdbad10f7c0226ac979a711d9921229aa6f15906d4574ba258eca1096a18f7  vl805.bin```
#### UART
```
Loading vl805.sig hnd: 0x0000005d
Loading vl805.bin hnd: 0x00000050
SHA256
00000000  a0 03 00 09 fb 26 3b 9e  6c 7a 2c 1b b2 7c c9 a0      | .....&;. lz,..|..|
00000010  2a e8 12 26 ab 59 73 d8  62 3a ca 58 f3 2b 1a 13      | *..&.Ys. b:.X.+..|
File has Invalid/missing signature: 'vl805.sig' (65)
FATAL @ 0x8000654a
```

cleverca22, 2019-12-12:
@timg236 why would it be reporting the wrong hash for a 35 byte file, but is perfectly fine at reporting the right hash for a 32 byte file?

it also reports the wrong hash for the original `vl805.bin`

timg236, 2019-12-12:
@cleverca22 You're attempting to debug the wrong thing. Clearly the hash generator does work because @mecko has stated that it works on some sd-cards and not others. This tool is routinely used by large numbers of people so there is obviously something different in that setup.

Clearly there is something different about the setup for @mecko 

Random experiment. Download the rescue image off the downloads page on a SANDISK 16GB SD card formatted as FAT32 using Windows.

1) Works for me.

2) Let's break it 

cp README.txt vl805.bin
Success: wrote 524288 bytes
Loading vl805.sig hnd: 0x000000c9
Loading vl805.bin hnd: 0x000000bc
SHA256
00000000  a0 b1 dd 6c 36 6a c6 a3  14 7d 99 41 fa ae 3b db      | ...l6j.. .}.A..;.|
00000010  49 30 1a 32 41 6d d4 c6  15 b4 a1 6f 74 b0 d5 2a      | I0.2Am.. ...ot..*|
File has Invalid/missing signature: 'vl805.sig' (65)
FATAL @ 0x8000654a

sha256sum.exe README.txt
a0b1dd6c366ac6a3147d9941faae3bdb49301a32416dd4c615b4a16f74b0d52a *README.txt

vl805 is now rejected because the signature is wrong but the computer signature matches the SHA256

sha256sum.exe vl805.bin  > vl805.sig

cleverca22, 2019-12-12:
yeah, it doesnt really make sense why its coming up with a different hash for files over 32 bytes in size

id need access to the source of recovery.bin to confirm it, otherwise i'm just guessing blindly

is it reading the sd card in blocks of 32 bytes or 512 bytes (sectors)
does recovery.bin bring the dram online first? the .bin files are too big to fit into the L2 cache
could the dram be faulty on this unit?, though 32 bytes seems too low to trigger that

timg236, 2019-12-12:
The normal suspects are 

1) The sd-card ... but they all work in Windows so let's assume they are good for now.
2) The sd-card reader on the Pi4. If you have another Pi4 then I suggest trying on that. It's quite possible to damage the sd-card reader in such a way that it will fail on some cards and work on others. Normal problem is damage to the underside of the board.
3) A file-system bug in recovery.bin.  Reformat the entire disk as FAT32 on Windows i.e. simplest possible setup. If you have a .img which fails on every card then raise a FW bug.
I wouldn't rule this out but it's tested on every cluster size etc, if the root dir is full of files and has loads of churn then that could expose a bug which is why we encourage people to use a cleanly formatted card.
4) An SD driver bug in recovery.bin, try the latest recovery.bin from the beta-tree. There were some fixes for EMMC + ancient SDv1 cards but all the ones listed are modernish SDHC.

We've yet to see an sd card driver bug that prevents recovery.bin from working with a modern SDHC card. I wouldn't completely rule it out but generally those sorts of bugs just cause the code to wait forever for data.

cleverca22, 2019-12-12:
wait a moment, something isnt right here...

@timg236 
```
Success: wrote 524288 bytes
Loading vl805.sig hnd: 0x000000c9
Loading vl805.bin hnd: 0x000000bc
SHA256
00000000 a0 b1 dd 6c 36 6a c6 a3 14 7d 99 41 fa ae 3b db | ...l6j.. .}.A..;.|
00000010 49 30 1a 32 41 6d d4 c6 15 b4 a1 6f 74 b0 d5 2a | I0.2Am.. ...ot..*|
File has Invalid/missing signature: 'vl805.sig' (65)
FATAL @ 0x8000654a
```
vs:
@mecko 
```
Loading vl805.sig hnd: 0x0000005d
Loading vl805.bin hnd: 0x00000050
SHA256
00000000  a0 03 00 09 fb 26 3b 9e  6c 7a 2c 1b b2 7c c9 a0      | .....&;. lz,..|..|
00000010  2a e8 12 26 ab 59 73 d8  62 3a ca 58 f3 2b 1a 13      | *..&.Ys. b:.X.+..|
File has Invalid/missing signature: 'vl805.sig' (65)
FATAL @ 0x8000654a
```

why did @timg236 's log start by writing the `pieeprom.bin` then fail over `vl805.bin`
why does @mecko 's log fail with `vl805.bin` first, and only complain about `pieeprom.bin` when `vl805.bin` is deleted?

why does it process them in a different order?

cleverca22, 2019-12-12:
@timg236 oh, what about the different modes used to access the SD card?

there are spi, 1bit, 4bit, and 8bit? modes

which one does the boot rom use? which one does `recovery.bin` use?

perhaps one of the pins is damaged, and it can only work in 1bit mode?

mecko, 2019-12-12:
@timg236 
> @cleverca22 You're attempting to debug the wrong thing. Clearly the hash generator does work because @mecko has stated that it works on some sd-cards and not others. This tool is routinely used by large numbers of people so there is obviously something different in that setup.

I didn't say it works on some sdcards and doesn't on others. It works on files smaller than 32 bytes but doesn't on files larger than that.

timg236, 2019-12-12:
"why did @timg236 's log start by writing the pieeprom.bin then fail over vl805.bin" 

The log wrapped :) That's from the previous boot

cleverca22, 2019-12-12:
ah, so that just leaves the question of SD card modes and potential damage to a pin in the SD slot, that only allows certain modes to work

timg236, 2019-12-12:
" I tried the procedure with 4 different sdcards, 1 of which is brand new and 1 is 100% working."

I misinterpreted 100% working. 

Assuming that the file-system isn't weird then unfortunately, I think it's a hardware failure on that board. 32 bytes could well be the size fifo inside the controller.

cleverca22, 2019-12-12:
hmmm, yeah, now that i think of it, 32 bytes is within a single sector, so if your reading a sector at a time, it doesnt matter if its 1byte or 512 bytes, it will read the same way over the SD bus

reviewing how the SD card works in the rpi...

mecko, 2019-12-12:
> Assuming that the file-system isn't weird then unfortunately, I think it's a hardware failure on that board. 32 bytes could well be the size fifo inside the controller.

Assuming the problem is with the board's sdcard reader, can I somehow flash the EEPROM not using the sdcard? If that'd work, it must help isolate the problem.
If it helps, I tried to save the rescue .zip as a recovery/, rename recovery.bin to bootcode.bin, and use https://github.com/raspberrypi/usbboot/ to boot into it, but it didn't work too well (green led blinked 4 times repeatedly).

cleverca22, 2019-12-12:
the SD controller is missing from https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2835/BCM2835-ARM-Peripherals.pdf and ive not found any more recent pdf's for other models

timg236, 2019-12-12:
The SD controller is same EMMC as used for WiFi SDIO on Pi3 but that's unlikely to be the issue since it's part of the SOC. Most likely it's an electrical problem somewhere between the SOC and the card or SOC io pads (resistors, voltage switch etc). Annoying sd-cards getting the wrong voltage / current will stagger on for quite a while before failing  e.g. you have to read enough data with the to fail.

If you get the repeated 4 long flashes without an SDCARD then the bootloader EEPROM is running so there's no need to update it. You should be able to put a Raspbian SDCARD in, boot that and if necessary use /usr/bin/vl805 to update the USB EEPROM directly.

If you get 4 long flashes repeated with Raspbian image with no recovery.bin in /boot then it can't read start.elf. Add uart_2ndstage=1 in config.txt for UART output. That would probably confirm an electrical failure.

USB device boot isn't available yet, recovery.bin might support it early next year but more work is required in order to get it to load the images and program them from USB device boot mode.

cleverca22, 2019-12-12:
with no usb or sd card present, the pi gives a single flash of the status led on power up, and thats it, nothing further

timg236, 2019-12-12:
ok, that means that the EEPROM is blank.

Have you tried the beta release of recovery.bin ?

cleverca22, 2019-12-12:
theres a beta release of it?

as far as i know, @mecko has only been using the copy from https://static.raspberrypi.org/files/rpi-boot-eeprom-recovery-2019-09-10-vl805-000137ab.zip

```
2019-12-11 07:12:45<clever> whats the hash of recovery.bin itself?
2019-12-11 07:13:03<clever> $ sha256sum recovery.bin 
2019-12-11 07:13:03<clever> 92d5d2d12002ace3e4ca9374219c86f9d1fee3b7a0d4c4b0d5dbbad3b3d06eb0  recovery.bin
2019-12-11 07:13:34<dakar> 92d5d2d12002ace3e4ca9374219c86f9d1fee3b7a0d4c4b0d5dbbad3b3d06eb0 *recovery.bin
```
and we have confirmed that the copy on his SD card perfectly matches the copy i have

mecko, 2019-12-12:
> Have you tried the beta release of recovery.bin ?

I also tried the most recent one I could find on this repo (firmware/beta/recovery.bin) at some stage.

cleverca22, 2019-12-12:
i think it would help some to include the `recovery.bin` in this github repo, so people can see when it has updated

cleverca22, 2019-12-12:
oh, its already in the repo?, nvm then, hadnt noticed

timg236, 2019-12-12:
Yes, look in the beta directory. If I get chance tomorrow I'll upload a hacked recovery.bin with 1-bit / slow card clock forced by default. 
No chance of it being included in the main Downloads .zip file any time soon (i.e. months not weeks), but if it allows someone to migrate a Pi with problematic SD to say network boot then that might be useful.

mecko, 2019-12-12:
@timg236 

I tried a very old recovery.bin (before hash check was implemented). It looks like it works though it doesn't act like it. Still one short flash without sdcard as well as with Raspbian. There's no output to UART whatsoever.

I hope this helps.

The output in UART is:
```
Loading pieeprom.bin hnd: 0x00000006
Reading EEPROM
103578 bytes differ
Writing EEPROM
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++.....................................................................
Verify
Reading EEPROM
0 bytes differ
Success: wrote 524288 bytes
```

timg236, 2019-12-12:
Might be worth running it twice. The second attempt should show all dots if it updated the bootloader EEPROM correctly. If it changes every time then it's getting random data.

Assuming it stabilizes the next step is enable BOOT_UART=1 in pieeprom.bin and enable_uart=1 in config.txt and see how far start.elf gets

cleverca22, 2019-12-12:
yep, we are already on to using https://github.com/raspberrypi/rpi-eeprom/blob/master/rpi-eeprom-config to update the config, then the old recovery.bin to flash it

cleverca22, 2019-12-12:
https://gist.github.com/cleverca22/fa08a2c303a54c1d70871229ca246bd0
i used these commands to create a new `.bin` file, and confirmed a minimal change

this includes `BOOT_UART=1` in the `bootconf.txt`

the first run of `recovery.bin` reported `++++...` and the second run all dots

but still nothing from the uart

cleverca22, 2019-12-12:
@timg236 how many bytes are covered for each `+`, and is that the erase block size for the SPI chip?

timg236, 2019-12-12:
A dot is 4K which is the erase block size.

As a last ditch attempt I've uploaded a version of recovery.bin which forces 1-bit mode. USB device boot (*) is enabled in that recovery.bin but you'll need to rebuild the rpiboot server.
https://github.com/timg236/usbboot/tree/tim/2711_file_server
https://github.com/timg236/rpi-eeprom/tree/rpiboot-1bit/firmware/beta/recovery-1bit.bin

(*) It may not work and rpiboot for actual booting won't be available soon but with luck it might help you migrate to a network boot.

cleverca22, 2019-12-12:
```
2019-12-12 15:57:29 <dakar> FILE_EXISTS? vl805.bin mode: 0
2019-12-12 15:57:30 <dakar> FILE_GET: vl805.bin (0x11) mode: 0 1
2019-12-12 15:57:30 <dakar> vl805.sig
2019-12-12 15:57:30 <dakar> 00000000  7e 48 30 31 8e 8c ca 2a  1a da c2 6d e5 b1 f7 1b      | ~H01...* ...m....|
2019-12-12 15:57:31 <dakar> 00000010  d3 b2 95 52 15 88 b0 88  3b fa a1 62 ce 70 05 ba      | ...R.... ;..b.p..|
2019-12-12 15:57:32 <dakar> FILE_GET: vl805.bin (0x53) mode: 0 2
2019-12-12 15:57:33 <dakar> SHA256
2019-12-12 15:57:34 <dakar> 00000000  ab 30 1d 84 9b 36 1f e3  ca 85 53 83 9e 2d 92 d2      | .0...6.. ..S..-..|
2019-12-12 15:57:35 <dakar> 00000010  a9 6f 07 22 54 24 e5 7c  83 d4 14 98 62 21 fc 2c      | .o."T$.| ....b!.,|
2019-12-12 15:57:36 <dakar> File has Invalid/missing signature: 'vl805.bin' (0)
2019-12-12 15:57:37 <dakar> FATAL @ 0x80006b96 0x00000006 0x80003bc4
2019-12-12 15:57:38 <dakar> i used his new recovery, with the "original" .bins and .sigs
```

looks like the hash still fails with the SD card, @mecko will be able to test over usb later