raspberrypi/rpi-eeprom #68: Cannot PXE boot from a TFTP directory

rgl, 2019-12-14:
I'm trying to make the pi PXE boot from a TFTP directory as described at [Raspberry Pi4 network boot (BETA)](https://github.com/raspberrypi/rpi-eeprom/blob/master/firmware/raspberry_pi4_network_boot_beta.md), but using `TFTP_PREFIX_STR=rpi1/` (or even using `rpi1/yadayada`) makes the firmware load `rpi1start4.elf` instead of `rpi1/start4.elf` (or `rpi1/yadayadastart4.elf`). It seems there's a bug somewhere that makes is only use the basename of the prefix, which is a real bummer.

I'm using the `/lib/firmware/raspberrypi/bootloader/beta/pieeprom-2019-11-18.bin` bootloader.

BTW, where is the source-code for the bootloader code that handles `TFTP_PREFIX_STR`?

These are my changes to stock `bootconf.txt` file:

```patch
--- bootconf.txt.orig	2019-12-14 18:56:25.007004012 +0000
+++ bootconf.txt	2019-12-14 18:56:25.207005728 +0000
@@ -1,15 +1,17 @@
 [all]
-BOOT_UART=0
+BOOT_UART=1
 WAKE_ON_GPIO=1
 POWER_OFF_ON_HALT=0
 DHCP_TIMEOUT=45000
 DHCP_REQ_TIMEOUT=4000
 TFTP_FILE_TIMEOUT=30000
 TFTP_IP=
-TFTP_PREFIX=0
-BOOT_ORDER=0x1
+TFTP_PREFIX=1
+BOOT_ORDER=0x2
 SD_BOOT_MAX_RETRIES=3
-NET_BOOT_MAX_RETRIES=5
+NET_BOOT_MAX_RETRIES=-1
+TFTP_PREFIX_STR=rpi1/
+
 [none]
 FREEZE_VERSION=0
```

timg236, 2019-12-14:
Very odd, I use a similar config and that EEPROM version works for me. The forward slash has no significance in config format and strings are assembled using simple string concatenation.

Does 'vcgencmd bootloader_config' show the expected output ? Maybe some unicode in there ?

With UART logging I see

ARP TX req: sender: dc:a6:32:00:15:96 192.168.1.157 target: 192.168.1.75
DHCP complete: 192.168.1.157 dc:a6:32:00:15:96 subnet 255.255.255.0 gateway 0.0.0.0 TFTP 192.168.1.75 c8:2a:14:15:c5:13
RX: 19 IP: 0 IPV4: 20 MAC: 10 UDP: 10 UDP RECV: 2 IP_CSUM_ERR: 0 UDP_CSUM_ERR: 0
TFTP_GET: c8:2a:14:15:c5:13 192.168.1.75 rpi1/start4.elf

and Wireshark shows
Frame 110: 87 bytes on wire (696 bits), 87 bytes captured (696 bits) on interface 0
Ethernet II, Src: dc:a6:32:00:15:96, Dst: c8:2a:14:15:c5:13
Internet Protocol Version 4, Src: 192.168.1.157, Dst: 192.168.1.75
User Datagram Protocol, Src Port: 21479, Dst Port: 69
Trivial File Transfer Protocol
    Opcode: Read Request (1)
    Source File: rpi1/start4.elf
    Type: octet
    Option: tsize = 0
        Option name: tsize
        Option value: 0
    Option: blksize = 1024
        Option name: blksize
        Option value: 1024

rgl, 2019-12-15:
Oh `vcgencmd bootloader_config` did not have the correct value... so none of the changes I was making were being applied at all.

I forgot to mention, but at some point, after network booting was working, I had removed the sd-card to see if it still worked, and indeed it did. Then, I tried to change that bootloader setting and it never worked again.

It turns out that the update mechanism only works from the sd-card, which is somewhat confusing, as the other files can be read from the network. So, to make this work, I had to rsync my `/boot` directory (which comes from NFS) to the boot partition on the sd-card.

Why can't everything be done without an sd-card?

Is/will the bootloader source-code be available?

timg236, 2019-12-15:
There’s no support to failover to the previous image in the event of a power failure risking corruption. You can update directly if USE_FLASHROM is set. See rpi-eeprom-update —help

A self update for network boot is a possibility in future eg if users are happy to risk power failure or have a UPS. However except for critical bugs or manufacturing support the highest bootloader priority right now is usb msd support and theres quite a lot of work still to do there

rgl, 2019-12-15:
How can USE_FLASHROM end up with a corrupt image in the flash (when interrupted) but when using recovery.bin it cannot? Aren't the two writing directly to the flash (albeit at different times)? Can't recovery.bin also be interrupted?

timg236, 2019-12-15:
recovery.bin is always loaded by the ROM if it exists on the SD card . After a successful update recovery.bin renames it self. If a power failure occurs it won’t have been renamed and it will complete the update.

rgl, 2019-12-15:
Got it now! Thank you!

closed 2019-12-15 by rgl

lurch, 2019-12-16:
@timg236 Seems like a couple of people have been caught out by rpi-eeprom-update not working as expected on a network-booted Pi - maybe it's worth adding a check and printing a warning message if `/boot` isn't mounted from the local SD-card? :man_shrugging:

rgl, 2019-12-16:
That warning is a welcomed addition. I would have noticed it.

The firmware that handles the loading of recovery.bin is burnt into the "SoC" ROM? Is it really a ROM or there is way to replace it?

timg236, 2019-12-16:
It really is a ROM.

A check which examines the resolved BOOTFS to see if is on mmcblk0 (might not be p1) and warns if not would be reasonable, but rpi-eeprom-update tweaks are fairly low priority right now. Of course if someone wants to write a PR it might happen sooner