raspberrypi/rpi-eeprom #69: Add support for PXE (67) Bootfile name

rgl, 2019-12-14:
The `/lib/firmware/raspberrypi/bootloader/beta/pieeprom-2019-11-18.bin` firmware requests the PXE `(67) Bootfile name` parameter, but the bootloader seems completely ignore it. At least when I return the filename as `rpi1/start4.elf`. Isn't this a bug?

Also, the PXE requests some unknown/vendor PXE options, as displayed in Wireshark:

```
#         Option: (55) Parameter Request List
#             Length: 14
#             Parameter Request List Item: (1) Subnet Mask
#             Parameter Request List Item: (3) Router
#             Parameter Request List Item: (43) Vendor-Specific Information
#             Parameter Request List Item: (60) Vendor class identifier
#             Parameter Request List Item: (66) TFTP Server Name
#             Parameter Request List Item: (67) Bootfile name
#             Parameter Request List Item: (128) DOCSIS full security server IP [TODO]
#             Parameter Request List Item: (129) PXE - undefined (vendor specific)
#             Parameter Request List Item: (130) PXE - undefined (vendor specific)
#             Parameter Request List Item: (131) PXE - undefined (vendor specific)
#             Parameter Request List Item: (132) PXE - undefined (vendor specific)
#             Parameter Request List Item: (133) PXE - undefined (vendor specific)
#             Parameter Request List Item: (134) PXE - undefined (vendor specific)
#             Parameter Request List Item: (135) PXE - undefined (vendor specific)
```

Where is documentation about it?

timg236, 2019-12-14:
Pi4 requests the same options as Pi3 but bootfile is no longer relevant because there is no boot file to load because the second-stage is loaded from the EEPROM

The GPU firmware loading pattern is exactly the same for all SD/USB/Network and we are very unlikely to change that and add special cases for network boot.

https://www.raspberrypi.org/documentation/configuration/boot_folder.md

rgl, 2019-12-15:
OK, I understand. Is the source-code somewhere so others can change it?

The documentation link you've provided does not seem to document the `(129 to 135) PXE - undefined (vendor specific)` request parameters. What are those used for?

timg236, 2019-12-15:
The firmware is not open source. There’s various posts on the forums about it

rgl, 2019-12-15:
Okay, thank you.

closed 2019-12-15 by rgl

timg236, 2019-12-15:
Can’t remember the history of why it requests those extensions. It’s probably  unnecessary and could maybe be tidied up in future