raspberrypi/rpi-eeprom #72: rpi-eeprom adds garbage to IP address of TFTP server

VladVons, 2019-12-22:
pieeprom-2019-12-03.bin has a problem while network booting to TFTP server.
if TFTP server's IP has lass than 3 digit  pieeprom pads value to 3 digit with a garbage.

/etc/dhcp/dhcpd.conf
option tftp-server-name "192.168.0.21";

tcpdump gets following string
00:12:47.464387 ARP, Ethernet (len 6), IPv4 (len 4), Request who-has 192.168.2.214 tell ...

our 192.168.0.21 becomes 192.168.2.214 and nothong comes to our real TFTP server

in case we set 3 digit it works good
option tftp-server-name "192.168.0.211";

timg236, 2019-12-23:
Might be a regression in the parsing code in that release. If you can specify the extension directly then appending a space might fix it. Should be trivial to fix, will take a look in the New Year

timg236, 2020-01-09:
This should be fixed in the 2020-01-09 release. 
Please could you give this a try ?
https://github.com/timg236/rpi-eeprom/tree/rpiboot-2020-01-19/firmware/beta

timg236, 2020-01-15:
I believe this is fixed. Closing bug - please re-open if this is not the case.

closed 2020-01-15 by timg236