raspberrypi/rpi-eeprom #73: recovery.bin uart output

cleverca22, 2019-12-22:
i'm attempting to debug and document the exact rules for when recovery.bin gets ran by the boot rom

other issues have shown uart output when its running, so i have tried running it with only `recover.bin` and no other files, to ensure it only prints messages and cant flash

after several days of being confused by 4 blink codes, i used `rpi-eeprom-update` to reflash the eeprom and enable `BOOT_UART=1`

and thats when i confirmed/discovered, that `recovery.bin` itself, was giving the 4 blink code, which does not mean `start*.elf` not found as the forums claim

what blink codes is `recovery.bin` able to emit, and how do i convince it to print debug over the uart?

timg236, 2019-12-23:
Try adding uart_2ndstage=1 to config.txt 
It is probably complaining that there is no eeprom images to update. Not seen any evidence of that happening via rpi-eeprom-update. It’s debatable whether it’s better to stop and indicate that the update process is broken (current behaviour) somehow or to rename itself and reboot. Maybe the later is a better option but inevitably you get issues where the update fails silently.

The ROM always runs recovery.bin if it finds it and it appears to be valid.

If rpi-eeprom-update is deploying the wrong files then that would be interesting.

cleverca22, 2019-12-24:
how does the rom know if `recovery.bin` is valid?

currently, i'm intentionally skipping `rpi-eeprom-update` and copying only a `recovery.bin` to the boot partition

i have seen `recovery.bin` do uart output in other issues on this repo, so i'm a bit puzzled as to why its only blinking for me

lurch, 2019-12-24:
@cleverca22 Is there a problem you're tying to resolve (and if so, could you describe the problem in more detail), or is this all just out of curiosity?

cleverca22, 2019-12-24:
i have compiled a custom `recovery.bin` file, but the boot rom entirely ignores it because its "not valid"

timg236, 2019-12-24:
For bare metal development you need to create an a .elf file on Pi4.

Other people have done this already on Pi4 e.g  https://github.com/ptesarik/vc4boot

Closing bug because because rpi-eeprom-update is working as expected and bare metal isn't something that I can justify spending time supporting. However, other people on the advanced users or bare metal forum will be able to help ue

closed 2019-12-24 by timg236

cleverca22, 2019-12-24:
ah nice, i was wondering if there was any documentation on the `start.elf` api

that covers some other questions ive had, but still doesnt explain the `recovery.bin` validation layers

i'll keep poking at things on my own

cleverca22, 2020-01-16:
if anybody else is looking for the answer, `uart_2ndstage=1` in `config.txt` causes the official `recovery.bin` to print debug to the uart at 115200 baud