raspberrypi/rpi-eeprom #74: Pi 4 TFTP boot fails with ERROR EUNDEF "Early terminate"

IntelMiner, 2019-12-27:
Hi all,

I'm trying to boot an RPi 4 4GB unit using the latest master branch

The Pi seems to be endlessly looping at downloading files over TFTP. Doing a tcpdump reveals the following output

```
01:20:54.190362 IP 192.168.1.35.6970 > 192.168.1.199.69:  48 RRQ "d9780558/start.elf" octet tsize 0 blksize 1024
01:20:56.190341 IP 192.168.1.35.6970 > 192.168.1.199.69:  48 RRQ "d9780558/start.elf" octet tsize 0 blksize 1024
01:20:58.190323 IP 192.168.1.35.6970 > 192.168.1.199.69:  48 RRQ "d9780558/start.elf" octet tsize 0 blksize 1024
01:21:00.190311 IP 192.168.1.35.6970 > 192.168.1.199.69:  48 RRQ "d9780558/start.elf" octet tsize 0 blksize 1024
01:21:02.190267 IP 192.168.1.35.6970 > 192.168.1.199.69:  48 RRQ "d9780558/start.elf" octet tsize 0 blksize 1024
01:21:04.190242 IP 192.168.1.35.6970 > 192.168.1.199.69:  48 RRQ "d9780558/start.elf" octet tsize 0 blksize 1024
01:21:06.190234 IP 192.168.1.35.6970 > 192.168.1.199.69:  48 RRQ "d9780558/start.elf" octet tsize 0 blksize 1024
01:21:08.190188 IP 192.168.1.35.6970 > 192.168.1.199.69:  48 RRQ "d9780558/start.elf" octet tsize 0 blksize 1024
01:21:10.190168 IP 192.168.1.35.6970 > 192.168.1.199.69:  48 RRQ "d9780558/start.elf" octet tsize 0 blksize 1024
01:21:12.190137 IP 192.168.1.35.6970 > 192.168.1.199.69:  48 RRQ "d9780558/start.elf" octet tsize 0 blksize 1024
01:21:14.190117 IP 192.168.1.35.6970 > 192.168.1.199.69:  48 RRQ "d9780558/start.elf" octet tsize 0 blksize 1024
01:21:16.190091 IP 192.168.1.35.6970 > 192.168.1.199.69:  48 RRQ "d9780558/start.elf" octet tsize 0 blksize 1024
01:21:18.190075 IP 192.168.1.35.6970 > 192.168.1.199.69:  48 RRQ "d9780558/start.elf" octet tsize 0 blksize 1024
01:21:20.190055 IP 192.168.1.35.6970 > 192.168.1.199.69:  48 RRQ "d9780558/start.elf" octet tsize 0 blksize 1024
01:21:22.190003 IP 192.168.1.35.6970 > 192.168.1.199.69:  48 RRQ "d9780558/start.elf" octet tsize 0 blksize 1024
01:21:24.189886 IP 192.168.1.35.6970 > 192.168.1.199.69:  20 ERROR EUNDEF "Early terminate"
01:21:24.190017 IP 192.168.1.35.6971 > 192.168.1.199.69:  40 RRQ "config.txt" octet tsize 0 blksize 1024
01:21:26.190000 IP 192.168.1.35.6971 > 192.168.1.199.69:  40 RRQ "config.txt" octet tsize 0 blksize 1024
01:21:28.189974 IP 192.168.1.35.6971 > 192.168.1.199.69:  40 RRQ "config.txt" octet tsize 0 blksize 1024
01:21:30.189986 IP 192.168.1.35.6971 > 192.168.1.199.69:  40 RRQ "config.txt" octet tsize 0 blksize 1024
01:21:32.189942 IP 192.168.1.35.6971 > 192.168.1.199.69:  40 RRQ "config.txt" octet tsize 0 blksize 1024
01:21:34.189915 IP 192.168.1.35.6971 > 192.168.1.199.69:  40 RRQ "config.txt" octet tsize 0 blksize 1024
01:21:36.189890 IP 192.168.1.35.6971 > 192.168.1.199.69:  40 RRQ "config.txt" octet tsize 0 blksize 1024
01:21:38.189840 IP 192.168.1.35.6971 > 192.168.1.199.69:  40 RRQ "config.txt" octet tsize 0 blksize 1024
01:21:40.189840 IP 192.168.1.35.6971 > 192.168.1.199.69:  40 RRQ "config.txt" octet tsize 0 blksize 1024
01:21:42.189791 IP 192.168.1.35.6971 > 192.168.1.199.69:  40 RRQ "config.txt" octet tsize 0 blksize 1024
01:21:44.189792 IP 192.168.1.35.6971 > 192.168.1.199.69:  40 RRQ "config.txt" octet tsize 0 blksize 1024
01:21:46.189764 IP 192.168.1.35.6971 > 192.168.1.199.69:  40 RRQ "config.txt" octet tsize 0 blksize 1024
01:21:48.189712 IP 192.168.1.35.6971 > 192.168.1.199.69:  40 RRQ "config.txt" octet tsize 0 blksize 1024
01:21:50.189696 IP 192.168.1.35.6971 > 192.168.1.199.69:  40 RRQ "config.txt" octet tsize 0 blksize 1024
01:21:52.189668 IP 192.168.1.35.6971 > 192.168.1.199.69:  40 RRQ "config.txt" octet tsize 0 blksize 1024
01:21:54.189522 IP 192.168.1.35.6971 > 192.168.1.199.69:  20 ERROR EUNDEF "Early terminate"
```

I've tried disabling things such as letting the Pi negotiate the TFTP blocksize inside dnsmasq without success. I'm not sure if this is a bug or regression or what

IntelMiner, 2019-12-28:
Tried swapping out `dnsmasq` for `in.tftpd` to see if it was causing issues

Upon booting, the Pi seems to request the same files repeatedly, both from within the TFTP root directory and `/tftproot/d9780558/` (last 8 digits of the Pi's serial number)

After requesting these files the Pi seems to simply hang, with the green LED flashing 4 times 
```
18:23:31.455242 IP 192.168.1.32.40084 > 192.168.1.199.69:  49 RRQ "d9780558/start4.elf" octet tsize 0 blksize 1024
18:23:31.455888 IP 192.168.1.32.40085 > 192.168.1.199.69:  48 RRQ "d9780558/start.elf" octet tsize 0 blksize 1024
18:23:31.456343 IP 192.168.1.32.40086 > 192.168.1.199.69:  40 RRQ "config.txt" octet tsize 0 blksize 1024
18:23:31.456815 IP 192.168.1.32.40087 > 192.168.1.199.69:  42 RRQ "recover4.elf" octet tsize 0 blksize 1024
18:23:31.457226 IP 192.168.1.32.40088 > 192.168.1.199.69:  42 RRQ "recovery.elf" octet tsize 0 blksize 1024
18:23:31.457639 IP 192.168.1.32.40089 > 192.168.1.199.69:  40 RRQ "start4.elf" octet tsize 0 blksize 1024
18:23:31.458059 IP 192.168.1.32.40090 > 192.168.1.199.69:  39 RRQ "start.elf" octet tsize 0 blksize 1024
18:23:31.472443 IP 192.168.1.32.40091 > 192.168.1.199.69:  49 RRQ "d9780558/start4.elf" octet tsize 0 blksize 1024
18:23:31.473055 IP 192.168.1.32.40092 > 192.168.1.199.69:  48 RRQ "d9780558/start.elf" octet tsize 0 blksize 1024
18:23:31.473566 IP 192.168.1.32.40093 > 192.168.1.199.69:  40 RRQ "config.txt" octet tsize 0 blksize 1024
18:23:31.474028 IP 192.168.1.32.40094 > 192.168.1.199.69:  42 RRQ "recover4.elf" octet tsize 0 blksize 1024
18:23:31.474484 IP 192.168.1.32.40095 > 192.168.1.199.69:  42 RRQ "recovery.elf" octet tsize 0 blksize 1024
18:23:31.474890 IP 192.168.1.32.40096 > 192.168.1.199.69:  40 RRQ "start4.elf" octet tsize 0 blksize 1024
18:23:31.475299 IP 192.168.1.32.40097 > 192.168.1.199.69:  39 RRQ "start.elf" octet tsize 0 blksize 1024
18:23:31.489692 IP 192.168.1.32.40098 > 192.168.1.199.69:  49 RRQ "d9780558/start4.elf" octet tsize 0 blksize 1024
18:23:31.490286 IP 192.168.1.32.40099 > 192.168.1.199.69:  48 RRQ "d9780558/start.elf" octet tsize 0 blksize 1024
18:23:31.490698 IP 192.168.1.32.40100 > 192.168.1.199.69:  40 RRQ "config.txt" octet tsize 0 blksize 1024
18:23:31.491119 IP 192.168.1.32.40101 > 192.168.1.199.69:  42 RRQ "recover4.elf" octet tsize 0 blksize 1024
18:23:31.491537 IP 192.168.1.32.40102 > 192.168.1.199.69:  42 RRQ "recovery.elf" octet tsize 0 blksize 1024
18:23:31.491958 IP 192.168.1.32.40103 > 192.168.1.199.69:  40 RRQ "start4.elf" octet tsize 0 blksize 1024
18:23:31.492370 IP 192.168.1.32.40104 > 192.168.1.199.69:  39 RRQ "start.elf" octet tsize 0 blksize 1024
18:23:31.506714 IP 192.168.1.32.40105 > 192.168.1.199.69:  49 RRQ "d9780558/start4.elf" octet tsize 0 blksize 1024
18:23:31.507319 IP 192.168.1.32.40106 > 192.168.1.199.69:  48 RRQ "d9780558/start.elf" octet tsize 0 blksize 1024
18:23:31.507787 IP 192.168.1.32.40107 > 192.168.1.199.69:  40 RRQ "config.txt" octet tsize 0 blksize 1024
18:23:31.508209 IP 192.168.1.32.40108 > 192.168.1.199.69:  42 RRQ "recover4.elf" octet tsize 0 blksize 1024
18:23:31.508613 IP 192.168.1.32.40109 > 192.168.1.199.69:  42 RRQ "recovery.elf" octet tsize 0 blksize 1024
18:23:31.509022 IP 192.168.1.32.40110 > 192.168.1.199.69:  40 RRQ "start4.elf" octet tsize 0 blksize 1024
18:23:31.509425 IP 192.168.1.32.40111 > 192.168.1.199.69:  39 RRQ "start.elf" octet tsize 0 blksize 1024
18:23:31.523784 IP 192.168.1.32.40112 > 192.168.1.199.69:  49 RRQ "d9780558/start4.elf" octet tsize 0 blksize 1024
18:23:31.524382 IP 192.168.1.32.40113 > 192.168.1.199.69:  48 RRQ "d9780558/start.elf" octet tsize 0 blksize 1024
18:23:31.524827 IP 192.168.1.32.40114 > 192.168.1.199.69:  40 RRQ "config.txt" octet tsize 0 blksize 1024
18:23:31.525234 IP 192.168.1.32.40115 > 192.168.1.199.69:  42 RRQ "recover4.elf" octet tsize 0 blksize 1024
18:23:31.525640 IP 192.168.1.32.40116 > 192.168.1.199.69:  42 RRQ "recovery.elf" octet tsize 0 blksize 1024
18:23:31.526039 IP 192.168.1.32.40117 > 192.168.1.199.69:  40 RRQ "start4.elf" octet tsize 0 blksize 1024
18:23:31.526437 IP 192.168.1.32.40118 > 192.168.1.199.69:  39 RRQ "start.elf" octet tsize 0 blksize 1024
18:23:31.540799 IP 192.168.1.32.40119 > 192.168.1.199.69:  49 RRQ "d9780558/start4.elf" octet tsize 0 blksize 1024
18:23:31.541448 IP 192.168.1.32.40120 > 192.168.1.199.69:  48 RRQ "d9780558/start.elf" octet tsize 0 blksize 1024
18:23:31.541893 IP 192.168.1.32.40121 > 192.168.1.199.69:  40 RRQ "config.txt" octet tsize 0 blksize 1024
18:23:31.542299 IP 192.168.1.32.40122 > 192.168.1.199.69:  42 RRQ "recover4.elf" octet tsize 0 blksize 1024
18:23:31.542703 IP 192.168.1.32.40123 > 192.168.1.199.69:  42 RRQ "recovery.elf" octet tsize 0 blksize 1024
18:23:31.543107 IP 192.168.1.32.40124 > 192.168.1.199.69:  40 RRQ "start4.elf" octet tsize 0 blksize 1024
18:23:31.543526 IP 192.168.1.32.40125 > 192.168.1.199.69:  39 RRQ "start.elf" octet tsize 0 blksize 1024
18:23:31.557871 IP 192.168.1.32.40126 > 192.168.1.199.69:  49 RRQ "d9780558/start4.elf" octet tsize 0 blksize 1024
18:23:31.558465 IP 192.168.1.32.40127 > 192.168.1.199.69:  48 RRQ "d9780558/start.elf" octet tsize 0 blksize 1024
18:23:31.558906 IP 192.168.1.32.40128 > 192.168.1.199.69:  40 RRQ "config.txt" octet tsize 0 blksize 1024
18:23:31.559307 IP 192.168.1.32.40129 > 192.168.1.199.69:  42 RRQ "recover4.elf" octet tsize 0 blksize 1024
18:23:31.559713 IP 192.168.1.32.40130 > 192.168.1.199.69:  42 RRQ "recovery.elf" octet tsize 0 blksize 1024
[... cut ...]
18:23:31.713034 IP 192.168.1.32.40191 > 192.168.1.199.69:  40 RRQ "config.txt" octet tsize 0 blksize 1024
18:23:31.713466 IP 192.168.1.32.40192 > 192.168.1.199.69:  42 RRQ "recover4.elf" octet tsize 0 blksize 1024
18:23:31.713885 IP 192.168.1.32.40193 > 192.168.1.199.69:  42 RRQ "recovery.elf" octet tsize 0 blksize 1024
18:23:31.714306 IP 192.168.1.32.40194 > 192.168.1.199.69:  40 RRQ "start4.elf" octet tsize 0 blksize 1024
18:23:31.714737 IP 192.168.1.32.40195 > 192.168.1.199.69:  39 RRQ "start.elf" octet tsize 0 blksize 1024
```

timg236, 2020-01-02:
It’s hard to tell more without uart log or full wireshark. Best guesses are timeouts or file permissions on the server. The tsize is being reported as zero which is odd.
Might also be worth trying TFTP command on Linux to double check server setup

timg236, 2020-02-27:
This issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

fmcbra, 2020-03-13:
I'm seeing the same thing (or something very similar) on one of my v4 (revision 1.1) boards.

Permissions are fine on the server. Timeouts are the defaults but I'll see if increasing them makes a difference.

Am unable to get a log from the serial console at present (new lead on order) but I can do some digging with wireshark and post the results.

fmcbra, 2020-03-14:
I've captured a .pcapng file from wireshark. Is that helpful? Never ventured beyond tcpdump so unsure if what I've saved is actually helpful. If it is I'd be happy to e-mail it to a developer.

JeffReeves, 2020-04-26:
I was encountering this same issue with the latest eeprom (2020-04-16). The issue ended up being that my files were inaccessible on the TFTP server. I was using nfs-server per the instructions of a few tutorials, but ended up needing to install tftpd-hpa. Once I did that and verified the server was listening on UDP port 69, and that I could get/put a file from another machine using tftp's interactive mode, I was able to PXE boot the RPi4 client and have it run off of an NFS mounted root filesystem.

For fmcbra and anyone else encountering this issue - try watching a tcpdump on your DHCP server, and narrow it down to just the MAC of the RPi:
`tcpdump -vnes0 -i <interface> ether host <rpi-mac-address>`

timg236, 2020-05-02:
The problem appears to be server-side, without full UART and Wireshark it's not possible to diagnose so closing bug.

closed 2020-05-02 by timg236

arcivanov, 2022-02-21:
I'm seeing exactly the same thing with RPI3b+ while actually developing my own DHCP/TFTP farm, which gives me an ability to see the whole protocol and make necessary changes as I go. 

There seems to be some sort of a retry race inside the `bootcode.bin` which causes this. Namely there are RRQ for the same file multiple times sometimes one of them and sometimes all of them terminating concurrently.

I'm attaching full wireshark, my TFTP server logs and a screenshot of the semi-booted RPi.

arcivanov, 2022-02-21:
.129 is RPi
.59 is DHCP/TFTP

```
./main.py 
<snip DHCP negotiation>
tftp: ('192.168.159.129', 49153) b'\x00\x0189eb9bb9/start.elf\x00octet\x00tsize\x000\x00'
tftp read: ('192.168.159.59', 47288) ('192.168.159.129', 49153) 89eb9bb9/start.elf  ==>  start.elf
tftp read connect: ('192.168.159.59', 47288) ('192.168.159.129', 49153) start.elf octet tsize= 0 blksize= None
tftp: ('192.168.159.129', 49153) start.elf 6 0 b'\x00\x06tsize\x002965216\x00'
tftp: ('192.168.159.129', 49153) start.elf ACK block 0
tftp read options: ('192.168.159.59', 47288) ('192.168.159.129', 49153) start.elf octet tsize= 2965216 blksize= 512
tftp: ('192.168.159.129', 49154) b'\x00\x0189eb9bb9/start.elf\x00octet\x00tsize\x000\x00'
Traceback (most recent call last):
  File "src/gevent/greenlet.py", line 906, in gevent._gevent_cgreenlet.Greenlet.run
  File "/home/user/Documents/src/pxefarm/src/main/python/_main.py", line 456, in setup_tftp_transfer
    send(s, packet, block)
  File "/home/user/Documents/src/pxefarm/src/main/python/_main.py", line 380, in send
    handle_ack(block, verbose)
  File "/home/user/Documents/src/pxefarm/src/main/python/_main.py", line 400, in handle_ack
    raise RuntimeError(f"tftp: {addr} {filename} block {block} bytes sent {bytes_sent} ERROR {ack_packet}")
RuntimeError: tftp: ('192.168.159.129', 49153) start.elf block 1 bytes sent 512 ERROR b'\x00\x05\x00\x00Early terminate\x00'
2022-02-21T01:07:40Z <Greenlet at 0x7fbc263ac700: setup_tftp_transfer(('192.168.159.129', 49153), '89eb9bb9/start.elf', 'octet', tsize=0)> failed with RuntimeError

tftp read: ('192.168.159.59', 47868) ('192.168.159.129', 49154) 89eb9bb9/start.elf  ==>  start.elf
tftp read connect: ('192.168.159.59', 47868) ('192.168.159.129', 49154) start.elf octet tsize= 0 blksize= None
tftp: ('192.168.159.129', 49154) start.elf 6 0 b'\x00\x06tsize\x002965216\x00'
tftp: ('192.168.159.129', 49154) start.elf ACK block 0
tftp read options: ('192.168.159.59', 47868) ('192.168.159.129', 49154) start.elf octet tsize= 2965216 blksize= 512
tftp: ('192.168.159.129', 49155) b'\x00\x0189eb9bb9/autoboot.txt\x00octet\x00tsize\x000\x00'
Traceback (most recent call last):
  File "src/gevent/greenlet.py", line 906, in gevent._gevent_cgreenlet.Greenlet.run
  File "/home/user/Documents/src/pxefarm/src/main/python/_main.py", line 456, in setup_tftp_transfer
    send(s, packet, block)
  File "/home/user/Documents/src/pxefarm/src/main/python/_main.py", line 380, in send
    handle_ack(block, verbose)
  File "/home/user/Documents/src/pxefarm/src/main/python/_main.py", line 400, in handle_ack
    raise RuntimeError(f"tftp: {addr} {filename} block {block} bytes sent {bytes_sent} ERROR {ack_packet}")
RuntimeError: tftp: ('192.168.159.129', 49154) start.elf block 1 bytes sent 512 ERROR b'\x00\x05\x00\x00Early terminate\x00'
2022-02-21T01:07:40Z <Greenlet at 0x7fbc263ac940: setup_tftp_transfer(('192.168.159.129', 49154), '89eb9bb9/start.elf', 'octet', tsize=0)> failed with RuntimeError

tftp read: ('192.168.159.59', 47118) ('192.168.159.129', 49155) 89eb9bb9/autoboot.txt  ==>  autoboot.txt
tftp read connect: ('192.168.159.59', 47118) ('192.168.159.129', 49155) autoboot.txt octet tsize= 0 blksize= None
tftp: ('192.168.159.129', 49155) autoboot.txt no such file!
tftp: ('192.168.159.129', 49156) b'\x00\x0189eb9bb9/config.txt\x00octet\x00tsize\x000\x00'
tftp read: ('192.168.159.59', 59077) ('192.168.159.129', 49156) 89eb9bb9/config.txt  ==>  config.txt
tftp read connect: ('192.168.159.59', 59077) ('192.168.159.129', 49156) config.txt octet tsize= 0 blksize= None
tftp: ('192.168.159.129', 49156) config.txt 6 0 b'\x00\x06tsize\x002075\x00'
tftp: ('192.168.159.129', 49156) config.txt ACK block 0
tftp read options: ('192.168.159.59', 59077) ('192.168.159.129', 49156) config.txt octet tsize= 2075 blksize= 512
tftp: ('192.168.159.129', 49156) config.txt blocks=5, bytes=2075 DONE!
tftp: ('192.168.159.129', 49157) b'\x00\x0189eb9bb9/recovery.elf\x00octet\x00tsize\x000\x00'
tftp read: ('192.168.159.59', 39741) ('192.168.159.129', 49157) 89eb9bb9/recovery.elf  ==>  recovery.elf
tftp read connect: ('192.168.159.59', 39741) ('192.168.159.129', 49157) recovery.elf octet tsize= 0 blksize= None
tftp: ('192.168.159.129', 49157) recovery.elf no such file!
tftp: ('192.168.159.129', 49158) b'\x00\x0189eb9bb9/start.elf\x00octet\x00tsize\x000\x00'
tftp read: ('192.168.159.59', 50998) ('192.168.159.129', 49158) 89eb9bb9/start.elf  ==>  start.elf
tftp read connect: ('192.168.159.59', 50998) ('192.168.159.129', 49158) start.elf octet tsize= 0 blksize= None
tftp: ('192.168.159.129', 49158) start.elf 6 0 b'\x00\x06tsize\x002965216\x00'
tftp: ('192.168.159.129', 49158) start.elf ACK block 0
tftp read options: ('192.168.159.59', 50998) ('192.168.159.129', 49158) start.elf octet tsize= 2965216 blksize= 512
Removed request (b"\xb8'\xeb\xeb\x9b\xb9", b'&\xf3\x039') at 343105.882668642
tftp: ('192.168.159.129', 49159) b'\x00\x0189eb9bb9/fixup.dat\x00octet\x00tsize\x000\x00'
tftp: ('192.168.159.129', 49158) start.elf blocks=5792, bytes=2965216 DONE!
tftp read: ('192.168.159.59', 47908) ('192.168.159.129', 49159) 89eb9bb9/fixup.dat  ==>  fixup.dat
tftp read connect: ('192.168.159.59', 47908) ('192.168.159.129', 49159) fixup.dat octet tsize= 0 blksize= None
tftp: ('192.168.159.129', 49159) fixup.dat 6 0 b'\x00\x06tsize\x007221\x00'
tftp: ('192.168.159.129', 49159) fixup.dat ACK block 0
tftp read options: ('192.168.159.59', 47908) ('192.168.159.129', 49159) fixup.dat octet tsize= 7221 blksize= 512
[... cut ...]
tftp read connect: ('192.168.159.59', 41893) ('192.168.159.129', 49182) kernel7.img octet tsize= 0 blksize= None
tftp: ('192.168.159.129', 49182) kernel7.img 6 0 b'\x00\x06tsize\x006596992\x00'
tftp: ('192.168.159.129', 49182) kernel7.img ACK block 0
tftp read options: ('192.168.159.59', 41893) ('192.168.159.129', 49182) kernel7.img octet tsize= 6596992 blksize= 512
tftp: ('192.168.159.129', 49182) kernel7.img blocks=12885, bytes=6596992 DONE!
Removed boot offer 192.168.159.129 at 343156.899704149
```

arcivanov, 2022-02-21:
[rpi3_boot_tftp_early_termination.tar.gz](https://github.com/raspberrypi/rpi-eeprom/files/8105427/rpi3_boot_tftp_early_termination.tar.gz)

arcivanov, 2022-02-21:
@timg236 please reopen

arcivanov, 2022-02-21:
bootcode.bin is used from the SD card and is this: https://github.com/raspberrypi/firmware/blob/a6496ae5cd5e9b4cd5b38f7274fa94dce315fa7a/boot/bootcode.bin

arcivanov, 2022-02-21:
[image]

arcivanov, 2022-02-21:
And same exact behavior on RPI 2B v1.1. And I don't have any more RPis to try :smile: 

[rpi2bv11_boot_tftp_early_termination.tar.gz](https://github.com/raspberrypi/rpi-eeprom/files/8105539/rpi2bv11_boot_tftp_early_termination.tar.gz)

arcivanov, 2022-02-21:
This is either failure to synchronize a queue for the multiple files across several cores or some sort of bug in a multi-core retry logic.

Because the requests are repeated with the time difference of 5-9 ms I suspect it's a multi-core/multi-threaded queue synchronization issue rather than the retry logic bug.

timg236, 2022-02-21:
Please don't post on closed issues unless you are the OP.

N.B. Since the rpi-eeprom is specific to the Raspberry Pi 4 family it's not the right place to to discuss Pi2 / Pi3 issues.