raspberrypi/rpi-eeprom #78: RPi4 netboot

jonwitts, 2020-01-14:
I have successfully followed this guide: https://github.com/raspberrypi/rpi-eeprom/blob/master/firmware/raspberry_pi4_network_boot_beta.md and added a RPi4 to our PiServer install and have it net-booting.

My question is what is the easiest way to update a lab full of RPi4 to use the ne-tboot eeprom without having to follow all of these steps on every RPi4?

Thanks,
Jon

timg236, 2020-01-14:
You're more likely to get a response on the Raspberry Pi General Forums instead of raising issues
https://www.raspberrypi.org/forums/viewforum.php?f=63
Try searching for PiServer

jonwitts, 2020-01-14:
Thanks... 
PiServer is running absolutely fine. The question was about flashing a whole lab of RPi to the latest eeprom version; hence why I thought this would be the better place to ask...

timg236, 2020-01-14:
There isn't anything in this repo which manages that, other people may have written flashrom scripts or just created and sd-card with pieeprom.bin (so it doesn't get renamed) and done this manually since it only takes about 15 seconds per board

jonwitts, 2020-01-14:
OK; thanks for responding. I could not see anything on the general forums; but I will see if I can create a custom SD Card with the pieeprom.bin on it...

closed 2020-01-14 by jonwitts

timg236, 2020-01-14:
Don't forget pieeprom.sig
sha256sum pieeprom.bin > pieeprom.sig

jonwitts, 2020-01-14:
Thanks!

Looking in the eeprom recovery zip it has the following files:

- pieeprom.bin
- pieeprom.sig
- README
- recovery.bin
- vl805.bin
- vl805.sig

Do I need any of those too?

timg236, 2020-01-14:
You only need to update vl805.bin if you want to upgrade it, although I would probably recommend switching to the 0137ad version (from the beta directory). It has all the power management savings + fixes for isochronous devices (typically webcams). At some point that will get pushed to 'critical'
The README is not required.

jonwitts, 2020-01-15:
Adding the sha256sum worked perfectly! Many thanks :-D