raspberrypi/rpi-eeprom #82: DHCP options to identify RPi, TFTP blocksize/windowsize

beldzhang, 2020-01-15:
_Post this to the forum firstly, and I think it's better be here._

Hi, I'm a developer focused on network boot for about 20 years, mainly support x86 pc boot to Windows and Linux.
I got RPi4 just one week, already tested raspbian, ubuntu, debian and network boot.

some issues about network boot

**0) testing environment**
EEPROM version: pieeprom-2020-01-09.bin
DHCP/PXE/TFTP server: developed by myself.

**1) in DHCP requests:**
other than the MAC address, there is no any field to identify a RPi client,
option 93 is set to "0" means IA x86 PC
option 97 UUID just repeat serial no 4 times
the OUI(dc:a6:32) can be use, but it not reliable as my experience
why this needed?
because the PXE options need a special menu item to continue and need to identify x86(bios/uefi) and RPI
suggestion:
option 97, change to magic number + serial no, there is enough length
-or-
option 93, acquire a unique number for RPi (it's complex with IEEE/IANA/RFC...)
-or-
any other option not defined by RFC

**2)in DHCP REQUEST packet, there is no PXEClient(60) option**
this is not consistency whit DHCP DISCOVER packet.
in my server, I just skip the packets without PXEClient(60), many desktop/laptop/mobile will sent request, and some ones running like flood

**3) in DHCP response option 0 (padding) is not handled properly**
then this response ignored.
in some(many) not implemented very well DHCP/PXE bootrom, padding zero is required to let them got correct hostname / bootfile / etc...

**4) TFTP download files**
blocksize is 1024, could be larger?
is there any future plan to implement windowsize ? since start.elf / kernel / initrd are big, set windowsize > 1 to speed up downloading.

timg236, 2020-01-15:
1/2) The PXE options might be made configurable via bootconf.txt as part of the IPV6 work but new network boot features are being added until after USB boot is complete.

3) Can you point to an example where (3) fails using the latest BETA pieprom.bin - January. e.g. dnsmasq config + wireshark ?

4) Very unlikely right now. When all the other features are in place we'll look at how much code / memory space is left for optimisations.
More likely we'd experiment with downloading a compressed blob for start.elf + kernel

beldzhang, 2020-01-15:
captured 2 files by wireshark (boot from my server):
  without padding: RPi boot normally
  with padding: DHCP Offer unrecongnized
after option 12 & 67, put a padding(0) to let ROM got string correctly, as C string is null-terminated.

don't know how to set a "Padding Option(0)" in dnsmasq by far... 

[rpi4-dhcp-padding-option.zip](https://github.com/raspberrypi/rpi-eeprom/files/4066782/rpi4-dhcp-padding-option.zip)

timg236, 2020-01-15:
Thanks.
Which version of the pieeprom.bin are you using ? btw uart-trace is also useful if you happen to have a USB serial cable

beldzhang, 2020-01-15:
testing is done with pieeprom-2020-01-09.bin
unfortunately do not has the cable
the software could give to you if like to.

thanks for your hard works.

timg236, 2020-01-15:
No problem, I see the issue. It looks as though Pi4 (and earlier models) assume that there is an explicit length field for all options except 0xff. It should be a trivial fix.

ghollingworth, 2020-01-15:
My bad...  :)

beldzhang, 2020-01-17:
about "mmc0: Timeout waiting for hardware cmd interrupt."

network boot passed!
there is a __lot__ of mmc0 related message from kernel

kernel version rpi-5.4.y(5.4.11) from here, cross build on Arch using bcm2711_defcofig
__no__ any microSD inserted, just boot from network.

it's not shown on any distro running from SD card.
any suggestion or anything I could have a try ?

Thanks.

timg236, 2020-01-17:
It's a bug in the upstream SDHCI driver not coping with a missing card if card-detect is not enabled
https://github.com/raspberrypi/linux/issues/2802

timg236, 2020-01-17:
Please can you try the fix for Option0 padding in this PR. I don't have suitable DHCP test environment for this but it's a very small code change
https://github.com/raspberrypi/rpi-eeprom/pull/84

beldzhang, 2020-01-17:
tested, not working ...

I could make a small DHCP/TFTP server to testing, would you like to?
If yes, please just let me know. (and windows or linux version?)

timg236, 2020-01-17:
I missed a bit but if this doesn't work a Linux server would be interesting, does it run on Raspberry Pi ?

[pieeprom.zip](https://github.com/raspberrypi/rpi-eeprom/files/4077601/pieeprom.zip)

beldzhang, 2020-01-17:
> 
> 
> I missed a bit but if this doesn't work a Linux server would be interesting, does it run on Raspberry Pi ?
> 
> [pieeprom.zip](https://github.com/raspberrypi/rpi-eeprom/files/4077601/pieeprom.zip)

this one is worked. 😄

timg236, 2020-01-17:
Great thanks for testing. I've updated the latest beta binary with the correction. The version changed because I also added an internal unit-test for this against the pcap file which you uploaded :)

beldzhang, 2020-01-20:
I make a video about RPi4 diskless boot to debian 😄
https://youtu.be/0AhnETthrgo

as a reference, there is another Linux Mint 19.3 diskless boot (in vbox)
https://youtu.be/jsGrSQXxI8g

timg236, 2020-01-21:
If the padding is fixed in the latest beta please could you close this bug ? Feel free to raise a bug for (1) / identifying clients as raspberry pi. 
I'd suggest an EEPROM config for customizing Option97 e.g. 
magic + board-revision + serial + reserved
The default may need to remain the serial repeated 4 times, at for a while in order to discover anyone relies upon this behaviour

beldzhang, 2020-01-21:
I thought it should be closed by you :)
Option 97 could be customized is a very good idea.
Now I'm optimizing my driver to boost booting time.(not a nfs-root style diskless boot)

closed 2020-01-21 by beldzhang

timg236, 2020-03-11:
@beldzhang I've pushed a new beta image with the Option97 change and quite a few other features which have been sat in the queue.
Hopefully, this can be promoted to stable fairly soon and the focus on the beta branch will switch to USB MSD boot.

beldzhang, 2020-03-12:
emmmm.....
@timg236 where can I download the files? I'm not familiar with git and github ...

timg236, 2020-03-12:
Sorry, I must have pushed to my fork and not completed the merge request. It's there now

timg236, 2020-03-12:
https://github.com/raspberrypi/rpi-eeprom/tree/master/firmware/beta

pieeprom-2020-03-04.bin