raspberrypi/rpi-eeprom #83: Dynamic Hostname

locutusofsand, 2020-01-16:
I am planning on deploying about 30 Pi's in an industrial environment at individual work centers. I am going to use them much like a kiosk to do various web based functions. The pages I need to serve up are unique to each work center. Ideally I would like to boot directly to the needed page. The only way that I have thought to do that is if I could give each of the Pi's its own host name and use that host name to populate the link in the script I use to boot from. Is there any way that I could tie the description field under the Clients tab in PiServer to the hostname of each device?

timg236, 2020-01-16:
The hostname is by the host OS (Linux) not the bootloader firmware . PiServer is not part of this repo

lurch, 2020-01-16:
> use that host name to populate the link in the script I use to boot from.

Alternatively I guess you could read the serial number or MAC address of each Pi (which will always be unique) in your boot-script, and use that to create your "boot link" ?

locutusofsand, 2020-01-16:
> The hostname is by the host OS (Linux) not the bootloader firmware . PiServer is not part of this repo

I'm sorry, I have too many open repositories, I actually intended to post it there. I will now.

> > use that host name to populate the link in the script I use to boot from.
> 
> Alternatively I guess you could read the serial number or MAC address of each Pi (which will always be unique) in your boot-script, and use that to create your "boot link" ?

However, along this comment, what would be the most consistent way to pull either of these that would work on both RPi 3 & 4?

Thanks.

timg236, 2020-01-16:
You might be better off asking this question on the Raspberry Pi forums. 
https://www.raspberrypi.org/forums//viewforum.php?f=63

locutusofsand, 2020-01-16:
OK, I will close since this is in the wrong spot. Thanks for the help.

closed 2020-01-16 by locutusofsand