raspberrypi/rpi-eeprom #86: After upgrade to 000137ad USB-stability problems

robsy, 2020-01-23:
Hi,

after the upgrade to 000137ad (latest stable) I have  the issue, that the USB-hub "disappears" after ~18h. 
[67672.041354] xhci_hcd 0000:01:00.0: xHCI host not responding to stop endpoint command.
[67672.057405] xhci_hcd 0000:01:00.0: Host halt failed, -110
[67672.057414] xhci_hcd 0000:01:00.0: xHCI host controller not responding, assume dead
[67672.057463] xhci_hcd 0000:01:00.0: HC died; cleaning up
[67672.057558] usb 1-1: USB disconnect, device number 2
[67672.057577] usb 1-1.4: USB disconnect, device number 3
[67672.057895] ftdi_sio ttyUSB0: ftdi_set_termios FAILED to set databits/stopbits/parity
[67672.057916] ftdi_sio ttyUSB0: ftdi_set_termios urb failed to set baudrate
[67672.057928] ftdi_sio ttyUSB0: failed to set flow control: -19
[67672.060856] ftdi_sio ttyUSB0: FTDI USB Serial Device converter now disconnected from ttyUSB0
[67672.060963] ftdi_sio 1-1.4:1.0: device disconnected

and lsusb:
root@xxxx:~ # lsusb 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

after a reboot (without any disconnecting of cables, only "reboot"):
root@xxxx:~ # lsusb 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

In my eyes the most important difference is in dmesg:
-usb 1-1: New USB device found, idVendor=2109, idProduct=3431, bcdDevice= 4.20
+usb 1-1: New USB device found, idVendor=2109, idProduct=3431, bcdDevice= 4.21

This issue happens ~18h after reboot. The same hardware worked without any problems with the previrous stable eeprom-image  for many weeks.

Any hints?

Robert

timg236, 2020-01-24:
It's probably best to check the firmware repo for USB issues, this is just the wrapper so it won't get much attention here.

There were some reports of USB issues possibly related to DVFS with the latest firmware.

lurch, 2020-01-25:
ping @P33M

P33M, 2020-01-27:
It's nothing to do with the VLI EEPROM image - it's most likely the DVFS firmware (which undervolts) being marginal on your board.
https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=262649

closed 2020-02-27 by timg236