raspberrypi/rpi-eeprom #92: Unnecessary check for '*.elf' in BOOTFS

kpfleming, 2020-02-17:
https://github.com/raspberrypi/rpi-eeprom/blob/30905b49096df59e50694fab05bcca55b66be5ef/rpi-eeprom-update#L433

I understand the desire to protect users from making mistakes, but I've just setup an RPi 4B with network boot (yay!), and as a result I have an SD card in the board with nothing on it. `rpi-eeprom-update` refused to put a bootloader update on that filesystem until I created a dummy .elf file. After doing that the bootloader file was placed there, I rebooted, and the board successfully applied the update and removed the file (and then proceeded on with its network boot).

If it's important to keep this check, then the documentation should probably be updated as a warning to others who are switching to network boot and won't have the usual contents of `/boot` in their SD card filesystem.

timg236, 2020-03-16:
For network boot there is now a self-update mode (default off) where if you mount the tftpboot directory as /boot then you can use rpi-update as thought it was on a sd-card.
Instead of using recovery.bin the bootloader updates itself, you just have to avoid powering off the Pi during the couple of seconds whilst an update is running.

timg236, 2020-05-02:
I think it would be reasonable for this to be a warning instead of an error

gukandrew, 2020-05-19:
Same here!
This prevents to update eprom using next command from tutorial:
`sudo rpi-eeprom-update -d -f ./pieeprom-new.bin`

timg236, 2020-05-26:
rpi-eeprom-update has been updated, it will be available the next time the APT package is updated or clone the script directly.

closed 2020-05-26 by timg236