raspberrypi/rpi-eeprom #93: SSH disabled while netbooting

Mavyre, 2020-02-17:
Hi,

It seems that the SSH isn't working while netbooting, even after activating it with raspi-config before rsyncing it to the servers. Also, adding ssh file to both the TFTP boot root, TFTP boot specific to that raspberry or adding it to its filesystem directly does not seem to work...

Any fixes recommended?

Thank you!

lurch, 2020-02-18:
Any ideas @XECDesign ?

XECDesign, 2020-02-18:
Not based on that information alone, since it works perfectly well here. I think this should be a forum question, as it's not an rpi-eeprom issue.

Mavyre, 2020-02-18:
Hi. Chrooting and using `systemctl enable ssh` wasn't enough.

Then, using NFSv4.1 and editing /etc/fstab to get the boot partition (which contains ssh file) was enough to enable it. I'm closing this issue as this was my mistake not to mount boot partition and not to pay attention on the version of NFS I was using.

Thank you for your fast responses!

closed 2020-02-18 by Mavyre