raspberrypi/rpi-eeprom #94: Python3 support

jvonau, 2020-02-18:
> apt-rdepends -r --state-follow=Installed --state-show=Installed python2
Reading package lists... Done
Building dependency tree       
Reading state information... Done
python2
  Reverse Depends: python (= 2.7.16-1)
python
  Reverse Depends: python-rpi.gpio (<< 0.6.5-1)
  Reverse Depends: rpi-eeprom (4.0-1)
python-rpi.gpio
rpi-eeprom

Was wondering if this program could be ported to python3 and the apt package updated to reflect the change. Trying to build a pure python3 image for the rpi4 with no python2 installed but there is a dependence from raspberrypi-bootloader upon rpi-eeprom. The other python-rpi.gpio dependence of python2 I can drop within pi-gen when building the image. Thanks in advance.

lurch, 2020-02-18:
Yeah, probably makes sense to migrate this to Python3, given that Python2 is now EOL.
Ping @XECDesign who I believe is responsible for Raspbian packaging.

> The other python-rpi.gpio dependence of python2

I believe there's already a `python3-rpi.gpio` package ?

XECDesign, 2020-02-18:
Some context - https://github.com/raspberrypi/rpi-eeprom/issues/43

timg236, 2020-02-21:
The program should work on either Python2 or Python3 but it’s probably reasonable for the package to require Python3 these days

timg236, 2020-05-02:
@XECDesign @MilhouseVH 
Happy to change this once we are sure that this won't break on LibreElec

MilhouseVH, 2020-05-02:
Debian packaging rules have no impact on LibreELEC, we only require that `rpi-eeprom` continues to work with Python2 as we distribute `rpi-eeprom with LibreELEC 9 (Python2-only) and LibreELEC 10 (Python3-only).

timg236, 2020-05-02:
Sorry, to be clear I was referring to #!/usr/bin/python and whether that will break if it becomes #!/usr/bin/python3

MilhouseVH, 2020-05-02:
Oh right, yes that will be a problem on a Python2-only system.

MilhouseVH, 2020-05-02:
If absolutely necessary we can patch it back to `#!/usr/bin/python` for LibreELEC 9, provided that Python2 compatibility in the rest of the code is maintained.

lurch, 2020-06-22:
Looks like #164 might fix the original issue?

timg236, 2020-06-22:
The Debian package should depend on Python 3 so that distributions don't have to include Python3. However, for distributions which only include Python2 the script shouldn't add a dependency on Python3. One workaround might be to make rpi-eeprom-config a module and import it via a different Python2/Python3 wrappers

andrum99, 2020-06-22:
> The Debian package should depend on Python 3 so that distributions don't have to include Python3. However, for distributions which only include Python2 the script shouldn't add a dependency on [...]

As I understand it, #164 makes the `rpi-eeprom` package depend on the `python` package OR the `python3` package, i.e. either one satisfies the dependency.

lpla, 2020-07-02:
Hi. Just my 2 cents. Let me show some cases:

- Some OS like Debian create a `/usr/bin/python` symlink only when `python2` is installed, so it won't be created if there is only `python3` in the system (Raspbian desktop-less). In that case the current shebang won't find Python.
- Some other OS like Ubuntu behave the same way as Debian but you can add a dependency to install a package like `python-is-python3` which creates this symlink pointing to `python3`, which is cleaner than manually creating that symlink, but still, can be fatal if there is `python2` installed along and any old script calls `/usr/bin/python` without Python 3 support.
- Other OS, like [Arch](https://wiki.archlinux.org/index.php/Python#Installation), forced users to update their scripts by pointing `/usr/bin/python` to `python3`. In this case, current shebang won't break as the script has compatibility for both Python 2 and 3.

Also, there is another issue with the current shebang, as discussed in [here](https://stackoverflow.com/questions/6908143/should-i-put-shebang-in-python-scripts-and-what-form-should-it-take) and [there](https://blogs.gnome.org/mcatanzaro/2018/02/16/on-python-shebangs/).

In my opinion, I guess that it would be easier to deprecate `python-minimal` dependency installation in Debian `control` file, instead of making it complementary like in #164, avoiding installing `python-minimal` in setups without any other Python 2 script.

So a shebang like `#!/usr/bin/env python3` will work in most modern systems that include a `python3`, and even older OS that also include `python2` along with, like Ubuntu 16.04.

If OS like OpenELEC prefer to point their packaged version to `#!/usr/bin/env python` because they don't include any `python3` at installation time in a specific release, that's fine, it will work.

Hope this is helpful.

timg236, 2020-07-20:
I think this is all sorted now - closing Issue

closed 2020-07-20 by timg236