raspberrypi/rpi-eeprom #103: DHCP_OPTION97: no ETH_MAC_LSB part in new GUID

beldzhang, 2020-03-14:
1) tested 2020-03-04, option 97 already in new spec, but the ETH_MAC_LSB part is missing, the 4 bytes are all zero, not copied from my rpi4.
tcpdump:  [rpi4.zip](https://github.com/raspberrypi/rpi-eeprom/files/4332137/rpi4.zip)

2) another discuss of PXE_OPTION43: PXE menu option, maybe too late but just let's talk about it
in previous version, the eeprom was check "Raspberry Pi" part only and can not modified, this version, it can be customized to any user specified string.
as my guess, it's used for identify a "regular DHCP server" or a "netboot enabled DHCP server", but it's no need to so complicated, in a "regular" server, it will never set option-60("PXEClient"), then check option-60 is well enough
is there any reason to make this mandatory?

3) any plan to support proxyDHCP mode? PXE spec support to get IP address from a "regular" server but get boot file/info from a proxyDHCP server.
in a large network, users may not want/allow multi DHCP server co-exists, and may not "touch" the DHCP server at all, so proxyDHCP mode is very necessary.

timg236, 2020-03-14:
1. I think I see the issue, some code got moved around breaking the mac-lsb copy in this section. It's a trivial fix so I'll try to pull that into the next beta.

2. The PXE_OPTION43 string just allows "Raspberry Pi Boot" to be overridden with a different constant string. I don't really recommend people change it and it's always better to do things at the server side but at least they have that option now.

3. We use proxyDHCP at Raspberry Pi and the implementation on Pi4 is the same as Pi3.
The real DHCP server returns the IP address and the proxyDHCP (dnsmasq) returns the the TFTP server IP or Option43. The "PiServer" uses dnsmasq to do this as well.

timg236, 2020-03-16:
I've pushed a new beta which should fix the Option97 issue.

beldzhang, 2020-03-16:
1) testing passed.

2) got that.

3) will test in next developing stage.

timg236, 2020-03-17:
Great. Thanks for the test report, it's really useful to get such detailed feedback.

beldzhang, 2020-03-17:
Thanks for your great works! and I think we could close this one.

closed 2020-03-17 by beldzhang