raspberrypi/rpi-eeprom #113: Last beta/pieeprom-2020-16-04 doesn't follow the date format YYYY-MM-DD

txenoo, 2020-04-16:
All pieprom files have been following the YYYY-MM-DD format but last one 'pieeprom-2020-16-04.bin'  is using  YYYY-DD-MM, when next pieeeprom file will be released it would be more difficult to detect the last one as pieeprom-2020-16-04 would be the last in order during 2020.

$dpkg -s rpi-eeprom-images 
Package: rpi-eeprom-images
Status: install ok installed
Priority: optional
Section: misc
Installed-Size: 10669
Maintainer: Serge Schneider <serge@raspberrypi.org>
Architecture: all
Source: rpi-eeprom
Version: 5.6-1
Description: Raspberry Pi 4 boot EEPROM images
 This package contains the bootloader EEPROM images for Raspberry Pi 4
Homepage: https://github.com/raspberrypi/rpi-eeprom/

$ ls /lib/firmware/raspberrypi/bootloader/beta/pieeprom-*|tail -n 4
/lib/firmware/raspberrypi/bootloader/beta/pieeprom-2020-03-16.bin
/lib/firmware/raspberrypi/bootloader/beta/pieeprom-2020-03-19.bin
/lib/firmware/raspberrypi/bootloader/beta/pieeprom-2020-04-09.bin
/lib/firmware/raspberrypi/bootloader/beta/pieeprom-2020-16-04.bin

timg236, 2020-04-17:
Weird,  I must have been working late to slip into a different timezone. Will fix.

timg236, 2020-04-17:
@XECDesign sorry - renamed it now

lurch, 2020-04-22:
Fixed by #114

timg236, 2020-04-22:
Thanks @lurch

closed 2020-04-22 by timg236