raspberrypi/rpi-eeprom #116: release notes, change and additions

pieter-groeneweg, 2020-04-24:
When I check for updates on my RPI4 raspbian 10 I see this:

Package | Current version | New version | Description
-- | -- | -- | --
rpi-eeprom-images | 5.7-1 | 5.8-1 | all Raspberry Pi 4 boot EEPROM images
rpi-eeprom | 5.7-1 | 5.8-1 | all Raspberry Pi 4 boot EEPROM updater

Of course it is super there are continious improvements. But I seem not to be able to relate these version numbers anywhere. Any search on Google seems to exclude the version numbers.

TX,

XECDesign, 2020-04-24:
The package versioning is a bit detached from the release notes versions here. Maybe https://github.com/raspberrypi/rpi-eeprom/blob/debian/buster/debian/changelog helps.

pieter-groeneweg, 2020-04-24:
O, great thank you very much! That helps indeed.

closed 2020-04-24 by pieter-groeneweg