raspberrypi/rpi-eeprom #119: rpi-eeprom-update doesn't say it is only for Pi 4B

andrum99, 2020-04-29:
The documentation built into the rpi-eeprom-update tool does not specify that it only works on the Pi 4B. Since the package gets installed on all Raspberry Pis (which run Raspbian), it needs to document that it only does anything useful on the Pi 4B.

timg236, 2020-04-29:
It already does the right thing if you run it i.e. nothing

timg236, 2020-04-29:
N.B. The tool is likely to be used on future products e.g. CM4. There's a general problem of not wanting to bake a list of every possible product / sku into documentation, there have been some internal discussions about how to describe that concisely. 
@JamesH65 for when we know what the right text is

andrum99, 2020-04-29:
> It already does the right thing if you run it i.e. nothing

I know. But the built-in documentation doesn't explain why it does nothing. There's also an argument that when you run it something without a BCM2711 it should tell you why it's done nothing.

andrum99, 2020-04-29:
> N.B. The tool is likely to be used on future products e.g. CM4. There's a general problem of not wanting to bake a list of every possible product / sku into documentation, there have been some [...]

rpi-eeprom-config specifies it is for Pi 4B only. Should that be amended to say 'Raspberry Pi 4', and rpi-eeprom-update have 'Raspberry Pi 4' added to it?

lurch, 2020-04-29:
Given the special-purpose nature of `rpi-eeprom-update`, I'm struggling to imagine a scenario in which somebody _without_ a Pi 4B would be running it? :confused:

mrankine, 2020-05-08:
> Given the special-purpose nature of `rpi-eeprom-update`, I'm struggling to imagine a scenario in which somebody _without_ a Pi 4B would be running it? 😕

It's in the Raspbian image, so everyone that hasn't specifically removed it will be running it on their Pi 0-3.

(It's also 13MB of unnecessary SD card space.)

lurch, 2020-05-08:
The "Raspbian philosophy" is to have a single SD card image that's bootable on **all** Raspberry Pis.

mrankine, 2020-05-08:
Exactly.

See the initial issue:
> The documentation built into the rpi-eeprom-update tool does not specify that it only works on the Pi 4B. Since the package gets installed on all Raspberry Pis (which run Raspbian), it needs to document that it only does anything useful on the Pi 4B.

ghollingworth, 2020-05-08:
Please supply a PR for changes we can discuss

closed 2020-05-08 by ghollingworth

lurch, 2020-05-11:
> (It's also 13MB of unnecessary SD card space.)

You'll be pleased to see that there are now plans in place to reduce the size of the rpi-eeprom packages in Raspbian #122 :wink: