raspberrypi/rpi-eeprom #120: Add support for booting from a "superfloppy" disk

worproject, 2020-05-02:
The boot ROM itself can boot just fine from a superfloppy disk (i've tested this with the recovery.bin image), but if there's no recovery image on the disk, it loads the EEPROM bootloader which can only boot from MBR-partitioned disks.

This worked fine on the Pi 3 as the bootcode was loaded by the boot ROM directly from the SD card.

timg236, 2020-05-02:
Right now, I can't

worproject, 2020-05-02:
Indeed, GPT support would be better. In fact I was using superfloppy to workaround the lack of native GPT support.

timg236, 2020-05-02:
Thanks that’s useful to know

timg236, 2020-05-20:
Since we are adding GPT and 0xef partition support we may as well add SuperFloppy whilst we are editing that bit of code.
Hopefully, fairly standard MBR (primary + extended) and GPT will then just work.

timg236, 2020-05-26:
This should be fixed by e87bc4da8d983aa6e7c3edcf8cd823b76e50e65e

The binaries are available in this release which can be flashed using the Raspberry Pi Imager
https://github.com/raspberrypi/rpi-eeprom/releases/tag/v2020.05.26-137ad

timg236, 2020-06-04:
Fix is now available via rpi-eeprom apt package

closed 2020-06-04 by timg236