raspberrypi/rpi-eeprom #121: SELF_UPDATE=1 disappears after every update

starbasessd, 2020-05-04:
Trying to have Pi4 4GB do auto eeprom updates due to being net-booted
eeprom beta pieeprom-2020-04-16.bin
pi@raspi1:~ $ vcgencmd bootloader_config
[all]
BOOT_UART=0
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0
DHCP_TIMEOUT=45000
DHCP_REQ_TIMEOUT=4000
TFTP_FILE_TIMEOUT=30000
TFTP_IP=
TFTP_PREFIX=0
BOOT_ORDER=0x21
SD_BOOT_MAX_RETRIES=3
NET_BOOT_MAX_RETRIES=5
[none]
FREEZE_VERSION=0
Follow instructions to extract, modify, re-apply, and re-install here:
https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711_bootloader_config.md
adding SELF_UPDATE=1 after the line [all], and before and after [none]
I have been able to have BOOT_ORDER=0x21 to do Net Boot, successfully.
Just can't get the SELF_UPDATE to stay.

timg236, 2020-05-04:
The bootloader doesn't modify the EEPROM config so there must be a stale pieeprom.bin/pieeprom.upd on either the TFTP boot or a local sd-card.

N.B. Since SELF_UPDATE is not set in the default bootloader configuration you will have to program it manually the first time.

starbasessd, 2020-05-04:
No local sd card. Using net boot
-----
pi@raspi1:~ $ cd /
pi@raspi1:/ $ sudo find -name *.upd
find: ‘./run/user/1000/gvfs’: Permission deniedpi@raspi1:/
pi@raspi1:/ $
-----
pi@raspi1:/ $ sudo find -name pieeprom.bin
find: ‘./run/user/1000/gvfs’: Permission denied
./home/pi/pieeprom.bin
-----
Only pieeprom.bin is the one I modified. No pieeprom.upd
-----
pi@raspi1:/ $ vcgencmd bootloader_config
[all]
BOOT_UART=0
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0
DHCP_TIMEOUT=45000
DHCP_REQ_TIMEOUT=4000
TFTP_FILE_TIMEOUT=30000
TFTP_IP=
TFTP_PREFIX=0
BOOT_ORDER=0x21
SD_BOOT_MAX_RETRIES=3
NET_BOOT_MAX_RETRIES=5
[none]
FREEZE_VERSION=0
-----
updated bootconf.txt from extracted version
-----
pi@raspi1:~ $ cat bootconf.txt
[all]
BOOT_UART=0
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0
DHCP_TIMEOUT=45000
DHCP_REQ_TIMEOUT=4000
TFTP_FILE_TIMEOUT=30000
TFTP_IP=
TFTP_PREFIX=0
BOOT_ORDER=0x21
SD_BOOT_MAX_RETRIES=3
NET_BOOT_MAX_RETRIES=5
SELF_UPDATE=1
[none]
FREEZE_VERSION=0
-----
command run to re-apply it:
pi@raspi1:~ $ rpi-eeprom-config --out pieeprom-netboot.bin --config
bootconf.txt pieeprom.bin ## copied/pasted from instructions
command to install it:
pi@raspi1:~ $ sudo rpi-eeprom-update -d -f ./pieeprom-netboot.bin  ##
copied/pasted from instructions
------
Are you saying that the instructions are wrong, that the pieeprom.bin
should be deleted prior to installing pieeprom-netboot.bin, or rename
pieeprom-netboot.bin should be renamed pieeprom.bin?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

timg236, 2020-05-04:
SELF_UPDATE is read from the "current" EEPROM image not the one in the TFTP boot directory so when you reboot the bootloader will see that SELF_UPDATE is not enabled and not look for an update.
i.e. SELF_UPDATE only applies to future updates not the one where you are requesting self-update. You have to use an SD-CARD for that.

starbasessd, 2020-05-04:
I am confused:
https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711_bootloader_config.md
-----
SELF_UPDATE
Allows the bootloader to update itself instead of requiring recovery.bin.
This is intended to make it easier to update the bootloader firmware via
network boot. To enable set SELF_UPDATE=1 and add bootloader_update=1 in
config.txt. N.B. There is no automatic rollback in the event of a power
failure during the firmware update. In the unlikely event of this happening
you will have to use Pi Imager to apply the rescue image.

If self update is enabled then the bootloader will look for (pieeprom.upd +
pieeprom.sig) and/or (vl805.bin + vl805.sig) on the boot partition (or TFTP
root). If the update files are different to the current image then the
update is applied and system is reset. Otherwise, if the images are
identical then boot continues as normal.

Since the updates files are in the same format as generated by
rpi-eeprom-update you can use rpi-eeprom-update to install the files to
/boot so long as /boot mounts the approprate boot device / tftp-root (via
NFS).

Default: 0
Version: stable/pieeprom-2020-03-19.bin
--------
Since currently I have to access the Pi4 and manipulate it with an SDCard,
I thought this was to allow the system to update itself while net booted.
I understood that vcgencmd bootloader_config reads the info from the
version in SPI memory, is this not correct?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

timg236, 2020-05-04:
To enable set SELF_UPDATE=1 and add bootloader_update=1 in config.txt

You haven't enabled SELF_UPDATE. Use an SD card to update the bootloader. Then verify using "vcgencmd bootloader_config" that you have enabled it. Only then can self-update be used. Closing bug as invalid, the forums are the best place for support threads.

closed 2020-05-04 by timg236

starbasessd, 2020-05-04:
Thanks. BTW I _AM_ booting from SD Card, doing the extractions, editing and applying the change from the SD CARD
I cannot get the SELF_UPDATE=1 to apply to the bootloader and stick.
Notice that the BOOT_ORDER setting does apply and stick.
Oh, well.

lurch, 2020-05-04:
If you've changed the boot-order, maybe it's still (successfully) booting from network and ignoring the modified EEPROM config on the SD card? Maybe temporarily unplugging the network cable will allow you to "force" an SD card boot? :shrug:

starbasessd, 2020-05-04:
BTDT. When booting from SD Card, I always disconnect ethernet cable after
running sudo halt just to prevent such issues.
Tim said go away, so I am.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

timg236, 2020-05-04:
You're welcome to continue discussing things on closed bugs, however, for the reasons above this doesn't appear to be a bug so I'm closing it. 

For unclear / support questions you are more likely to get a response on the forums and if you have something more concrete then that's the time to re-open / raise a new bug.

lurch, 2020-06-01:
I wonder if https://github.com/raspberrypi/documentation/pull/1580 (i.e. the option being mis-documented) was the cause of this problem? :shrug:

starbasessd, 2020-06-01:
> I wonder if raspberrypi/documentation#1580

Thanks for noticing this, I appreciate it.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?