raspberrypi/rpi-eeprom #123: rpi-eeprom-update with USE_FLASHROM=1 fails if SPI modules are already loaded.

trejan, 2020-05-11:
[rpi-eeprom-update tries to modprobe spi-bcm2835 after loading the spi-gpio40-45 overlay.](spi-gpio40-45) If spi-bcm2835 is already loaded then modprobe will do nothing and the SPI pins won't change. flashrom will try to reflash whatever is connected to the SPI pins on the GPIO header.

Unloading spi-bcm2835 will be difficult so it is probably best to tell the user to disable SPI themselves to do the update.

timg236, 2020-05-15:
The current release of the bootloader supports self-update and that will be enabled by default in the next release so that bootloader updates work with USB mass storage and network boot.

The USE_FLASHROM script code will then just be deleted.

trejan, 2020-05-25:
Another issue related to this that needs documentation. If you're on the 5.4 kernel and have audio enabled in /boot/config.txt then pinctrl will refuse to let spi-bcm2835 take over the GPIOs. The 4.19 pinctrl doesn't complain and works.

`5.4`

```
[   66.684136] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /scb/mailbox@7e00b840/bcm2835_audio/status
[   66.730318] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/spi@7e204000/cs-gpios
[   66.730329] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/spi@7e204000/status
[   66.730337] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/gpio@7e200000/spi0_cs_pins/brcm,pins
[   66.730344] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/gpio@7e200000/spi0_cs_pins/brcm,function
[   66.730351] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/gpio@7e200000/spi0_cs_pins/status
[   66.730360] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/gpio@7e200000/spi0_pins/brcm,pins
[   66.730367] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/gpio@7e200000/spi0_pins/brcm,function
[   66.730374] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/gpio@7e200000/spi0_pins/status
[   66.738433] pinctrl-bcm2835 fe200000.gpio: pin gpio40 already requested by bcm2835_audio; cannot claim for fe204000.spi
[   66.738443] pinctrl-bcm2835 fe200000.gpio: pin-40 (fe204000.spi) status -22
[   66.738451] pinctrl-bcm2835 fe200000.gpio: could not request pin 40 (gpio40) from group gpio40  on device pinctrl-bcm2835
[   66.738456] spi-bcm2835 fe204000.spi: Error applying setting, reverse things back
[   66.738477] spi-bcm2835: probe of fe204000.spi failed with error -22
```

**4.19**

```
[   66.322773] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/audio/status
[   66.372362] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/spi@7e204000/cs-gpios
[   66.372372] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/spi@7e204000/status
[   66.372379] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/gpio@7e200000/spi0_cs_pins/brcm,pins
[   66.372386] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/gpio@7e200000/spi0_cs_pins/brcm,function
[   66.372391] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/gpio@7e200000/spi0_cs_pins/status
[   66.372399] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/gpio@7e200000/spi0_pins/brcm,pins
[   66.372406] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/gpio@7e200000/spi0_pins/brcm,function
[   66.372411] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/gpio@7e200000/spi0_pins/status
[   67.258947] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /soc/audio/status
[   67.260975] bcm2835_audio soc:audio: card created with 8 channels

```

timg236, 2020-05-26:
Closing issue because USE_FLASHROM is going to be removed so I'm not spending time on that area. Device tree issues are best discussed on the Linux repo

closed 2020-05-26 by timg236