raspberrypi/rpi-eeprom #131: USB MSD timeout message - incorrect units

andrum99, 2020-05-21:
**Describe the bug**
When testing USB MSD boot, I received some messages like this:

`USB MSD timed out after 10000 seconds`

The timeout appears to actually be 10000 milliseconds, i.e. 10 seconds.

Originally reported in the forums at https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=274595#p1663699

**To Reproduce**
Steps to reproduce the behavior:

Boot Pi 4 with no boot devices attached, and USB MSD boot enable in the bootloader config (e.g. BOOT_ORDER=0xf41).

**Expected behaviour**
seconds -> milliseconds

**Screenshots**
[image]

**Bootloader version and configuration**

pi@pi4b2hdd:~ $ vcgencmd bootloader_version
May 15 2020 11:05:52
version 23a9f59b85f5a81bb2eec455e064ef9905216322 (release)
timestamp 1589537152

and

pi@pi4b2hdd:~ $ vcgencmd bootloader_version
May 21 2020 18:02:07
version ae867e96f2d7d0613dfb7e21b98b7cf99a606efc (release)
timestamp 1590080527

Config is unmodified in both cases.

timg236, 2020-05-22:
Thanks, fixed, I'll include that in an update sometime next week

timg236, 2020-05-26:
This should be fixed by e87bc4da8d983aa6e7c3edcf8cd823b76e50e65e

The binaries are available in this release which can be flashed using the Raspberry Pi Imager
https://github.com/raspberrypi/rpi-eeprom/releases/tag/v2020.05.26-137ad

andrum99, 2020-05-26:
> This should be fixed by [e87bc4d](https://github.com/raspberrypi/rpi-eeprom/commit/e87bc4da8d983aa6e7c3edcf8cd823b76e50e65e)
> 
> The binaries are available in this release which can be flashed using the Raspberry Pi Imager
> https://github.com/raspberrypi/rpi-eeprom/releases/tag/v2020.05.26-137ad

Confirmed - LGTM.

closed 2020-05-26 by timg236