raspberrypi/rpi-eeprom #134: VL805 readback failure in the bootloader

timg236, 2020-05-27:
**Describe the bug**
On some boards, the VL805 self or recovery.bin update fails. This seems to be due to the read of the VLI SPI EEPROM failing causing the bootloader to think an update is required.

**To Reproduce**
Use the Raspberry Pi Imager to create an EEPROM recovery image and observe the UART output.

**Additional context**
Forum post - logs https://www.raspberrypi.org/forums/viewtopic.php?p=1667830#p1667830

https://pastebin.com/dicQb0b0
https://pastebin.com/n1rGrJM8
https://pastebin.com/2YcP69kL

alkersan, 2020-05-27:
Thank you @timg236 for moving the issue here. 

Here is a simpler way to reproduce the infinite reboot cycle, just using sdcard boot and the latest firmware (the one where SELF_UPDATE is enabled by default).

1. Flash the EEPROM (checked 2020-05-27 and 2020-05-26). The only change in the conf is `BOOT_UART=1`, to get the logs. Everything else is left by default:
```
[all]
BOOT_UART=1
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0
DHCP_TIMEOUT=45000
DHCP_REQ_TIMEOUT=4000
TFTP_FILE_TIMEOUT=30000
ENABLE_SELF_UPDATE=1
DISABLE_HDMI=0
SD_BOOT_MAX_RETRIES=1
USB_MSD_BOOT_MAX_RETRIES=1
BOOT_ORDER=0xf41
```

2. Prepare fresh raspbian sdcard using rpi-imager. Append following parameters to the end of `config.txt`:
```
enable_uart=1
bootloader_update=1
```

3. Copy the `vl805.bin` and `vl805.sig` files to the raspbian boot directory (pieeprom files aren't event needed).
Now we're in the state when during next power cycle the bootloader should check the vl805 files and decide whether or not to flash them. This is the sole purpose of SELF_UPDATE, right?

3. Boot the RPi. It goes into infinite reset cycle. Here are the serial logs:
- [url=https://pastebin.com/raw/jpd6rCeZ]sdcard_boot_2020-05-27_eeprom_vl805_present_on_boot.log[/url]

alkersan, 2020-05-27:
Interestingly - if instead of sdcard use USB thunmb drive - the vl805.bin is correctly identified as unchanged. Serial output is [here](https://pastebin.com/raw/b8aKWUe2).

timg236, 2020-05-27:
Thanks, I managed to reproduce it, probably some sort of dumb PCIe init ordering regression which I missed due to focussing on making sure that this worked with USB boot!

timg236, 2020-05-27:
This should fix self-update. I think the beta recovery.bin needs a patch as well I'll kick off a build

[pieeprom-vl805-self-update-fix.zip](https://github.com/raspberrypi/rpi-eeprom/files/4690670/pieeprom-vl805-self-update-fix.zip)

timg236, 2020-05-27:
This works for me now via SD or USB boot, found a small issue with the previous attempt where the VLI ROM had not quite finished running before the SPI read.

https://github.com/raspberrypi/rpi-eeprom/releases/tag/v2020.05.27-137ad

alkersan, 2020-05-27:
Tested and can also confirm that now vl805 update works as expected:
- [recovery.bin update](https://pastebin.com/raw/zyw9vTzG)
- [sdcard self update](https://pastebin.com/raw/aqxQvECb)
- [USB self update](https://pastebin.com/raw/1JkrH7Dk)
- [TFTP self update](https://pastebin.com/raw/pu3wiMND)

In all cases no flash is happening, bootloader detects presence of eeprom.bin and vl805.big blobs, skips update and proceeds to kernel boot.

timg236, 2020-05-28:
Thanks for verifying @alkersan 

The Github binaries didn't get pushed properly but the rescue release was fine. I created a new version today to make it clearer. Closing bug, hopefully, this is all ok now!

closed 2020-05-28 by timg236