raspberrypi/rpi-eeprom #139: Seagate 5Gb HDD backup drive - unable to boot from

vintozver, 2020-05-30:
**Describe the bug**
Unable to boot up from the Seagate USB 3.0 5Gb Backup Plus HDD.

**To Reproduce**
Partition the disk (GPT), add one FAT32 partition, copy working boot partition from the SD card, replace the *.bin & *.dat files with USB MSD support.

$ sudo parted /dev/sda
GNU Parted 3.2
Using /dev/sda
Welcome to GNU Parted! Type 'help' to view a list of commands.
(parted) print
Model: Seagate BUP BK (scsi)
Disk /dev/sda: 5001GB
Sector size (logical/physical): 512B/4096B
Partition Table: gpt
Disk Flags:

Number  Start   End    Size   File system  Name  Flags
 1      17.4kB  271MB  271MB  fat32        boot  boot, esp`

**Expected behaviour**
Boot of the kernel, even with panic that root partition is not found

**Bootloader version and configuration**
2020 05 28

**Additional context**
On the boot screen: the last line is xHC-err CMD failed -3
for a moment
then screen goes blank

NetConsole logs with XHCI_DEBUG=0xf

SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SD HOST: 250000000 CTL0: 0x00000f00 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
EMMC
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SDV1
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SD CMD: 0x371a0010 (55) 0x0 0x1fff0001
Boot mode: USB-MSD (04) order f
XHCI-STOP
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
xHC ports 5 slots 32 intrs 4
USB3 rport 2 status 00000000 -> 000002a0
USB3 rport 3 status 00000000 -> 000002a0
USB3 rport 4 status 00000000 -> 000002a0
USB3 rport 5 status 00000000 -> 000002a0
USB2 rport 1 status 00000000 -> 400002e1
USB2 rport 1 status 40000331 -> 40200e03
REQ [01:00] 0.00 000000:01 ep 0 128,6,256,0 18 @ 0x00000012
REQ [01:00] 2.16 000000:01 ep 0 128,6,512,0 9 @ 0x00000009
REQ [01:00] 2.16 000000:01 ep 0 128,6,512,0 25 @ 0x00000019
DEV [01:00] 2.16 000000:01 class 9 VID 2109 PID 3431
HUB init [01:00] 2.16 000000:01
REQ [01:00] 2.16 000000:01 ep 0 160,6,10496,0 7 @ 0x00000007
REQ [01:00] 2.16 000000:01 ep 0 128,6,1536,0 10 @ 0x0000000a
REQ [01:00] 2.16 000000:01 ep 0 0,9,1,0 0 @ 0x00000000
REQ [01:00] 2.16 000000:01 ep 0 35,3,8,1 0 @ 0x00000000
REQ [01:00] 2.16 000000:01 ep 0 35,3,8,2 0 @ 0x00000000
REQ [01:00] 2.16 000000:01 ep 0 35,3,8,3 0 @ 0x00000000
REQ [01:00] 2.16 000000:01 ep 0 35,3,8,4 0 @ 0x00000000
REQ [01:00] 2.16 000000:01 ep 0 163,0,0,1 4 @ 0x00000004
REQ [01:00] 2.16 000000:01 ep 0 163,0,0,2 4 @ 0x00000004
REQ [01:00] 2.16 000000:01 ep 0 163,0,0,3 4 @ 0x00000004
REQ [01:00] 2.16 000000:01 ep 0 163,0,0,4 4 @ 0x00000004
REQ [01:00] 2.16 000000:01 ep 0 35,1,16,4 0 @ 0x00000000
REQ [01:00] 2.16 000000:01 ep 0 35,3,4,4 0 @ 0x00000000
USB3 rport 3 status 000002a0 -> 000002e0
[... 359 lines cut ...]
REQ [01:00] 2.16 000000:01 ep 0 163,0,0,1 4 @ 0x00000004
REQ [01:00] 2.16 000000:01 ep 0 163,0,0,2 4 @ 0x00000004
MSD [03:00] 3.00 000000:03 READ lun 0 addr: 8222 count: -8158
CBW 00000045 ffc04400 fff8f074 00000000
REQ [01:00] 2.16 000000:01 ep 0 163,0,0,3 4 @ 0x00000004
REQ [01:00] 2.16 000000:01 ep 0 163,0,0,4 4 @ 0x00000004
   EVT (32) c0 65 f9 3f 00 00 00 00 0d 00 00 03 00 80 03 03
   CMD (1) 88 f0 f8 3f 00 00 00 00 0d 00 00 00 21 04 00 00
SLOT IN
00 00 40 18 00 00 03 00
00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00
SLOT OUT
00 00 40 20 00 00 03 00
00 00 00 00 03 00 00 18
00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00
EP0 CTX
01 00 00 00 26 00 00 02
01 94 f4 3f 00 00 00 00
08 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00
Read autoboot.txt failed
PM_RSTS: 0x00001000

timg236, 2020-06-01:
I presume you mean 5TB not 5GB. Drives larger than 2TB aren't really supported although if the ESD is within the first 2TB is should work.

It would be useful to get a dump of the entire GPT area or at least the first 128 sectors in order to determine why count is negative

MSD [03:00] 3.00 000000:03 READ lun 0 addr: 34 count: 1
CBW 00000044 00000200 fff8f074 00000000
MSD [03:00] 3.00 000000:03 CSW 53425355 00000044 00000000 00000000
REQ [01:00] 2.16 000000:01 ep 0 163,0,0,1 4 @ 0x00000004
REQ [01:00] 2.16 000000:01 ep 0 163,0,0,2 4 @ 0x00000004
MSD [03:00] 3.00 000000:03 READ lun 0 addr: 8222 count: -8158

vintozver, 2020-06-01:
# dd if=/dev/sda of=disk_hdr.img count=130
[disk_hdr.zip](https://github.com/raspberrypi/rpi-eeprom/files/4712954/disk_hdr.zip)

# sudo gdisk  /dev/sda
GPT fdisk (gdisk) version 1.0.3

Partition table scan:
  MBR: protective
  BSD: not present
  APM: not present
  GPT: present

Found valid GPT with protective MBR; using GPT.

Disk /dev/sda: 9767541167 sectors, 4.5 TiB
Model: BUP BK
Sector size (logical/physical): 512/4096 bytes
Disk identifier (GUID): 29468380-BD6B-4BAC-AF08-DA8DBC3B0EC5
Partition table holds up to 128 entries
Main partition table begins at sector 2 and ends at sector 33
First usable sector is 34, last usable sector is 9767541133
Partitions will be aligned on 8-sector boundaries
Total free space is 685 sectors (342.5 KiB)

Number  Start (sector)    End (sector)  Size       Code  Name
   1              34          530144   258.8 MiB   EF00  boot
   2      5472573440      9767540735   2.0 TiB     0700  stuff
   3          530432       545789951   260.0 GiB   8E00  system
   4       545789952      5472573439   2.3 TiB     8E00  data

timg236, 2020-06-01:
Thanks, very useful! I found a problem in some SCSI read-ahead caching code which would explain the negative count in the debug logs. On my USB device it wasn't fatal but it if I'm correct then it would be reasonable for the drive to reject the command.

Please could you try the attached test bootloader to see if this helps?

[rpi-eeprom-recovery-3bbcda012.zip](https://github.com/raspberrypi/rpi-eeprom/files/4713198/rpi-eeprom-recovery-3bbcda012.zip)

vintozver, 2020-06-01:
just tried new attached bootloader. Explaining the results.

My 512G Kingston SSD worked with the previous, 2020-05-28 version before.
The attached bootloader boots fine when the PI has just powered up, but fails to boot after the OS reboot.

```
XHCI-STOP
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
Boot mode: RESTART (0f) order 0
Boot mode: SD (01) order f4
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SD HOST: 250000000 CTL0: 0x00000f00 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
EMMC
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SDV1
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SD CMD: 0x371a0010 (55) 0x0 0x1fff0001
Retry SD 1/1
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SD HOST: 250000000 CTL0: 0x00000f00 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
EMMC
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SDV1
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SD CMD: 0x371a0010 (55) 0x0 0x1fff0001
Boot mode: USB-MSD (04) order f
XHCI-STOP
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
USB3 rport 2 status 00000000 -> 000002a0
USB3 rport 3 status 00000000 -> 000002b1
USB3 rport 4 status 00000000 -> 000002a0
USB3 rport 5 status 00000000 -> 000002a0
USB2 rport 1 status 00000000 -> 400002e1
USB2 rport 1 status 40000331 -> 40200e03
REQ [01:00] 0.00 000000:01 ep 0 128,6,256,0 18 @ 0x00000012
REQ [01:00] 2.16 000000:01 ep 0 128,6,512,0 9 @ 0x00000009
REQ [01:00] 2.16 000000:01 ep 0 128,6,512,0 25 @ 0x00000019
USB3 rport 3 status 000002b1 -> 00281203
HUB init [01:00] 2.16 000000:01
REQ [01:00] 2.16 000000:01 ep 0 160,6,10496,0 7 @ 0x00000007
REQ [01:00] 2.16 000000:01 ep 0 128,6,1536,0 10 @ 0x0000000a
XHCI-STOP
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
Boot mode: RESTART (0f) order 0
Boot mode: SD (01) order f4
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SD HOST: 250000000 CTL0: 0x00000f00 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
EMMC
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SDV1
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SD CMD: 0x371a0010 (55) 0x0 0x1fff0001
Retry SD 1/1
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SD HOST: 250000000 CTL0: 0x00000f00 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
EMMC
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SDV1
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SD CMD: 0x371a0010 (55) 0x0 0x1fff0001
Boot mode: USB-MSD (04) order f
XHCI-STOP
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
xHC ports 5 slots 32 intrs 4
USB3 rport 2 status 00000000 -> 000002a0
USB3 rport 3 status 00000000 -> 000002b1
USB3 rport 4 status 00000000 -> 000002a0
USB3 rport 5 status 00000000 -> 000002a0
[... 461 lines cut ...]
CBW 0000004d 00000000 fff8f074 00000000
MSD [03:00] 3.00 000000:03 CSW 53425355 0000004d 00000000 00000001
MSD [03:00] 3.00 000000:03 lun 0 TEST_UNIT_READY
CBW 0000004e 00000000 fff8f074 00000000
REQ [01:00] 2.16 000000:01 ep 0 163,0,0,1 4 @ 0x00000004
MSD [03:00] 3.00 000000:03 CSW 53425355 0000004e 00000000 00000000
MSD [03:00] 3.00 000000:03 lun 0 READ_CAPACITY
CBW 0000004f 00000008 fff8f074 00000000
REQ [01:00] 2.16 000000:01 ep 0 163,0,0,2 4 @ 0x00000004
MSD [03:00] 3.00 000000:03 CSW 53425355 0000004f 00000000 00000000
REQ [01:00] 2.16 000000:01 ep 0 163,0,0,3 4 @ 0x00000004
REQ [01:00] 2.16 000000:01 ep 0 163,0,0,4 4 @ 0x00000004
MSD [03:00] 3.00 000000:03 READ lun 0 addr: 0 count: 4
CBW 00000050 00000800 fff8f074 00000000
MSD [03:00] 3.00 000000:03 CSW 53425355 00000050 00000000 00000000
MBR: 0x00000001,4294967295 type: 0xee
MBR: 0x00000000,       0 type: 0x00
MBR: 0x00000000,       0 type: 0x00
MBR: 0x00000000,       0 type: 0x00
GPT 808346296bbdac4baf08da8dbc3b0ec5 000000001 24630d9ae num-partitions 128 entry-size 128
GPT partition: 0 28732ac11ff8d211ba4b00a0c93ec93b 000000022..0000816e0
MSD [03:00] 3.00 000000:03 READ lun 0 addr: 34 count: -30
CBW 00000051 ffffc400 fff8f074 00000000
REQ [01:00] 2.16 000000:01 ep 0 163,0,0,1 4 @ 0x00000004
```

timg236, 2020-06-02:
I think the Kingston issue is different possible this problem https://github.com/raspberrypi/rpi-eeprom/issues/132 I think a powered hub if it is failing before address device. XHCI_DEBUG 0x1f might help here.

For the 5TB issue I think SCSI_READ_CAPACITY_10 must be returning zero if the drive has more than 2^32 sectors. XHCI_DEBUG 0xf would confirm that. 
Assuming that I've interpreted the logs correctly and that is the issue then the following fix should resolve that, at least in the bootloader. Unfortunately, I had to fake the SCSI responses because I don't have access to any large USB hard drives right now.

[rpi-eeprom-recovery-265d9482.zip](https://github.com/raspberrypi/rpi-eeprom/files/4716707/rpi-eeprom-recovery-265d9482.zip)

vintozver, 2020-06-02:
fix works (for 5Tb HDD boot)! able to boot to the kernel, initrd is loaded

tested on
`
$ vcgencmd bootloader_version
Jun  2 2020 12:06:35
version 265d9482081e3f2b37d3e2eb51f7040becf90d90 (release)
timestamp 1591095995

$ vcgencmd bootloader_config
[all]
BOOT_UART=0
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0
DHCP_TIMEOUT=45000
DHCP_REQ_TIMEOUT=4000
TFTP_FILE_TIMEOUT=30000
ENABLE_SELF_UPDATE=1
DISABLE_HDMI=0
SD_BOOT_MAX_RETRIES=1
USB_MSD_BOOT_MAX_RETRIES=3
USB_MSD_LUN_TIMEOUT=10000
USB_DEV_FILE_TIMEOUT=5000
BOOT_ORDER=0xf41
BOOT_UART=1
NETCONSOLE=6665@169.254.1.1/eth0,6666@/
XHCI_DEBUG=0xf
`

logs 
`
SD HOST: 250000000 CTL0: 0x00000f00 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
EMMC
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SDV1
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SD CMD: 0x371a0010 (55) 0x0 0x1fff0001
Retry SD 1/1
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SD HOST: 250000000 CTL0: 0x00000f00 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
EMMC
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SDV1
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SD CMD: 0x371a0010 (55) 0x0 0x1fff0001
Boot mode: USB-MSD (04) order f
XHCI-STOP
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
xHC ports 5 slots 32 intrs 4
USB3 rport 2 status 00000000 -> 000002a0
USB3 rport 3 status 00000000 -> 000002a0
USB3 rport 4 status 00000000 -> 000002a0
USB3 rport 5 status 00000000 -> 000002a0
USB2 rport 1 status 00000000 -> 400002e1
USB2 rport 1 status 400002f1 -> 40200e03
REQ [01:00] 0.00 000000:01 ep 0 128,6,256,0 18 @ 0x00000012
REQ [01:00] 2.16 000000:01 ep 0 128,6,512,0 9 @ 0x00000009
REQ [01:00] 2.16 000000:01 ep 0 128,6,512,0 25 @ 0x00000019
DEV [01:00] 2.16 000000:01 class 9 VID 2109 PID 3431
HUB init [01:00] 2.16 000000:01
REQ [01:00] 2.16 000000:01 ep 0 160,6,10496,0 7 @ 0x00000007
REQ [01:00] 2.16 000000:01 ep 0 128,6,1536,0 10 @ 0x0000000a
REQ [01:00] 2.16 000000:01 ep 0 0,9,1,0 0 @ 0x00000000
REQ [01:00] 2.16 000000:01 ep 0 35,3,8,1 0 @ 0x00000000
REQ [01:00] 2.16 000000:01 ep 0 35,3,8,2 0 @ 0x00000000
REQ [01:00] 2.16 000000:01 ep 0 35,3,8,3 0 @ 0x00000000
REQ [01:00] 2.16 000000:01 ep 0 35,3,8,4 0 @ 0x00000000
REQ [01:00] 2.16 000000:01 ep 0 163,0,0,1 4 @ 0x00000004
[... 1198 lines cut ...]
CBW 0000013e 00000800 fff8f074 00000000
MSD [03:00] 3.00 000000:03 CSW 53425355 0000013e 00000000 00000000
MSD [03:00] 3.00 000000:03 READ lun 0 addr: 224308 count: 4
CBW 0000013f 00000800 fff8f074 00000000
MSD [03:00] 3.00 000000:03 CSW 53425355 0000013f 00000000 00000000
MSD [03:00] 3.00 000000:03 READ lun 0 addr: 8981 count: 4
CBW 00000140 00000800 fff8f074 00000000
MSD [03:00] 3.00 000000:03 CSW 53425355 00000140 00000000 00000000
MSD [03:00] 3.00 000000:03 READ lun 0 addr: 71 count: 4
CBW 00000141 00000800 fff8f074 00000000
MSD [03:00] 3.00 000000:03 CSW 53425355 00000141 00000000 00000000
MSD [03:00] 3.00 000000:03 READ lun 0 addr: 8982 count: 6
CBW 00000142 00000c00 fff8f074 00000000
REQ [01:00] 2.16 000000:01 ep 0 163,0,0,1 4 @ 0x00000004
MSD [03:00] 3.00 000000:03 CSW 53425355 00000142 00000000 00000000
REQ [01:00] 2.16 000000:01 ep 0 163,0,0,2 4 @ 0x00000004
REQ [01:00] 2.16 000000:01 ep 0 163,0,0,3 4 @ 0x00000004
REQ [01:00] 2.16 000000:01 ep 0 163,0,0,4 4 @ 0x00000004
MSD [03:00] 3.00 000000:03 READ lun 0 addr: 72 count: 4
CBW 00000143 00000800 fff8f074 00000000
MSD [03:00] 3.00 000000:03 CSW 53425355 00000143 00000000 00000000
MSD [03:00] 3.00 000000:03 READ lun 0 addr: 8988 count: 10
CBW 00000144 00001400 fff8f074 00000000
MSD [03:00] 3.00 000000:03 CSW 53425355 00000144 00000000 00000000
`

vintozver, 2020-06-02:
Not able to get access to the Kingston 512G SATA SSD right now, had to re-purpose it urgently for another application. I will let you know when I get it back in my possession.

timg236, 2020-06-02:
Great news. Many thanks for the detailed bug report it makes things so much easier for us!

I'll close this once it's in APT.

Please raise a new bug report for the Kingston 512G SATA with full details of the model and even better the Amazon link. Once the set of "awkward" drives is clearer we can hopefully get them into the CI system.

carleberhart, 2020-06-02:
I am currently trying to boot a raspberry pi 4 on a cheap kingston 120 gig
ssd without much success.
On boot the eeprom is reported uptodate, but then it is reported that the
fixup4rc.dat is empty or missing
I attach a snapshot of the boot sequence.
Please advise.  Thank you
Carl Eberhart

carleberhart, 2020-06-02:
By the way, I can send a video of the entire boot cycle if needed

carleberhart, 2020-06-02:
One more thing.  The drive was purchased from Amazon last week.
Kingston 120GB A400 SATA 3 2.5" Internal SSD SA400S37/120G - HDD
Replacement for Increase Performance

vintozver, 2020-06-02:
please create a **new** report

update the bootloader with the latest version attached here (https://github.com/raspberrypi/rpi-eeprom/files/4716707/rpi-eeprom-recovery-265d9482.zip)

apply the config with XHCI_DEBUG=0x1f and use NETCONSOLE to collect the logs (https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711_bootloader_config.md)

attach the NETCONSOLE logs to the bug report

timg236, 2020-06-02:
Removed unrelated comments

timg236, 2020-06-04:
The fix is now available via apt in the rpi-eeprom package

closed 2020-06-04 by timg236