raspberrypi/rpi-eeprom #140: Throttle netboot attempts on failure

rgl, 2020-05-30:
**Describe the bug**

When the TFTP server is not correctly configured and does not (yet) have the expected file (e.g. start4.elf), the booloader (pieeprom-2020-04-16.bin) will loop as fast as it can trying to get the file.

**To Reproduce**

Configure the bootloader as:

```
BOOT_ORDER=0x2
NET_BOOT_MAX_RETRIES=-1
TFTP_PREFIX=2
```

**Expected behaviour**

It should throttle itself, or just sleep a given number of seconds, before retrying the whole booting process.

timg236, 2020-06-01:
The bootloader generally has to do DHCP and link negotiation between retries for network boot which is quite slow. Are you using a static IP address configuration?

timg236, 2020-06-01:
Actually, you might want to try the latest bootloader. This is implicitly throttled by the LED error pattern which is displayed after each failed boot mode so it will retry every few seconds instead of multiple times.

lurch, 2020-06-01:
> This is implicitly throttled by the LED error pattern which is displayed after each failed boot mode so it will retry every few seconds instead of multiple times.

Oooh, interesting. Is there an option to disable that for people who want faster switching between failing boot modes? (Yes, I realise the irony of this actually being the opposite of what the OP is requesting! No, I don't actually need this myself, I'm just playing devil's advocate.)

timg236, 2020-06-01:
Yes, make sure the first boot mode doesn't fail

timg236, 2020-06-02:
This has a 5-second delay between network boot retries

[rpi-eeprom-recovery-265d9482.zip](https://github.com/raspberrypi/rpi-eeprom/files/4716737/rpi-eeprom-recovery-265d9482.zip)

timg236, 2020-06-04:
APT update now contains this change

closed 2020-06-04 by timg236