raspberrypi/rpi-eeprom #141: Featurerequest RPi4 - extendable Firmwarefiles

derestle-htwg, 2020-06-02:
It seems that there is no big Interest to provide bootoptions through GPIO Headersin the EEPROM :-(

How about a hook in the firmware? The Firmware itself could stay closed, but a defined hook would be great.
4 Bytes in the Firmware and a little bit of documentation.
As an example:
For each firmwarefile is a ReadMe that states where in the Firmware is a function pointer and how long the Firmware is.

The User takes the Firmware image, appends his payload after the firmwareimage and modifies the functionpointer in the firmware. That gets flashed to the EEPROM.

When the Firmware boots it should have a look at the Fuses. If the Fuses say that it should be booted from the appended Payload then it looks a the the functionpointer. If the pointer is valid (e.g. not null, points in the payload) it jumps to the payload. 

Maybe, as an optional feature, you can provide a table with function pointers as payload with some basic stuff.

I know this is far away from John Doe standard user.
But if you don't let us boot from GPIO Headers, give us atleast the possibility to get somehow our own stuff running. Or does the community really have to reverse engineer again?

It will happen - the question is, do you give us a helping hand or do you give us a hard time?

Best regards

timg236, 2020-06-02:
The best place to discuss major new features like this is on the forums to build a case for how this would benefit the majority of Pi users. 
It's not something that I'm personally interested in

closed 2020-06-02 by timg236

lurch, 2020-06-02:
> Or does the community really have to reverse engineer again?
It will happen - the question is, do you give us a helping hand or do you give us a hard time?

I'm sure "the community" would prefer for Tim to continue working on USB-boot support, rather than adding SPI-boot firmware-options that would only be of interest to a small minority of users?? :man_shrugging:

derestle-htwg, 2020-06-03:
@lurch , this is about a Firmwarehook, SPI-Boot is in another thread.

@timg236 this will never benefit a majority. Embedded developers are indeed a minority of the RPI-Users.

Well, of course USB-Boot befefits way more users, is also great and i'd love to use it for my personal projects.

And i don't want that Firmwarehook to be done immediatelly. As a Firmware developer myself i am aware  that even small things like that took take their time to be developed and tested.

But instead of giving it a low priority it is closed. That tells me that there is absolutely no interest in ever doing such.