raspberrypi/rpi-eeprom #142: Questions about rpi-eeprom-update options

lurch, 2020-06-02:
Just a couple of minor things... looking at the option descriptions here https://github.com/raspberrypi/rpi-eeprom/blob/master/rpi-eeprom-update#L354-L368

1. Now that USE_FLASHROM is deprecated, is the warning about "This command should only be run from console mode" still relevant?

1. Reading the options, I thought that `sudo rpi-eeprom-update -d` might overwrite my boot-config-options with the default ones, but as my EEPROM was already up to date (and I didn't use `-f`) then it didn't do anything, and my boot-config-options didn't change. If this _is_ the intended behaviour, I wonder if it's worth clarifying? :man_shrugging:

timg236, 2020-06-03:
Yes the warning is still relevant because the behaviour hasn't changed.

-d and -f are independent, feel free to raise a PR to improve the inline help for -d

lurch, 2020-06-03:
> Yes the warning is still relevant because the behaviour hasn't changed.

Hmm? On e.g. https://github.com/raspberrypi/documentation/blob/master/hardware/raspberrypi/bcm2711_bootloader_config.md it tells the user to run `sudo rpi-eeprom-update -d -f /lib/firmware/raspberrypi/bootloader/beta/pieeprom-2020-05-15.bin` and then reboot. AFAICT, this just copies the eeprom update files to the `/boot` partition, so why is the "WARNING: This command should only be run from console mode in order to avoid conflicts/deadlock with dtoverlay/dtparam settings." relevant? :confused:

timg236, 2020-06-03:
The warning is specific to USE_FLASHROM so could be moved to that or just deleted

andrum99, 2020-06-03:
I've filed #143 to clarify the -d flag for `rpi-eeprom-update`.

lurch, 2020-06-04:
Both issues now resolved - thanks @andrum99 and @timg236

closed 2020-06-04 by lurch