raspberrypi/rpi-eeprom #147: Yellow led are blinking constantly when I boot from USB Mass Storage

ridgarou, 2020-06-04:
**Describe the bug**
Recently I adquired a RPi4 with 8Gb and I configure it to boot from an SSD disk ina HDD external Case plugged to the Pi through a USB 3.0 port

The problem is that the Yellow led are blinking constantly and I found several process (called _kworker_) consuming a minimal amoung of CPU
[image]

I solved "the problem" inserting an microSD card in the slot. In this moment, the yellow led dont blink and the amount os CPU consumed by this process are 0.0% (but I suspect they keep running)
[image]

Is like the are constantly trying read a microSD card.

I have been search through Internet and the only thing similar I found are this: https://www.raspberrypi.org/forums/viewtopic.php?t=222352 but this was two years ago

There are no fix for this annoying issue??

Thank you very much in advance :)

**To Reproduce**
1. I had installed the last version of _Raspberry Pi OS (32-bit) Lite (Version: May 2020 - Release date: 2020-05-27 - Kernel version: 4.19)_
2. I updated the firmware with `sudo rpi-update` to `Raspbian GNU/Linux 10 (buster) 5.4.44-v7l+`
3. I update the bootloader with the last beta version: `Wed May 27 17:47:29 UTC 2020 (1590601649)`
4. I updated the `cmdline.txt` file adding `usb-storage.quirks=152d:0578:u` because de system dont recognice my SSD usb case as USB Mass Storage

**Bootloader version and configuration**
>  **$ sudo rpi-eeprom-update**
> BCM2711 detected
> VL805 firmware in bootloader EEPROM
> BOOTLOADER: up-to-date
> CURRENT: Wed May 27 17:47:29 UTC 2020 (1590601649)
>  LATEST: Wed May 27 17:47:29 UTC 2020 (1590601649)
>  FW DIR: /lib/firmware/raspberrypi/bootloader/beta
> VL805: up-to-date
> CURRENT: 000137ad
>  LATEST: 000137ad

> **$ vcgencmd bootloader_version**
> May 27 2020 18:47:29
> version d648db3968cd31d4948341e09cb8a925c49d2ea1 (release)
> timestamp 1590601649

> **$ vcgencmd bootloader_config**
> [all]
> BOOT_UART=0
> WAKE_ON_GPIO=1
> POWER_OFF_ON_HALT=0
> DHCP_TIMEOUT=45000
> DHCP_REQ_TIMEOUT=4000
> TFTP_FILE_TIMEOUT=30000
> ENABLE_SELF_UPDATE=1
> DISABLE_HDMI=0
> SD_BOOT_MAX_RETRIES=1
> USB_MSD_BOOT_MAX_RETRIES=1
> BOOT_ORDER=0xf41

**USB boot (please complete the following information):**
> **$ cat /boot/cmdline.txt**
> usb-storage.quirks=152d:0578:u console=serial0,115200 console=tty1 root=PARTUUID=3ec167a4-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait

> **$ sudo lsusb -vvv**
> Bus 002 Device 002: ID 152d:0578 JMicron Technology Corp. / JMicron USA Technology Corp. JMS567 SATA 6Gb/s brid                                              ge
> Device Descriptor:
>   bLength                18
>   bDescriptorType         1
>   bcdUSB               3.00
>   bDeviceClass            0
>   bDeviceSubClass         0
>   bDeviceProtocol         0
>   bMaxPacketSize0         9
>   idVendor           0x152d JMicron Technology Corp. / JMicron USA Technology Corp.
>   idProduct          0x0578 JMS567 SATA 6Gb/s bridge
[... 181 lines cut ...]
> [    1.181541] usb usb2: New USB device strings: Mfr=3, Product=2, SerialNumber=1
> [    1.181573] usb usb2: Product: xHCI Host Controller
> [    1.181599] usb usb2: Manufacturer: Linux 5.4.44-v7l+ xhci-hcd
> [    1.181625] usb usb2: SerialNumber: 0000:01:00.0
> [    1.184409] usbcore: registered new interface driver uas
> [    1.184581] usbcore: registered new interface driver usb-storage
> [    1.194211] usbcore: registered new interface driver usbhid
> [    1.194236] usbhid: USB HID core driver
> [    1.544893] usb 2-2: new SuperSpeed Gen 1 USB device number 2 using xhci_hcd
> [    1.576210] usb 2-2: New USB device found, idVendor=152d, idProduct=0578, bcdDevice= 5.08
> [    1.576248] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> [    1.576277] usb 2-2: Product: USB to ATA/ATAPI Bridge
> [    1.576304] usb 2-2: Manufacturer: JMicron
> [    1.576329] usb 2-2: SerialNumber: 0123456789ABCDEF
> [    1.579663] usb 2-2: UAS is blacklisted for this device, using usb-storage instead
> [    1.579794] usb 2-2: UAS is blacklisted for this device, using usb-storage instead
> [    1.579829] usb-storage 2-2:1.0: USB Mass Storage device detected
> [    1.580273] usb-storage 2-2:1.0: Quirks match for vid 152d pid 0578: 1800000
> [    1.580445] scsi host0: usb-storage 2-2:1.0
> [    1.724511] usb 1-1: new high-speed USB device number 2 using xhci_hcd
> [    1.907169] usb 1-1: New USB device found, idVendor=2109, idProduct=3431, bcdDevice= 4.21
> [    1.907206] usb 1-1: New USB device strings: Mfr=0, Product=1, SerialNumber=0
> [    1.907235] usb 1-1: Product: USB2.0 Hub
> [    6.151975] usbcore: registered new interface driver brcmfmac
>

lurch, 2020-06-04:
ping @XECDesign in case this is a "Linux thing" rather than a "bootloader thing" ? :wink:

ghollingworth, 2020-06-04:
Yes, the SD driver occasionally checks the SD card to see if you've inserted an SD card, at the same time time by default the /sys/class/leds/led0/trigger is set to mmc0 so every time it checks the LED blinks...

This is by design...

lurch, 2020-06-04:
https://www.google.com/search?q=led+trigger+linux+sysfs might provide enough info for you to change the trigger so that the LED doesn't blink every time it checks the SD card? :slightly_smiling_face:

timg236, 2020-06-04:
"The problem is that the Yellow led are blinking constantly and I found several process (called kworker) consuming a minimal amoung of CPU"
Closing because Linux behaviour is not under the control of the bootloader

closed 2020-06-04 by timg236

lurch, 2020-06-05:
@ridgarou Perhaps https://github.com/raspberrypi/linux/issues/3286 is useful too?

fjgsanandres, 2020-06-08:
very useful! Thanks a lot! 😄