raspberrypi/rpi-eeprom #149: [documentation needed] Will bootloader config survive eeprom update?

Harvie, 2020-06-05:
Hello,
i understood that in order to change bootloader settings i have to patch the eeprom image (using `rpi-eeprom-config`) before flashing it into the eeprom (using `rpi-eeprom-update`).

Let's say that i've changed `POWER_OFF_ON_HALT` value.
How do i make sure that it will not get overwritten once new stable eeprom image is released? I want to get latest eeprom updates, but also be sure that i will not lose my configuration during automatic update. Is the configuration extracted from current eeprom and automaticaly patched into the downloaded update before flashing eeprom during automatic update?

I wasn't able to find this information anywhere. I believe this should be made clear at minimum on these two pages:

https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711_bootloader_config.md
https://www.raspberrypi.org/documentation/hardware/raspberrypi/booteeprom.md

timg236, 2020-06-05:
Yes, the settings are migrated. This should be raised as documentation PR

lurch, 2020-06-05:
ping @JamesH65

JamesH65, 2020-06-05:
So, in order to get this change right, an I right in thinking that bootloader updates are now all done via apt? Or is it still all done via installing rpi-eeprom? Or are both supported, and both will automagically migrate settings from the previous bootloader?

lurch, 2020-06-05:
Is `rpi-eeprom` always installed by default @XECDesign , even in the "Lite" OS images?

timg236, 2020-06-05:
Nothing has changed. rpi-eeprom is an APT package installed in all OS images. At startup the systemd services runs and updates the EEPROMs if necessary migrating the previous images. The EEPROM images are under APT under /lib/firmware/raspberrypi and new versions only appear when the package is updated i.e. no side-channel online updater.

If you use the Raspberry Pi Imager the RECOVERY image applies the factory default settings.

JamesH65, 2020-06-05:
So we could remove the references in the docs to having to install rpi-eeprom now I guess. See https://github.com/raspberrypi/documentation/blob/JamesH65-patch-1%3Dboot-config-migration/hardware/raspberrypi/booteeprom.md, Updating the bootloader section

timg236, 2020-06-05:
Yes, I think we just want dist-upgrade or full-upgrade to get the latest package.

lurch, 2020-06-05:
Also, something which I mentioned at the time, but I guess must have slipped through the net:
https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711_bootloader_config.md has a part saying
```
# Check the current version
sudo rpi-eeprom-update
# Update to latest
sudo rpi-eeprom-update
```
which IMHO doesn't look right? Looking at the rpi-eeprom-update help text, should the latter one have been `sudo rpi-eeprom-update -a` ?

timg236, 2020-06-05:
Yes the -a is missing

lurch, 2020-06-07:
This is being addressed in https://github.com/raspberrypi/documentation/pull/1585

timg236, 2020-07-02:
Docs PR merged closing issue

closed 2020-07-02 by timg236