raspberrypi/rpi-eeprom #151: Bootloader can't boot via USB-HDD after system reboot

wildrun0, 2020-06-06:
After rebooting (sudo reboot), im getting stuck in bootloader with this loop:
```
Failed to open device: 'sdcard' (cmd 371a0010 status 1fff0001)
Failed to open device: 'sdcard' (cmd 371a0010 status 1fff0001)
root HUB port 1 init
root HUB port 3 init
USB MSD timed out after 20 seconds
```
If at this moment i replug usb-hdd, it will boot correctly, And sometimes it says that: 'SD: card detected' but there isnt no sd card in rpi.

Im using latest bootloader version at this moment(jun 3 2020 release).

timg236, 2020-06-06:
The bug report doesn't provide any of the information in the template so ignoring for now

wildrun0, 2020-06-06:
> The bug report doesn't provide any of the information in the template so ignoring for now

My fault, sorry. This information is what you needed?

timg236, 2020-06-07:
It doesn't say which hard drive you are using. A few other people are successfully using JPMicron SATA adapters. It might be worth trying a powered hub as well.
If you are able to collect XHCI_DEBUG logs that would be very useful, although, we'll also continue to collect details of USB devices where failure occurs so that we can try to get hold of a representative and do some interop testing.

wildrun0, 2020-06-07:
> XHCI_DEBUG

I'm using Seagate ST9320325AS connected via sata to usb3 adapter(there's 2 cables, one for data and second for additional power supply)
And what do exactly i need to get from XHCI_DEBUG? 0x3 logs?

timg236, 2020-06-08:
XHCI_DEBUG is documented here,
https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711_bootloader_config.md

The UART or NETCONSOLE trace would determine whether the USB port shows the connected state or not after a reboot. Ideally, it would be set to 0xff for full traces, which is very verbose so a one off trace log.

Before, that, if you could try the following bootloader which forces a reset of port power at reboot that would be useful. It's just a guess so really I just want to know if there is any observable change in behaviour.

[rpi-eeprom-recovery-c1325cd20.zip](https://github.com/raspberrypi/rpi-eeprom/files/4745181/rpi-eeprom-recovery-c1325cd20.zip)

wildrun0, 2020-06-08:
> [rpi-eeprom-recovery-c1325cd20.zip](https://github.com/raspberrypi/rpi-eeprom/files/4745181/rpi-eeprom-recovery-c1325cd20.zip)

I've tried this bootloader and it works fine. Thanks! Now Pi successfully boots after rebooting.
Also i've tried to connect second power usb(as i said above, i use sata to usb3 adapter w/ 2 cables) to power supply (just for test) and it couldn't boot (same issue)

lurch, 2020-06-08:
I guess there's no way for the external power-supply to know that it's supposed to reset port power?

timg236, 2020-06-08:
The external power supply is not under the control of the USB host controller (it's invisible) and I don't think external power supplies know about the USB state. 
The disadvantage of toggling the power at boot is that it makes other devices (e.g. external raid controllers) much slower to boot which are fine after a reboot.
@p33m Does this seem familiar?

P33M, 2020-06-08:
I am assuming that the "two cables" statement means one with the data pairs, and one with extra power wires. If both the cables are plugged into the Pi, then the device gets power cycled across a reboot. It could be getting upset because the power cycle isn't long enough - is there a configurable wait parameter for pausing prior to attempting MSD boot?

timg236, 2020-06-08:
In the attached patch it's a non-configurable 100ms in the off state. The default bootloader does not toggle port power. What's not clear is why a device works from a cold reboot but needs the port power to be toggled after a soft reset.

P33M, 2020-06-08:
Does the assertion of PCIe fundamental reset (as would happen over a reboot) not disable port power?

timg236, 2020-06-08:
No, PCIe fundamental reset does appear to disable port power. The reboot case is more confusing now because if there is no SD-card or the board has the SD PWR switch (8GB) then the PMIC won't be reset which I think does reset the port power.

PCIE_0_RGR1_SW_INIT_1 |= PCIE_0_RGR1_SW_INIT_1_PCIE_BRIDGE_SW_INIT_SET;
PCIE_0_RGR1_SW_INIT_1 |= PCIE_0_RGR1_SW_INIT_1_PCIE_SW_PERST_SET;

geerlingguy, 2020-06-08:
I'm trying this with a USB SSD, and getting the same issue. On first start, I hit:

[image]

If I unplug the Pi's and re-plug it, it still sees no 'sdcard', but then boots from the USB SSD no problem.

If I do a restart (either via GUI 'Reboot' option or `sudo reboot`), it goes back to getting stuck on the screenshot attached above. While it's stuck there, unplugging the drive and switching to another port makes no difference, but if I then unplug and replug power to the Pi itself, it boots up again, identifying the external SSD USB boot volume successfully:

[image]

(After this it boots up correctly and takes me to the desktop.)

geerlingguy, 2020-06-08:
I just made a comment that I thought may be relevant to this issue, but it was just deleted... I was encountering the same issue and had some extra screenshots/recordings to help with those who may also encounter the issue. Should I repost that information, or did I do something wrong?

timg236, 2020-06-08:
@geerlingguy The comment adds a slightly annoying animation indicating a power issue but gave no information about the hardware being used. Create a new bug report and fill out the required information. 
Interoperability reports are of limited use if the hardware description is just SSD

geerlingguy, 2020-06-08:
That sounds fine; I was just trying to assist others who may encounter the same issue as in the OP and that I was having—basically stating in simple terms that the fix (without further modifying the firmware) is to power down the Pi completely (e.g. unplug it), then power it back on (a 'hard' reboot).

It took me a few cycles and searching for this issue before I connected the dots that this bug (regardless of what kind of USB volume is used) is only related to soft reboots.

timg236, 2020-06-09:
Understood but the assertion that this happens "regardless of what kind of USB volume is used" is not correct. We have successfully tested with multiple adapters and SSDs and so have people on the forums. There may well be a class of devices which require an extended power off on reset but we'll never know what that is if bug reports aren't specific.

timg236, 2020-06-09:
> ST9320325AS

If you have details of the SATA adapter (e.g. an Amazon link) that would be useful.

lurch, 2020-06-09:
Just as a point of comparison, I wonder if @ghollingworth encountered similar issues with the USB MSD boot on 3B / 3B+ ?

wildrun0, 2020-06-09:
> > ST9320325AS
> 
> If you have details of the SATA adapter (e.g. an Amazon link) that would be useful.

I've searched this adapter in internet and found nothing useful. I buy this in a local store so i guess this adapter from china (aliexpress i think). All the useful information i found: 
Connectors: USB Am 4pin,USB 3.0 Am 9pin, SATA+powerSATA

timg236, 2020-06-09:
No problem, I think we are going to have to get hold of a selection of adapters because logging or even protocol analysers don't really help if the USB port doesn't detect the device.

wildrun0, 2020-06-09:
I can send a photo of an adapter if it helps

wildrun0, 2020-06-09:
So here it is
[image]

geerlingguy, 2020-06-09:
FYI the adapter I'm currently testing with is the Inateck 2.5" SATA HDD External Enclosure, [FEU3NS-1](https://www.amazon.com/Inateck-Drive-Enclosure-External-FE2001/dp/B00DW374W4) on Amazon. With that enclosure, the Pi 4 does not soft reboot successfully.

I also have an Inland SATA adapter and one from Startech.com, so I'll try with those and see if they allow a soft reboot or if they fail in the same way.

timg236, 2020-06-10:
> FYI the adapter I'm currently testing with is the Inateck 2.5" SATA HDD External Enclosure, [FEU3NS-1](https://www.amazon.com/Inateck-Drive-Enclosure-External-FE2001/dp/B00DW374W4) on Amazon. With [...]

Thanks, I can now add that to the purchase order :)

Please could you try the soft reboot with rpi-eeprom-recovery-c1325cd20.zip which is attached to this bug, flash using the Raspberry Pi Imager. If you happen to have a powered USB hub then it would be useful to know if that changes behaviour with the experimental bootloader.

timg236, 2020-06-16:
Resolved in pieeprom-2020-06-15

closed 2020-06-16 by timg236

lurch, 2020-06-16:
Just out of curiosity @timg236 , did you manage to work out why some devices _need_ a power-cycle after a warm-reboot? (but are fine from a cold-boot)

matonb, 2020-08-25:
Just an FYI

I recently bought 4 RPi 4's, which I immediately updated all 4 to the latest firmware:
```
Jul 31 2020 14:43:39
version f8d1dc69144b10b4bd78b4b6d08658ac3005726d (release)
timestamp 1596203019
```

Two of the pi's boot from SSD no problem at all, but two failed with `failed to find sd card` etc. etc.

After a bit of trial and error I eventually updated/downdraded the bootloader to the 16th July 2020 version `pieeprom-2020-07-16.bin` on the two troublesome Pi's, which then succesfully booted from SSD.

I then updated the bootloader to the 31st July version and they still booted.

It would appear that the 16th July version does 'something' which the 31st July version does not...

spaskalev, 2020-09-05:
I am also experiencing a soft-reboot stuck with the latest stable eepron on a 8gb rpi 4, an non-powered orico usb-c to sata adapter and a sandisk sata ssd, powered only through the adapter. On a soft reboot the bootloader screen will report that an SD card is found while there is none, iterate through the usb devices and appears stuck afterwards. A hard reset will make it start.

`
Jun  8 2020 10:30:11
version c1325cd20901c069d2af151508b910b35c7a2810 (release)
BOOT_ORDER=0xf41
`

Laser128128, 2020-09-14:
Hi All

I also have the same issue, but even an extended power down (with the cable removed) is not solving my issue.  I think my issue is that I used a NVME SSD and used a NVME to USB3 adaptor instead of SATA to USB3?  The drive does show up as removable storage in raspbian and I can copy the SDCard contents to it but it doesn't get recognised during boot.  

If someone with more knownledge could just confirm that my issue is due to using NVME rather than SATA then that will be great.  Thanks

matonb, 2020-09-14:
@Laser128128 Did you try upgrading to `pieeprom-2020-07-16.bin` first ?

Laser128128, 2020-09-14:
Hi @matonb 

Yes I did.  I managed to get it working in the end by following this guide:
https://jamesachambers.com/raspberry-pi-4-usb-boot-config-guide-for-ssd-flash-drives/

It is lightning fast now and I have more storage space to work with.  I still can't get it to work completely without the SD card. From my limited understanding it still needs the SD card to tell the pi to boot from the SSD, I can live with this for now and will probably look at moving away from the SD card completely at a later stage.

Thanks

matonb, 2020-09-14:
Hi @Laser128128,

  Maybe your combination of NVME drive and Adpator is missing the seet spot at the mo.
  I'm using Inateck USB 3.0 External Hard Drive Case FE2004C, with either SANDisk or Crucial SSD drives.

  When I got USB Boot to work I no longer need an SD card installed at all.
  I used Jeff Geerlings guide as a starting point: [https://www.jeffgeerling.com/blog/2020/im-booting-my-raspberry-pi-4-usb-ssd](https://www.jeffgeerling.com/blog/2020/im-booting-my-raspberry-pi-4-usb-ssd)

Regards,
Brett

geerlingguy, 2020-09-14:
@Laser128128 - That blog post was written before the Pi Foundation released new firmware updates to support direct booting from USB drives.

You need to follow the updated process to make sure your Pi's firmware has this feature enabled. See my more recent blog post: https://www.jeffgeerling.com/blog/2020/im-booting-my-raspberry-pi-4-usb-ssd

matonb, 2020-09-14:
@geerlingguy I had problems getting USB Boot to work if I didn't update to `pieeprom-2020-07-16.bin` first.

That happend on two fo the 8GB pi's I bought and one 2GB so it's a bit hit and miss...

HarvsG, 2020-11-11:
I'm having the same problem with a USB flash drive on RPI 400

marianopw, 2020-11-27:
I can confirm this is happening with an SSD Intel 454s (Using 2020-07-16 eeprom and also latest update)
I have a cluster with 4 Raspberry pi 4 and only the one with the intel SSD is failing to boot

USB Adapter: All raspis are using ADATA ED600 adapter.
SSD Drives: 3 ADATA SU630 (Working OK)
                    1 Intel 454s Not booting (Showing Failed to open device: 'sdcard')

Swapping drives confirms that the problem is the intel drive

timg236, 2020-11-27:
The original issue was resolved. If other configurations are failing then it will be a different issue albeit with similar symptoms. For hardware interop please create a separate issue if the USB devices are not the same as the original report.