raspberrypi/rpi-eeprom #161: Fastest Net Boot - Http

Synoia, 2020-06-18:
This is not a bug, it is a suggested enhancement

TCP is the fastest protocol we have. It was implemented on a PI 3 in u-boot. It reduced net boot from AWS from 4 minutes to 20 seconds. Is it possible;e to add it to the native PI net boot?

I am willing to do the work. Or share the code.

lurch, 2020-06-18:
Http? TCP? AWS?

I don't have anything to do with the bootloader, but perhaps more details, and an explanation of why it's faster than the current netboot method, would be useful? :shrug:

ghollingworth, 2020-06-18:
We are likely to provide some capability to enable the downloading of an image over http in due course.

Gordon

Synoia, 2020-06-18:
Most protocols ask for a packet of data, wait, and then ask for the next packet. The historical boot protocol  TFTP works in that manner. Most communication protocols from the 70 and 80s work in that manner, poll/response behavior. Networks were slow then, 9600 bps was considered fast, and latency very small.

With HTTP there is one request for the whole file, and the file is sent as quickly as possible. Overruns are catered for by the error recovery built into http, the sliding window, which does not slide if the stream has gaps in it..

Synoia, 2020-06-18:
Gordon, want some help?

Synoia, 2020-06-18:
I will add that NFS is also an extremely inefficient protocol, too much chat between client and server,  and not well suited to today's systems.

ghollingworth, 2020-06-18:
Maybe when it comes to testing, but for the moment it's best to wait until we've moved forward with the functionality

Synoia, 2020-06-18:
Gordon
The TCP code written for u-boot is under my Copyright in my spare time (Which I have in abundance).  It is not a general purpose stack.  I understand your desire for confidentiality, control and ownership, I was suggesting something expedited.
Regards

Duncan Hare

714 931 7952

Synoia, 2020-06-18:
Andrew: 
HTTP - A Streaming protocol used by web browsers. H - Hyertext, Transposrt, T ?, P - Protocol.
AWS Amazon Web Services, a set of server farms available for rent.
TCP the Transmission Control Protocol, used by HTTP.

closed 2020-06-18 by Synoia