raspberrypi/rpi-eeprom #162: Kingston A400 480Gb & 120Gb SSD's Not working with latest RPi firmware USB Boot

jtremblant, 2020-06-19:
Does NOT work with Kingston A400 480Gb and A400 240Gb. Shows some nasty messages like "start4.elf is NOT compatible" and "USB-MSD boot requires newer software" I tested both brand new ssd drives on windows and Mac, checked if there's a new firmware from kingston, and both were up to date. Run some heavy tests just to make sure they were not defective. All came up clean. Tried again with etcher, then pi imager. Same results. Tried on new rpi 4 8gb, same results. Tried using different brands of USB to SSD adapters, same result. Tried with different official and non-official 5.1V 3A and 5.1 2.5A AC adapters. All my USB to SSD adapters and caddys are using ASmedia 1153E which is a well known linux compatible chipset(UAS and TRIM enabled without QUIRKS and runs out of the box beautifully for raspbian 32 bit and ubuntu mate). Went back to my old trusted method, microsd 32Gb to boot and USB SSD drive for regular operations.Tested it with latest stable pieeprom-2020-06-15.bin.

lurch, 2020-06-19:
I believe that for USB boot to work, then you need both an updated eeprom and _also_ an updated start4.elf - if you're seeing a message saying "start4.elf is NOT compatible" then I guess you're missing the latter?
Read the stuff about `rpi-update` in https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711_bootloader_config.md and let us know if it still doesn't work after that.

jtremblant, 2020-06-19:
Thank you for the quick response, I followed this guide:
https://www.youtube.com/watch?v=pJdIG0J7QOA
I Basically replaced all 14 files on /boot/ including all *.ELF files from
a backup he made from a working 64bit Raspian. Still getting some messages
at cold boot trying to boot from SDcard first, after doing a raspian 32
bit's full-upgrade, it goes thru that stage really quick.. I hope it will
be fixed when a production Rpi firmware rolls out. Other than that, once it
loads up, seems to be working as intended. Still going to run some
benchmarks on it. I guess you can now mark it up my inquiry as resolved.
Thank you again.

lurch, 2020-06-20:
Unfortunately "cargo cult" methods like "copy all 14 files from the /boot directory of one OS to the /boot a different OS" have the potential to break things if the OS being copied FROM uses a different kernel-version to the OS being copied TO.

timg236, 2020-06-22:
Thanks @lurch closing issue

closed 2020-06-22 by timg236