raspberrypi/rpi-eeprom #163: Can not change bootloader configuration

mkreisl, 2020-06-19:
I try  to change my bootloader configuration from
```
[all]
BOOT_UART=0
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0
DHCP_TIMEOUT=45000
DHCP_REQ_TIMEOUT=4000
TFTP_FILE_TIMEOUT=15000
TFTP_IP=
BOOT_ORDER=0x21
SD_BOOT_MAX_RETRIES=3
NET_BOOT_MAX_RETRIES=5
[none]
FREEZE_VERSION=0
```
to
```
[all]
BOOT_UART=0
WAKE_ON_GPIO=0
POWER_OFF_ON_HALT=1
DHCP_TIMEOUT=45000
DHCP_REQ_TIMEOUT=4000
TFTP_FILE_TIMEOUT=15000
TFTP_IP=
BOOT_ORDER=0xf421
SD_BOOT_MAX_RETRIES=3
NET_BOOT_MAX_RETRIES=5
ENABLE_SELF_UPDATE=1
DISABLE_HDMI=0
[none]
FREEZE_VERSION=0
```

I have done this several times as described here:
https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711_bootloader_config.md

However, I already have the latest firmware installed and according to `rpi-eeprom-update -h` the configuration is not changed if the firmware is up to date.

I'm sorry, am I misunderstanding something or if not, who is making this up?
```
   -d Use the default bootloader config, or if a file is specified using the -f
      flag use the config in that file. This option only applies when a
      bootloader EEPROM update is needed; if the bootloader EEPROM is up-to-date
      then its config will not be changed.
   -f Install the given file instead of the latest applicable update
      Ignores the FREEZE_VERSION flag in bootloader and is intended for manual
      firmware updates.
```
In my opinion the whole rpi-eeprom concept is a chaos and totally user unfriendly

This is what I did:
```
root@kmxbilr2:/tmp# cp /lib/firmware/raspberrypi/bootloader/stable/pieeprom-2020-06-15.bin pieeprom.bin
root@kmxbilr2:/tmp# cat bootconf.txt 
[all]
BOOT_UART=0
WAKE_ON_GPIO=0
POWER_OFF_ON_HALT=1
DHCP_TIMEOUT=45000
DHCP_REQ_TIMEOUT=4000
TFTP_FILE_TIMEOUT=15000
TFTP_IP=
BOOT_ORDER=0xf421
SD_BOOT_MAX_RETRIES=3
NET_BOOT_MAX_RETRIES=5
ENABLE_SELF_UPDATE=1
DISABLE_HDMI=0
[none]
FREEZE_VERSION=0
root@kmxbilr2:/tmp# rpi-eeprom-config --out pieeprom-new.bin --config bootconf.txt pieeprom.bin
root@kmxbilr2:/tmp# rpi-eeprom-update -d -f ./pieeprom-new.bin
BCM2711 detected
Dedicated VL805 EEPROM detected
BOOTFS /boot
*** INSTALLING ./pieeprom-new.bin  ***
BOOTFS /boot
EEPROM update pending. Please reboot to apply the update.
root@kmxbilr2:/tmp# reboot^C
root@kmxbilr2:/tmp# ls -la /boot/pi*
-rwxr-xr-x 1 root root     65 Jun 19 11:56 /boot/pieeprom.sig
-rwxr-xr-x 1 root root 524288 Jun 19 11:56 /boot/pieeprom.upd
root@kmxbilr2:/tmp# reboot
```
After reboot:
```
root@kmxbilr2:~# vcgencmd bootloader_config
[all]
BOOT_UART=0
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0
DHCP_TIMEOUT=45000
DHCP_REQ_TIMEOUT=4000
TFTP_FILE_TIMEOUT=15000
TFTP_IP=
BOOT_ORDER=0x21
SD_BOOT_MAX_RETRIES=3
NET_BOOT_MAX_RETRIES=5
[none]
FREEZE_VERSION=0
```
You can see, NOTHING has been changed

lurch, 2020-06-19:
Are you booting from SD-card or network or USB? If it's one of the latter two, then note that the `ENABLE_SELF_UPDATE` can only be enabled by booting from SD-card (as mentioned on https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711_bootloader_config.md).

mkreisl, 2020-06-19:
Of course I'm booting from sd-card, because I WANT TO ENABLE USB BOOT

lurch, 2020-06-19:
Can you verify what `vcgencmd bootloader_version` says?

According to https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711_bootloader_config.md in `pieeprom-2020-05-15.bin` the default `BOOT_ORDER` was changed to `0xf41` :confused:

mkreisl, 2020-06-19:
```
root@kmxbilr2 ~ # vcgencmd bootloader_version
Jun 15 2020 14:36:19
version c302dea096cc79f102cec12aeeb51abf392bd781 (release)
timestamp 1592228179
```
I already told, bootloader ist up2date

mkreisl, 2020-06-19:
But anyway, somehow I managed to get the Pi4 to boot from USB, network and SD just like the Pi3. After over 4 hours!

In the meantime the update process even broke my /boot, so the Pi didn't want to boot at all. The green LED was always blinking somehow.

I will still think about putting it into XBian, but the whole procedure is much too chaotic, the users should be able to deal with Raspberry Pi OS if they want to have a Raspberry Pi4 bootloader EEPROM update

lurch, 2020-06-20:
Glad to hear that you got it working eventually, sorry that it took you so long!
(as a disclaimer: I don't work on the bootloader myself, I just try to be helpful where I can)

If it helps, I did some investigation into the various bootloader update methods a couple of weeks ago - see my big comment [here](https://github.com/raspberrypi/documentation/pull/1585#discussion_r436354112) (but that's just my best guess, there's no guarantee it's 100% correct).

From what I can tell, if you're updating the bootloader using `rpi-eeprom-update` and an SD-card, then it should be using the `recovery.bin` update method, which (AFAIK) should write your bootloader-with-modified-config to EEPROM _regardless_ of what version is already in EEPROM.

Do you happen to remember what "the magic trick" was that got it working for you?

mkreisl, 2020-06-20:
First of all, thanks for your help.

No, unfortunately I can't understand this anymore. after it somehow broke the /boot directory, I copied the necessary bootloader (pieeprom.* and recovery.bin) files back in after recovering /boot, and when I rebooted, it ran without errors and the bootconfig was also taken over

By the way, the day before, I had similar problems updating the bootloader alone without changing the configuration, but somehow it worked after several attempts

Unfortunately I can't remember if I had similar problems the first time about 8 months ago when the netboot was available

But no matter, now it works the way I want it to (boot from SD -> USB -> network) and as long as there are no serious errors, I will definitely not change that anymore

timg236, 2020-06-22:
It seems unlikely that rpi-eeprom-update broke the boot directory given that it just copies some files. If the file-system was in an inconsistent state then it's possible that that the bootloader would not see pieeprom.upd / recovery.bin. 
If you have a UART log of the update failing or an image of the boot partition which fails then please attach, otherwise, I'll close the bug since this local test and forum comments indicate that the  recovery.bin in the stable release directory is working for other people.

timg236, 2020-07-02:
Not reproducible so closing bug.

closed 2020-07-02 by timg236

jonalmeida, 2021-07-11:
Hello folks, I was running into the same issues as @mkreisl and came across this issue, so I thought I'd share my experiences.

For me at least, it turned out to be my SD card had reached a corrupted state and part of it's safety mechanism is to turn into a permanent read-only system. I confirmed this by trying to write any file to the disk and rebooting (the edited files revert to the prior state). What you might have guessed is that it means, after changes made to `boot/config.txt` those changes are lost during the reboot, and hence the USB boot configuration would never be set.

My solution to this was to make a copy of this SD card with `dd` that I could then boot from with read/write and performt the changes after that.

Another `dd` of the new SD card onto the USB, and I have it booting from it now.

Hope that helps close this loop @timg236 .  🙂