raspberrypi/rpi-eeprom #166: USB device breaks wireless mouse/keyboard

rtfmoz2, 2020-06-23:
For general boot questions please check the read the [Boot Problems](https://www.raspberrypi.org/forums/viewtopic.php?t=58151) sticky post on the forums.  

**Describe the bug**
Booting with [UGREEN USB 3.0 SSD enclosure](https://www.amazon.com.au/gp/product/B06XWRRMYX) containing [Intel 600GB SSD](https://ark.intel.com/content/www/us/en/ark/products/56569/intel-ssd-320-series-600gb-2-5in-sata-3gb-s-25nm-mlc.html) while using [Logitech USB wireless](https://www.logitech.com/assets/41520/2/wireless-combo-mk220-qsg.pdf)  keyboard/mouse renders the keyboard/mouse unusable.

**To Reproduce**
Boot with devices attached. Cannot login as keyboard only responds occasionally. In desktop mode the mouse only moves in pulses.

**Expected behaviour**
Keyboard and Mouse work normally as they do when USB drive is not present.

**Bootloader version and configuration**
https://gist.github.com/rtfmoz2/37f0c3579415c74e4b5d7d772c923215

**USB boot (please complete the following information):**
lsusb -vvv
https://gist.github.com/rtfmoz2/1d1ad48f4cdd5987bd688c4472c2b367

rtfmoz2, 2020-06-23:
Ok where should I be posting this.

timg236, 2020-06-23:
The Linux USB stack is independent of the bootloader, there are some similar bugs on the Linux repo, although this may be an RF interference issue

rtfmoz2, 2020-06-23:
Ok thanks, will post there. Sorry, i just wasn't sure.

closed 2020-06-23 by rtfmoz2

timg236, 2020-06-23:
No problem, if you have a HUB or try connecting the mouse or keyboard via a that because that may help avoid any RF interference from the USB.

lurch, 2020-06-23:
You also ought to try connecting your SSD _after_ Linux has booted (from an SD card) to see if that also causes keyboard & mouse problems.
And if you have one, you ought to try using a powered USB hub in case the Pi's USB ports aren't able to supply enough current to all devices.

rtfmoz2, 2020-06-23:
I posted this over at linux issues. Suggested a cable to push the wireless reciever futher away from the Pi. This resolved the issue. Seems the SSD puts out some signal interference. I may try wrapping it in alfoil and see if that confirms it.

rtfmoz2, 2020-06-23:
> I posted this over at linux issues. Suggested a cable to push the wireless reciever futher away from the Pi. This resolved the issue. Seems the SSD puts out some signal interference. I may try wrapping the SSD in aluminium foil and see if that confirms it.

I did this and it didn't help. Will investigate further...