raspberrypi/rpi-eeprom #167: Add a way to hide the HDMI diagnostics screen during a successful system bootup

dividuum, 2020-06-23:
**Describe the bug**
It's more of a feature request. Starting with version pieeprom-2020-06-15.bin, the HDMI diagnostics screens seems to always start. It shows some boot messages and (on a correctly configured system) the boot continues normally and Linux boots up eventually.

In previous releases the HDMI diagnostics screen only appeared if the boot process couldn't continue beyond the initial eeprom. This also seems how it's supposed to work, at least according to the [documentation](https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711_bootloader_config.md) (jump to the DISABLE_HDMI section).

For aesthetic reasons, I wish to disable the HDMI diagnostics screen unless it's actually useful to debug a failed boot. Unless I missed an option somewhere, there's no way to achieve this at the moment.

**To Reproduce**
Install pieeprom-2020-06-15.bin eeprom release and boot. The HDMI diagnostic screen shows up briefly while the system boot successfully into Linux.

**Expected behaviour**
A clear and concise description of what you expected to happen. The HDMI diagnostic screen only shows up if the boot fails.

timg236, 2020-06-23:
Yes, this is a change in behaviour because we decided that it was no longer appropriate to force the user to disconnect the power in order to reboot. Consequently, it's not a "screen of death" and needs to be enabled early on.

disable_splash in config.txt is no longer checked by the bootloader because it might not be able to load config.txt

Right now, the only way to block this is via DISABLE_HDMI and revert to UART or NETCONSOLE diagnostics.

It might be reasonable to repurpose DISABLE_HDMI (or add yet another config variable) to only enable after the first restart has been attempted. 

Adding this as a low priority feature request.

dividuum, 2020-06-23:
Not sure t.b.h. if I understand your first paragraph. I figured the code looked something like this:

```
while ((boot_method = next_boot_method())) {
   if (retrieve_firmware_files(boot_method)) {
      jump_to_firmware() // does not return
   }
}
show_diagnostics_screen()
```
and the "only" change was to move the diagnostics_screen call at the start of that, so I'm not sure how power-cycling plays into that. I'm probably totally wrong :)

I'll use `DISABLE_HDMI` for now. Thanks for the quick response.

timg236, 2020-06-23:
The pseudo code is valid but there's a few complications

- If 0xf is set in BOOT_ORDER then the loop will never terminate, this is enabled by default in the updated raspi-config. This means that if a user forgets to insert an SD or USB device then it can be hot-plugged and it will boot.
- Network boot and even USB-MSD (with spinning rust) can be slow to boot so progress information is useful
- It's common to put SD first even if Network is desired because it's useful to be able to override network boot if the network connection has failed or worse is unusably slow.

So, at what point has boot failed?

There has to be some method for never enabling HDMI for kiosk/signage applications.

Another alternative is that if we add minimal USB keyboard (boot not full HID) then there could be some minimal user interaction but I really don't want to end up with GRUB here :)

dividuum, 2020-06-23:
Thanks a lot for the insight! Totally makes sense now.

lurch, 2020-06-23:
Maybe this'd be a good place to use a [conditional filter](https://www.raspberrypi.org/documentation/configuration/config-txt/conditional.md) ?
E.g. I _think_ if you had
```
[gpio4=1]
DISABLE_HDMI=1
[gpio4=0]
DISABLE_HDMI=0
[all]
```
in your EEPROM config, then the HDMI diagnostics screen would disabled if GPIO4 is high, but shown if you pull GPIO4 low?

cleverca22, 2020-07-11:
last i checked, there is ~280kb free (before the compressed memsys stuff came around) on the eeprom

what if the main `bootcode.bin` just had very very basic usb keyboard stuff, that could only detect if you hit F12 or similar

when that is detected, it would load a second binary from eeprom (and dram is up, so no more L2 cache limitation)

and that secondary binary then gave a bit of a BIOS config UI, so you could change any setting in `bootconf.txt` or temporarily override things, such as forcing it to netboot or usb boot just once

timg236, 2020-07-11:
I’m very much against a bios settings UI. It’s unnecessary and confusing for most users. There may be other uses for remaining space eg for industrial customers to provision certs, localised content or improved ui

timg236, 2020-11-06:
@dividuum The pieeprom-2020-10-28 beta now defers rendering to the HDMI display for 5 seconds unless a fatal error (or self-update failure) occurs. This can be overridden via the HDMI_DELAY property.

dividuum, 2020-11-06:
Sounds great. I'll give it a test run tomorrow.

lurch, 2020-11-09:
> This can be overridden via the HDMI_DELAY property.

Should that be added to https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711_bootloader_config.md ?

timg236, 2020-11-09:
> > This can be overridden via the HDMI_DELAY property.
> 
> Should that be added to https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711_bootloader_config.md ?

I'm only updating the docs when a feature is available in the stable release

dividuum, 2020-11-10:
Using the firmware seems to work well. The diagnostics screen didn't show up during any of the normal bootups and only when I removed the SD card. From my point of view this issue can be closed, unless you want to keep it open until it's officially in the stable release. Thanks!

timg236, 2020-11-10:
Thanks for the test. I'll close this now and update the docs as soon as it goes stable. I'll update the docs as soon as it's stable but I can't see this feature changing between now and then.

closed 2020-11-10 by timg236