raspberrypi/rpi-eeprom #168: 500ms may be too short for usb port-power reset

hhmob, 2020-06-23:
When reboot a raspberry4 which boots from usb,  most of u-disks can boot,  
but some usb device (such as Geekworm X872) fails unless "unplug then plug". 
Could you please extend the time resetting USB port-power after "Failed t open device: 'sdcard'"? 
Maybe 3 seconds?

timg236, 2020-06-23:
It supports up 1 second (1000ms) already so please try that first, it seems unlikely that another two seconds would make any difference.
Marking as low priority enhancement, since it's just upping a limit. The limit is just there to avoid accidentally blocking boot for a very long time.

hhmob, 2020-06-23:
> It supports up 1 second (1000ms) already so please try that first, it seems unlikely that another two seconds would make any difference.
> Marking as low priority enhancement, since it's just upping a limit. The limit is just there to avoid accidentally blocking boot for a very long time.

Thanks, how can I set it to 1000ms? I would like to have a try.

timg236, 2020-06-23:
There's a new parameter which has yet to make it into the official docs. Update eeprom config  to include the following line https://www.raspberrypi.org/documentation/hardware/raspberrypi/booteeprom.md

USB_MSD_PWR_OFF_TIME=1000

hhmob, 2020-06-23:
> There's a new parameter which has yet to make it into the official docs. Update eeprom config to include the following line https://www.raspberrypi.org/documentation/hardware/raspberrypi/booteeprom.md
> 
> USB_MSD_PWR_OFF_TIME=1000

Thanks, I find it in https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711_bootloader_config.md

hhmob, 2020-06-23:
Great! 
After changing USB_MSD_PWD_OFF_TIME to 1000, X872 works after REBOOT !!!!!
It seems SSD2USB boards like X872 need longer USB_MSD_PWD_OFF_TIME than u-disk.
I suggest that default value of USB_MSD_PWD_OFF_TIME set to 1000. 

How to make change to bootloader configuration:
```
pi@raspberrypi:~/newfirmware $ cp /lib/firmware/raspberrypi/bootloader/stable/pieeprom-2020-06-15.bin ./pieeprom.bin
pi@raspberrypi:~/newfirmware $ rpi-eeprom-config pieeprom.bin > bootconfig.txt
pi@raspberrypi:~/newfirmware $ cat bootconfig.txt 
[all]
BOOT_UART=0
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0
DHCP_TIMEOUT=45000
DHCP_REQ_TIMEOUT=4000
TFTP_FILE_TIMEOUT=30000
ENABLE_SELF_UPDATE=1
DISABLE_HDMI=0
BOOT_ORDER=0xf41

pi@raspberrypi:~/newfirmware $ vi bootconfig.txt 
pi@raspberrypi:~/newfirmware $ cat bootconfig.txt 
[all]
BOOT_UART=0
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0
DHCP_TIMEOUT=45000
DHCP_REQ_TIMEOUT=4000
TFTP_FILE_TIMEOUT=30000
ENABLE_SELF_UPDATE=1
DISABLE_HDMI=0
USB_MSD_PWR_OFF_TIME=1000
BOOT_ORDER=0xf41

pi@raspberrypi:~/newfirmware $ rpi-eeprom-config --out pieeprom-new.bin --config bootconfig.txt pieeprom.bin 
pi@raspberrypi:~/newfirmware $ rpi-eeprom-config pieeprom-new.bin 
[all]
BOOT_UART=0
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0
DHCP_TIMEOUT=45000
DHCP_REQ_TIMEOUT=4000
TFTP_FILE_TIMEOUT=30000
ENABLE_SELF_UPDATE=1
DISABLE_HDMI=0
USB_MSD_PWR_OFF_TIME=1000
BOOT_ORDER=0xf41

pi@raspberrypi:~/newfirmware $ sudo rpi-eeprom-update -d -f ./pieeprom-new.bin
BCM2711 detected
VL805 firmware in bootloader EEPROM
BOOTFS /boot
*** INSTALLING ./pieeprom-new.bin  ***
BOOTFS /boot
EEPROM update pending. Please reboot to apply the update.
pi@raspberrypi:~/newfirmware $ vcgencmd bootloader_config
[all]
BOOT_UART=0
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0
DHCP_TIMEOUT=45000
DHCP_REQ_TIMEOUT=4000
TFTP_FILE_TIMEOUT=30000
ENABLE_SELF_UPDATE=1
DISABLE_HDMI=0
USB_MSD_PWR_OFF_TIME=1000
BOOT_ORDER=0xf41

pi@raspberrypi:~/newfirmware $ sudo reboot
```

lurch, 2020-06-23:
> I suggest that default value of USB_MSD_PWD_OFF_TIME set to 1000

That would make the boot take (slightly) longer for all users though.

hhmob, 2020-06-23:
> > I suggest that default value of USB_MSD_PWD_OFF_TIME set to 1000
> 
> That would make the boot take (slightly) longer for all users though.

500ms can save countless people, just 500ms :)

lurch, 2020-06-23:
This is why user-testing is so invaluable, to find out just how many users need the port-power reset to be increased to 1000ms :)

@hhmob if you have the time to spare, it might be worth experimenting to find out roughly where the cut-off between not working (with 500ms) and working (with 1000ms) is?

hhmob, 2020-06-23:
> This is why user-testing is so invaluable, to find out just how many users need the port-power reset to be increased to 1000ms :)
>
> @hhmob if you have the time to spare, it might be worth [...]

OK, I will do it asap. I think many of those who use ssd2usb boards may encounter the reboot issue due to short USB_MSD_PWD_OFF_TIME.

ericf, 2020-06-28:
I was running into this same issue with my RPi 4s that have the same Geekworm X872 USB/NVMe SSD boards attached. I wanted to add a note that I was also able to fix `sudo reboot` issue using:
```
USB_MSD_PWR_OFF_TIME=1000
```

hhmob, 2020-06-29:
> I was running into this same issue with my RPi 4s that have the same Geekworm X872 USB/NVMe SSD boards attached. I wanted to add a note that I was also able to fix `sudo reboot` issue using:
> 
> ```
> USB_MSD_PWR_OFF_TIME=1000
> ```

@lurch @ericf 

Great !!! I have tested USB_MSD_PWR_OFF_TIME=600, that also works for X872!

timg236, 2020-07-07:
In the 2020-07-06 release, the default is now 1000ms and the valid range is 250ms to 5000ms. Zero may still be used to prevent the USB port power off.

closed 2020-07-07 by timg236

Giuseppe-P, 2021-01-04:
i installed raspbian lite on sd card and followed the procedure to change the boot file to USB_MSD_PWR_OFF_TIME ... but this file i can't find it is empty when edited ...
I had an active hassOS on SSD but it no longer boots with the latest updates

can someone help me understand why the boot file is empty in the installed raspbian lite OS in order to perform firmware update

timg236, 2021-01-05:
Removed Pi3 specific comments from an already closed bug.