raspberrypi/rpi-eeprom #172: Does not boot 2020-05-27-raspios-buster-armhf.zip

ilg-ul, 2020-06-28:
**Describe the bug**

The June 15 bootloader does not boot the 2020-05-27 **32-bit** image (but the 64-bit image is ok).

**To Reproduce**

- copy the `2020-05-27-raspios-buster-armhf.zip` to any USB stick known to work (I used a SanDisk Ultra USB 3.0 64 GB with balenaEtcher)
- plug the USB stick into RPi 4 USB 3.0 lower port
- power on

**Expected behaviour**

- the system boots and starts install

**Screenshots**

```
bootloader: c302dea0 Jun 15 2020

Read config.txt bytes 1784 ...
recovery4.elf not found (6)
Read start4.elf bytes 2252416 ...
Read fixup4.dat bytes 5400 ...
0x00c03112 0x0 0x0f
start4.elf: is not compatible
USB-MSD boot requires newer software
Get the latest software from https://www.raspberrypi.org/gownloads
ERROR: 00000044
```

**Bootloader version and configuration**

```
$ sudo vcgencmd bootloader_version
Jun 15 2020 14:36:19
version c302dea096cc79f102cec12aeeb51abf392bd781 (release)
timestamp 1592228179
```

**Additional context**

Exactly the same system with the same SanDisk USB stick boots and installs properly the 64-bit image `2020-05-27-raspios-buster-arm64.zip`.

I saw several messages in the forum suggesting all kind of wizardry, like overwriting files, but I'm not sure this is the right way, in this case the boot image is the official Raspberry Pi OS .zip file, normally it should be consistent.

I friend of mine had the same experience, so I guess the issue can be reproduced easily.

lurch, 2020-06-28:
This is because to be able to USB-boot you need both a new-enough bootloader (in the EEPROM) **and** a new-enough `start4.elf` (on the USB stick).
The `start4.elf` in the (stable) 32-bit Raspberry Pi OS isn't new enough to USB-boot (which is why you get the "start4.elf: is not compatible" message), but the `start4.elf` in the (beta) 64-bit Raspberry Pi OS _is_ new enough to USB-boot.

The documentation at https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711_bootloader_config.md explains what you need to do. (basically you need to copy the *.elf and *.dat files from https://github.com/raspberrypi/firmware/tree/master/boot to your USB stick)

ilg-ul, 2020-06-28:
Thank you @lurch for the explanation, it makes sense.

However I could not find any explanation why the stable 32-bit image, which was released in the same day as the 64-bit image, does not include the updated file.

> The documentation at https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711_bootloader_config.md explains what you need to do.

Indeed, down the page there are some details, but they are very cryptic. 

Perhaps you can update the README in this project to better explain the procedure to make the boot disk for both 32 and 64-bit.

Any idea when the 32-bit image will be updated to include the new files?

lurch, 2020-06-28:
> However I could not find any explanation why the stable 32-bit image, which was released in the same day as the 64-bit image, does not include the updated file(s).

I believe the stable release of Raspberry Pi OS will always use the (possibly older) known-stable `start4.elf` rather than the latest  (possibly unstable) beta version of `start4.elf`.

>> The documentation at https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711_bootloader_config.md explains what you need to do.

> Indeed, down the page there are some details, but they are very cryptic.

Feel free to create an issue in the [docs repo](https://github.com/raspberrypi/documentation) explaining which parts you think need clarifying :slightly_smiling_face:

ilg-ul, 2020-06-28:
I confirm that after replacing the `*.elf` and `*.dat` files in the 2020-05-27 32-bit image with those from the master/boot, the system was able to boot properly.

However, getting the files from this Git project is not easy, the whole repo is about 13.5 GB.

I extracted the `*.elf` and `*.dat` files and archived them, for future use.

In case others face the same issue, the archive can be downloaded from ([firmware-master-20200628.tgz](https://github.com/ilg-ul/raspbery-pi-os-files/releases/download/files/firmware-master-20200628.tgz)).

AaronDewes, 2020-06-29:
You don't need to extract them, you can download them from https://github.com/raspberrypi/firmware/raw/master/boot/fixup4.dat (replace fixup4.dat with the filename you want to download) or download all with this bash script:
```
for FILE in fixup4cd.dat fixup4.dat fixup4db.dat fixup4x.dat start4cd.elf start4db.elf start4.elf start4x.elf;
do wget https://github.com/raspberrypi/firmware/raw/master/boot/$FILE;
done
```

ilg-ul, 2020-06-29:
Thank you @AaronDewes, right, now that I know the repo is huge, I wouldn't clone it but download individual files, as you suggest.

Unfortunately this information was initially missing... :-(

timg236, 2020-07-02:
The requirement to manually upgrade the GPU firmware for the beta release is covered in the EEPROM BETA documentation
https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711_bootloader_config.md

closed 2020-07-02 by timg236