raspberrypi/rpi-eeprom #182: RPi4 Netboot - DHCP option 93

stevem2, 2020-07-11:
Submitting at the request of timg236 on the raspberrypi forum.

The DHCP DISCOVER message sent by rpi4 during netboot contains the value 0 in the parameter for Option 93 "Client System Architecture".  IANA assigns this value to 'IA x86 PC'.

I have an existing network of x86 machines, with Dnsmasq (acting in DHCP proxy mode) providing different filenames to netboot clients, depending on whether they  are in BIOS or UEFI boot mode, based on the value of the 'Client System Architecture' parameter.  Attempting to netboot the RPi4 using this setup results in the RPi4 being sent the filename for x86 machines.  The RPi repeats the DHCP DISCOVER message a couple of times and then moves to the next boot method without trying tftp. 

This appears to be standard behaviour with RPi4 netboot (and possibly RPi3 as well, but I'm unable to try this).

If the RPi4 could advertise a unique 'Client System Architecture' value, then dnsmasq could return the appropriate filename to allow the RPi to continue netbooting.

June 2020 'stable' bootloader, with netbooting enabled..

# Configuration file for dnsmasq.
resolv-file=/usr/local/etc/resolv.conf
dhcp-range=192.168.4.254, proxy, 255.255.255.0
pxe-service=x86PC, "ipxe boot", ipxe.pxe, 192.168.4.253
pxe-service=X86-64_EFI,"ipxe uefi boot", ipxe.efi, 192.168.4.253
enable-tftp
tftp-root=/var/ftpd
tftp-no-fail

Thanks
Steve

lurch, 2020-07-11:
Sounds like this might be similar to #87 ?

cleverca22, 2020-07-11:
the new rpi4 firmware should be sending a combination of 0x34695052 (RPi4 as 4 bytes, backwards), the board revision (so you can detect the exact model, ram, and everything), part of the mac, and the full serial#

that should be enough to detect that its a pi, and do more intelligent stuff in response to it

separate from the above, ive not tested it recently, but does the rpi4 firmware respect the `next-server` and `filename` options in the dhcp server?
ive wanted the tftp and dhcp to be on different IP's, and the `filename` field can be used for the dhcp server to set a custom `TFTP_PREFIX`

trejan, 2020-07-11:
Repeating what I posted on the forum.

None of the IANA registered processor architecture types at https://www.iana.org/assignments/dhcpv6-parameters/dhcpv6-parameters.xhtml#processor-architecture match the Pi netboot system. You'd need to officially request a number for the Pi, squat on something like 0x0015 which is "arm 32 uboot" or pick a random number high up in the range and hope nobody else uses it.

PiServer and netboot setups will need to be updated to add an extra pxe-service line to the dnsmasq configuration as current netboot guide assumes type 0 aka x86 PC.

timg236, 2020-07-11:
@lurch is right the updated id from #87 is the right way to go here because it also includes the board if and serial number
Pi3 never supported the file name option for where it might have made sense for bootcode.bin. For Pi4 it doesn’t really fit since it’s not PXE
Changing from x86 to a different architecture might be an interesting long term thing but I think it’s better to avoid requiring eeprom updates in order to add a pi to a network boot

Ruffio, 2021-03-23:
stevem2 can this issue be closed?

timg236, 2021-10-11:
There’s already enough information to identify a raspberry pi and changing the DHCP request would cause risk unnecessary disruption. Closing