raspberrypi/rpi-eeprom #184: Minimize power consumption when in `halt`

seamusdemora, 2020-07-15:
I [read this on the RPi website some time ago](https://www.raspberrypi.org/forums/viewtopic.php?p=1484347#p1484347), and I took it seriously: i.e. 3-4 mA current draw in `halt` mode. I assumed that once the firmware was rolled out, this would be the default behavior when the system was placed in `halt`/`poweroff` mode. Clearly, that's not the case, as my RPi4 still draws ~ 380mA in `halt`.

Then I heard "rumors" recently that a true "low power state" had been scrapped because a consortium of "HAT makers" didn't like the idea. Then I began exploring [the documentation](https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711_bootloader_config.md), and found this: 

>POWER_OFF_ON_HALT
>
>If 1 and WAKE_ON_GPIO=0 then switch off all PMIC outputs in halt. This is lowest possible power state for halt but may cause problems with some HATs because 5V will still be on. GLOBAL_EN must be shorted to ground to boot. 

And so I've followed the instructions, updated my EEPROM (using stable `pieeprom-2020-06-15.bin` as the basis) with the following values: WAKE_ON_GPIO=0, POWER_OFF_ON_HALT=1. Then, I test. And this is where my question arises: 

Instead of the claimed 3-4 mA, I measure about 40 mA. Everything is dosconnected from the RPi except the USB-C power input. This is far too much to use in the remote battery-powered application I've planned. What must I do to get the current draw down to 3-4 mA?

  * Have I missed something? Am I on the wrong firmware release, etc? Are there other values to be changed?
  * If the PMIC is truly disabled, why does the board still draw 40mA? 
  * Why is the red Power LED on?

cleverca22, 2020-07-16:
even when the PMIC is disabled, the 5v rail is still live, there is nothing that can cut that off from the board

if you want things any lower, your probably going to need an external mosfet that can just cut the entire 5v rail off, when signaled by a GPIO

and there should be something in config.txt to signal a GPIO on shutdown

seamusdemora, 2020-07-16:
>and there should be something in config.txt to signal a GPIO on shutdown

I'm not sure I follow you. Do you mean use a GPIO pin after power has been removed? IIUC, the only way to reboot from low power mode is to pull GLOBAL_EN low, and the only way to do that is using external circuitry. Am I missing something?

timg236, 2020-07-16:
3-4 mA looks like a typo, I'd expect about 20 mA which is what I see on 4GB and 8GB boards with that software configuration

seamusdemora, 2020-07-16:
Well that's disappointing - even 20 mA (about twice what I'm getting) isn't quite suited to remote/battery-powered applications.

lurch, 2020-07-17:
@seamusdemora Slightly off-topic, but have you seen https://raspi.tv/2019/how-much-power-does-the-pi4b-use-power-measurements ? Nice comparison graphs :chart_with_upwards_trend: 
Although be aware that that page was written shortly after the Pi 4 was released, and so doesn't include the power-improvements mentioned [here](https://www.raspberrypi.org/blog/thermal-testing-raspberry-pi-4/).

timg236, 2020-07-17:
There’s only one halt state in the bootloader ie petition 63 in PMRSTS register so halt and poweroff should do the same thing.

closed 2020-07-17 by timg236

seamusdemora, 2022-05-29:
> 3-4 mA looks like a typo...

Maybe, but the OP had ample opportunity to edit or update it. And [subsequent dialog on that post](https://forums.raspberrypi.com/viewtopic.php?t=243421&sid=0f48ebe794aa5615ffb4f675e5dd8cc4&start=25#p1825272) offered other explanations - other than "typo" iow. Really quite an amusing exchange :)