raspberrypi/rpi-eeprom #188: NFS only booting does not auto-update firmware

Bl00d-B0b, 2020-07-21:
Hello.

I have boot from NFS, no any other boot devices are connected.
I manage to install new firmware v(15-06-2020) using USE_FLASHROM=1 key, which is deprecated . but as I can see, I still can't update new flash version:

before update:
# rpi-eeprom-update
BCM2711 detected
Dedicated VL805 EEPROM detected
*** UPDATE AVAILABLE ***
BOOTLOADER: update available
CURRENT: Mon 15 Jun 13:36:19 UTC 2020 (1592228179)
LATEST: Thu 16 Jul 15:15:46 UTC 2020 (1594912546)
FW DIR: /lib/firmware/raspberrypi/bootloader/stable
VL805: update available
CURRENT: 000137ad
LATEST: 000138a1

now i make update:
# rpi-eeprom-update -a
BCM2711 detected
Dedicated VL805 EEPROM detected
BOOTFS /boot
*** INSTALLING EEPROM UPDATES ***
BOOTLOADER: update available
CURRENT: Mon 15 Jun 13:36:19 UTC 2020 (1592228179)
LATEST: Thu 16 Jul 15:15:46 UTC 2020 (1594912546)
FW DIR: /lib/firmware/raspberrypi/bootloader/stable
VL805: update available
CURRENT: 000137ad
LATEST: 000138a1
BOOTFS /boot
EEPROM updates pending. Please reboot to apply the update.
# reboot

after reboot:
# rpi-eeprom-update
BCM2711 detected
Dedicated VL805 EEPROM detected
*** UPDATE AVAILABLE ***
BOOTLOADER: update available
CURRENT: Mon 15 Jun 13:36:19 UTC 2020 (1592228179)
LATEST: Thu 16 Jul 15:15:46 UTC 2020 (1594912546)
FW DIR: /lib/firmware/raspberrypi/bootloader/stable
VL805: up-to-date
CURRENT: 000138a1
LATEST: 000138a1

as you can see VL805 firmware is updated, but boot loader is still not !!!

Edward

P.S. this is my boot config:
[all]
BOOT_UART=0
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0
DHCP_TIMEOUT=45000
DHCP_REQ_TIMEOUT=4000
TFTP_FILE_TIMEOUT=30000
ENABLE_SELF_UPDATE=1
DISABLE_HDMI=1
BOOT_ORDER=0xf21

timg236, 2020-07-21:
Best guess is that the permissions were wrong in the /tftpboot directory so pieeprom.bin files were not installed. A UART log would help clarify.

Bl00d-B0b, 2020-07-21:
I use FreeNAS for NSF.
Permissions:
root@nas0[/mnt/Volume/Data/tftpboot]# ls -al
total 96
drwxr-xr-x  6 root  wheel   6 Feb 26 13:47 .
drwxrwxr-x  4 root  wheel   5 Feb 25 14:47 ..
drwxr-xr-x  3 root  wheel  47 Jul 21 15:43 558e8211
drwxr-xr-x  3 root  wheel  49 Jul 21 15:28 a8931679
drwxr-xr-x  3 root  wheel  47 Jul 21 15:33 adf58e87
drwxr-xr-x  3 root  wheel  49 Jul 21 15:28 f3c96ec9
root@nas0[/mnt/Volume/Data/tftpboot]# cd 558e8211
root@nas0[/mnt/Volume/Data/tftpboot/558e8211]# ls -al
total 42502
drwxr-xr-x  3 root  wheel        47 Jul 21 15:43 .
drwxr-xr-x  6 root  wheel         6 Feb 26 13:47 ..
-rw-r--r--  1 root  wheel        41 Jul 21 15:09 .firmware_revision
-rw-r--r--  1 root  wheel     23389 Apr 14 22:03 bcm2708-rpi-0-w.dtb
-rw-r--r--  1 root  wheel     25296 Jul 21 15:08 bcm2708-rpi-b-plus.dtb
-rw-r--r--  1 root  wheel     24644 Jul 21 15:08 bcm2708-rpi-b-rev1.dtb
-rw-r--r--  1 root  wheel     25033 Jul 21 15:08 bcm2708-rpi-b.dtb
-rw-r--r--  1 root  wheel     24766 Jul 21 15:08 bcm2708-rpi-cm.dtb
-rw-r--r--  1 root  wheel     25931 Jul 21 15:08 bcm2708-rpi-zero-w.dtb
-rw-r--r--  1 root  wheel     24738 Jul 21 15:08 bcm2708-rpi-zero.dtb
-rw-r--r--  1 root  wheel     26299 Jul 21 15:08 bcm2709-rpi-2-b.dtb
-rw-r--r--  1 root  wheel     26448 Jul 21 15:08 bcm2710-rpi-2-b.dtb
-rw-r--r--  1 root  wheel     28565 Jul 21 15:08 bcm2710-rpi-3-b-plus.dtb
-rw-r--r--  1 root  wheel     27946 Jul 21 15:08 bcm2710-rpi-3-b.dtb
-rw-r--r--  1 root  wheel     26255 Jul 21 15:08 bcm2710-rpi-cm3.dtb
-rw-r--r--  1 root  wheel     47437 Jul 21 15:08 bcm2711-rpi-4-b.dtb
-rw-r--r--  1 root  wheel     47542 Jul 21 15:08 bcm2711-rpi-cm4.dtb
-rw-r--r--  1 root  wheel     52480 Jul 21 15:08 bootcode.bin
-rwxr-xr-x  1 root  wheel       209 Feb 26 21:03 cmdline.txt
-rwxr-xr-x  1 root  wheel      1906 Jul  3 13:17 config.txt
-rw-r--r--  1 root  wheel     18693 May 27 13:57 COPYING.linux
-rw-r--r--  1 root  wheel      3148 Jul 21 15:08 fixup_cd.dat
-rw-r--r--  1 root  wheel     10266 Jul 21 15:08 fixup_db.dat
-rw-r--r--  1 root  wheel     10266 Jul 21 15:08 fixup_x.dat
-rw-r--r--  1 root  wheel      7274 Jul 21 15:08 fixup.dat
-rw-r--r--  1 root  wheel      5407 Jul 21 15:08 fixup4.dat
-rw-r--r--  1 root  wheel      3148 Jul 21 15:08 fixup4cd.dat
-rw-r--r--  1 root  wheel      8414 Jul 21 15:08 fixup4db.dat
-rw-r--r--  1 root  wheel      8418 Jul 21 15:08 fixup4x.dat
-rwxr-xr-x  1 root  wheel       145 Feb 13 18:09 issue.txt
-rw-r--r--  1 root  wheel   5247472 Jul 21 15:08 kernel.img
-rw-r--r--  1 root  wheel   5599624 Jul 21 15:08 kernel7.img
-rw-r--r--  1 root  wheel   5997544 Jul 21 15:08 kernel7l.img
-rw-r--r--  1 root  wheel  15483392 Jul 21 15:08 kernel8.img
-rw-r--r--  1 root  wheel      1594 May 27 14:22 LICENCE.broadcom
drwxr-xr-x  2 root  wheel       205 Jul 21 12:24 overlays
-rw-r--r--  1 root  wheel        65 Jul 21 15:43 pieeprom.sig
-rw-------  1 root  wheel    524288 Jul 21 15:43 pieeprom.upd
-rw-r--r--  1 root  wheel     98196 Jul 21 15:43 recovery.bin
-rw-r--r--  1 root  wheel    819100 Jul 21 15:08 start_cd.elf
-rw-r--r--  1 root  wheel   4850760 Jul 21 15:08 start_db.elf
-rw-r--r--  1 root  wheel   3759720 Jul 21 15:08 start_x.elf
-rw-r--r--  1 root  wheel   3001440 Jul 21 15:08 start.elf
-rw-r--r--  1 root  wheel   2277376 Jul 21 15:08 start4.elf
-rw-r--r--  1 root  wheel    819100 Jul 21 15:08 start4cd.elf
-rw-r--r--  1 root  wheel   3778856 Jul 21 15:08 start4db.elf
-rw-r--r--  1 root  wheel   3036168 Jul 21 15:08 start4x.elf

Ed

Bl00d-B0b, 2020-07-21:
I do not have possibility to get UART logs, but as you can see pieeprom.upd is here. 

when i change permission to this file to -rw-r--r-- and run rpi-eeprom-update -a after reboot I see permissions are reset to -rw-------

Ed

timg236, 2020-07-21:
rpi-eeprom-update runs as root and /boot is normally a FAT partition, I think it should be possible to set the permissions on the server to give all the files the equivalent of /boot permissions. I'll see if rpi-eeprom-update can set the permissions on file-systems that support that

Bl00d-B0b, 2020-07-21:
Its looks strange for me, because VL firmware do not have this issue. and my boot partition is also on NFS mount, not FAT.

Bl00d-B0b, 2020-07-21:
and file is deleted after successful update.

timg236, 2020-07-21:
The files are deleted by the rpi-eeprom-update daemon running on the Pi which will have access. The read issue is with the TFTP server not running as root.
Try https://github.com/raspberrypi/rpi-eeprom/blob/master/rpi-eeprom-update with an update (e.g. use -d -f to force the update) to verify the permissions now

Bl00d-B0b, 2020-07-21:
I already update all my raspberries with permission update. not sure I can test now. I need to wait new version of firmware.???

or I can just rename file?

timg236, 2020-07-21:
It's probably best you leave it for now. Resolving Isssue as fixed by a5794dde78340a9bceffe31fdc58cd5d355cfc86

closed 2020-07-21 by timg236